mchehab/zbar #3: PDF417 decoding doesn't work?

Naahuel, 2019-02-02:
Hello!

I am attempting to make a [PDF417][1] barcode reader for the web using Web Assembly (there's a few out there but the only reliable one is licenced). The idea is simple, just build a binary file with C code that can be called via JavaScript.

I followed [this guide][2] which got me almost there! I had to add `--enable-codes=pdf417` as a parameter to `configure` to enable PDF417 symbols and disable all the others I'm not interested in.

The code actually works amazingly well for any other type of symbol ([see demo][3]) but if I compile ZBar to accept PDF417, it detects it but throws a warning:

    WARNING: zbar/decoder/pdf417.c:73: pdf417_decode8: Assertion "clst >= 0 && clst < 9" failed.
    dir=0 sig=5a44 k=9 buf[0000]=
    
    WARNING: zbar/decoder/pdf417.c:89: pdf417_decode8: Assertion "g[0] >= 0 && g[1] >= 0 && g[2] >= 0" failed.
    dir=1 sig=ca03 k=6 g0=ffffffff g1=e71 g2=585 buf[0000]=

And of course, no decoding is done. Those assertions are on lines [71][4] and [86][5] of ZBar.

You can see all the code I'm using in the guide I mentioned earlier (the only difference is that I enabled PDF417 before compiling)

For any other type of barcode it works amazingly! Is PDF417 not fully implemented? Is there *anything* I can do to make it work?

  [1]: https://en.wikipedia.org/wiki/PDF417
  [2]: https://barkeywolf.consulting/posts/barcode-scanner-webassembly/
  [3]: https://barkeywolf.consulting/barcode-scanner-webassembly-demo/index.html
  [4]: https://github.com/ZBar/ZBar/blob/master/zbar/decoder/pdf417.c#L71
  [5]: https://github.com/ZBar/ZBar/blob/master/zbar/decoder/pdf417.c#L86

Naahuel, 2019-02-03:
Thanks for answering. That's what I thought, yes. Unfortunately I'm not skilled enough to do that and it looks awfully complex. So I guess I'll just keep looking for some other c library that has that

Naahuel, 2019-02-04:
Thank you @mchehab Hopefully it'll be implemented soon!

I ended up using a [cpp port of ZXing](https://github.com/nu-book/zxing-cpp) which can compile to wasm and be used with WebAssembly. It works Ok, but not better than the pure JavaScript implementation. The output wasm file is also quite large.

mchehab, 2019-02-05:
> I ended up using a [cpp port of ZXing](https://github.com/nu-book/zxing-cpp) 

Unfortunately, ZXing is Apache licensed. For us, the better would be either if someone with time could finish the implementation from scratch or if we could use some code from another LGPL 2.1 compatible project. 

I did a quick search for other ZBar patches that might be floating around, but didn't see anything finishing the PDF417 implementation.

I probably won't have any time myself to do such coding. 

So, volunteers for this task are wanted!

cb49747, 2020-10-15:
Is there any update on getting pdf417 support for zbar?  I would help but way above my skill set.  I'm looking a writing a perl solution to reading these barcodes.

dearsina, 2021-07-14:
Any update to PDF417 support? It's becoming an important barcode format. Thanks.

haimat, 2021-12-05:
Same here - any update on this PDF417 support in zbar?

jameshilliard, 2022-11-06:
> Unfortunately, ZXing is Apache licensed. For us, the better would be either if someone with time could finish the implementation from scratch or if we could use some code from another LGPL 2.1 compatible project.

The [zint implementation](https://github.com/zint/zint/blob/2.11.1/backend/pdf417.c) is in c an should be LGPL2.1 comaptible AFAIU([BSD-3-Clause](https://github.com/zint/zint/blob/2.11.1/backend/pdf417.c#L33)).

vishaljangid1729, 2023-04-18:
I am attempting to use the ZBar library to decode PDF417 barcodes on the web using Web Assembly. I need help with detecting PDF417. 

I have tested the ZBar library with other types of barcodes, and it works perfectly, so the issue is specific to PDF417 decoding.

Could someone from the community or the ZBar development team help me troubleshoot this issue? Is there a workaround or a fix that I can apply to resolve this problem?

jameshilliard, 2023-04-18:
> Is there a workaround or a fix that I can apply to resolve this problem?

From my understanding pdf417 support is effectively completely broken in zbar and would need significant development work to make functional.

vishaljangid1729, 2023-04-18:
> > Is there a workaround or a fix that I can apply to resolve this problem?
> 
> From my understanding pdf417 support is effectively completely broken in zbar and would need significant development work to make functional.

I am facing this issue in production. I need urgent help on this. Can anyone help me on this please.

haimat, 2023-04-18:
> I am facing this issue in production. I need urgent help on this. Can anyone help me on this please.

I had the same issue, but was not able to solve it via **zbar**. Thus I ended up with [pdf417decoder](https://pypi.org/project/pdf417decoder/).
If you are using Python you could give that library a try. Good luck!

jameshilliard, 2023-04-18:
> I am facing this issue in production. I need urgent help on this. Can anyone help me on this please.

You probably need to look into using something like [ZXing](https://github.com/zxing-cpp/zxing-cpp) for pdf417 decoding at the moment.

> I am attempting to use the ZBar library to decode PDF417 barcodes on the web using Web Assembly.

The ZXing library [supports webassembly](https://github.com/zxing-cpp/zxing-cpp/blob/master/wrappers/wasm/README.md).