mchehab/zbar #21: ZBar is too sensible to image rotation

mchehab, 2019-02-10:


mchehab, 2019-02-10:
Funny enough, for QR, only 4 (integer) rotation angles got it wrong:

```
for i in qr-code.png; do for j in $(seq 1 359); do n=${i/.png/-$j.png}; convert -rotate $j $i ../rotated/$n;done; done
for i in ../rotated/*.png; do if [ "$(zbarimg $i 2>/dev/null)" != 'QR-Code:https://github.com/mchehab/zbar' ]; then echo $i; fi; done
../rotated/qr-code-183.png
../rotated/qr-code-342.png
../rotated/qr-code-72.png
../rotated/qr-code-73.png
```

jameshilliard, 2019-02-10:
QR encoding has pretty solid error correction which allows recovery of the data even if up to 30% of codewords are missing it seems, that's also how QR Code Art is possible.

mchehab, 2019-02-10:
> QR encoding has pretty solid error correction which allows recovery of the data even if up to 30% of codewords are missing it seems, that's also how QR Code Art is possible.

I see. Yet, it sounds funny that there is a small range that are "blind spots" to the detection algorithm. Probably not worth fixing it, but for the 1D barcodes, it would be good if ZBar could do a lot better and read them with higher inclinations.

jasp00, 2019-02-10:
ZBar does not handle any inclination (exception: SQ code), it only processes space/bar pairs horizontally and vertically. In QR code, it detects the corner finders like 1D barcodes.