mchehab/zbar #27: Migrate onto Python 3 and GTK+ 3

hosiet, 2019-02-13:
Python 2 will stop maintenance by 2020. I guess it's reasonable to open an issue to migrate away from python2 (as well as gtk2).

mchehab, 2019-02-13:
> Python 2 will stop maintenance by 2020. I guess it's reasonable to open an issue to migrate away from python2 (as well as gtk2).

Makes sense. 

One note though: there's already an independent ZBar python 2/3 library:
PyZBar: https://pypi.org/project/pyzbar/. I never tested it, but, assuming it works properly, one option would be to just drop python2 support, assuming that python developers would use PyZbar instead.

For Fedora rawhide, we actually had to disable Python 2, because the pygtk2 module is orphaned and will likely be retired from it soon.

Another option would be to merge PyZbar into ZBar (after talking with the original author of it, and provided that we solve any licensing issues - if any), replacing what we have right now. That would very likely break API/ABI.

In any case, I suspect we should solve the Python issue before doing a Gtk+2 migration to Gtk+3 (and to the upcoming Gtk 4 version), due to pygtk2 dependencies.

jameshilliard, 2019-03-04:
> I never tested it, but, assuming it works properly, one option would be to just drop python2 support, assuming that python developers would use PyZbar instead.

Well python2 will probably be supported by some enterprise distros well past 2020(centos 7 is supported until June 30th, 2024 and uses python2), not sure if that's something we care about supporting though. Usually it's feasible to write single source python code that can support both python2 and python3 at the same time as long as python3 only features aren't used. I'll try and look into how difficult that would be if I get some time.

> I never tested it, but, assuming it works properly, one option would be to just drop python2 support, assuming that python developers would use PyZbar instead.

I looked into PyZbar a little while ago and it didn't seem to support a number of features the python2 version bundled with zbar does such as video scanning. Since I needed to integrate zbar video scanning with my application that supports python3 only I ended up creating the dbus API instead.

mchehab, 2019-03-04:
> Well python2 will probably be supported by some enterprise distros well past 2020(centos 7 is supported until June 30th, 2024 and uses python2), not sure if that's something we care about supporting [...]

Enterprise distros have a team of people responsible to ensure long term maintainership. One of their principles is that APIs never change. So, if a package goes to CENTOS, RHEL, SUSE, ... (and even LTS distros like Debian) they won't have any strong reason to remove them. Also, although uncommon, they may eventually remove packages or features along the time (I've seen this on some LTS distros). RHEL for example has the concept of "technology preview" with are experimental features that may eventually be fully supported or be removed on a future update.

In any case, I don't see why we would keep maintaining a python2-only support up to 2024 upstream. This is a distro's maintainers task.

> Usually it's feasible to write single source python code that can support both python2 and python3 at the same time as long as python3 only features aren't used. I'll try and look into how difficult that would be if I get some time.

If this can be done for ZBar, it would be nice. Right now, the Python 2 bindings are written in C, though.  So, in order to use the current code, it will depend on how Python internals changed from version 2 to 3. Another alternative would be to (partially) port the binding code to Python. That could  make easier to support different Python versions.

> > I never tested it, but, assuming it works properly, one option would be to just drop python2 support, assuming that python developers would use PyZbar instead.

> I looked into PyZbar a little while ago and it didn't seem to support a number of features the python2 version bundled with zbar does such as video scanning. Since I needed to integrate zbar video scanning with my application that supports python3 only I ended up creating the dbus API instead.

Ok, if PyZBar doesn't carry the needed functionality, then it makes a lot more sense to change ZBar to work with Python 3.

TheJJ, 2019-05-05:
I've done the C-API port from 2 to 3. Py2 is still supported. Please test!

mchehab, 2019-05-06:
This issue was covering two things: Python 3 and Gtk3.

While we don't address Gtk3, better to keep it opened.

@TheJJ how easy/hard would be to port the stuff under the pygtk dir to be compatible with Gtk3? I suspect that, for Gtk3, it should be using this, right? 
  - https://pygobject.readthedocs.io/en/latest/

TheJJ, 2019-05-06:
Phew, I'm not experienced with Gtk stuff. `pygobject` is the right way though as far as I've heard.
How one does the `zbarpygtk.defs` the modern way, I have no idea.

It seems like the `defs` file is converted to generated C code. With `pygobject`, I don't think that's necessary any more.

Maybe the best approach would be to just write some wrapper code in plain Python and reuse the `zbar` python module directly. What is the `pygtk` module designed/used for anyway?

mchehab, 2019-05-06:
Something like this makes it build:
```
diff --git a/Makefile.am b/Makefile.am
index 44347aaadc17..0247917669fc 100644
--- a/Makefile.am
+++ b/Makefile.am
@@ -34,6 +34,9 @@ pkgconfig_DATA += zbar-gtk.pc
 if HAVE_PYGTK2
 include $(srcdir)/pygtk/Makefile.am.inc
 endif
+if HAVE_PYGOBJECT
+include $(srcdir)/pygtk/Makefile.am.inc
+endif
 endif
 if HAVE_QT
 include $(srcdir)/qt/Makefile.am.inc
diff --git a/configure.ac b/configure.ac
index 696e80b9d02b..76c4b3d27f56 100644
--- a/configure.ac
+++ b/configure.ac
@@ -514,7 +514,6 @@ Install the development package for python-$am_cv_python_version, or configure
    CPPFLAGS="$CPPFLAGS_save"
 
    dnl PyGTK
-   dnl disable pygtk when we're on Python 3
    AS_IF([test "x$with_gtk$PYTHON_VERSION_MAJOR" = "xyes2"],
      [PKG_CHECK_MODULES([PYGTK], [pygtk-2.0], [with_pygtk2=yes], [with_pygtk2=no])
       AC_CHECK_PROGS([PYGTK_CODEGEN], [pygobject-codegen-2.0 pygtk-codegen-2.0 pygtk-codegen], [:])
@@ -524,11 +523,15 @@ Install the development package for python-$am_cv_python_version, or configure
          AS_IF([test -f "$PYGTK_H2DEF"], [], [PYGTK_H2DEF=":"])])
       AS_IF([test "x$PYGTK_DEFS" = "x"],
         [PYGTK_DEFS=`$PKG_CONFIG pygtk-2.0 --variable=defsdir`])
-   ])
+   ],
+   [AS_IF([test "x$with_gtk$PYTHON_VERSION_MAJOR" = "xyes3"],
+     [PKG_CHECK_MODULES([PYGOBJECT], [pygobject-3.0], [with_pygobject=yes], [with_pygobject=no])
+    ])])
 ])
 
 AM_CONDITIONAL([HAVE_PYTHON], [test "x$with_python" = "xyes"])
 AM_CONDITIONAL([HAVE_PYGTK2], [test "x$with_pygtk2" = "xyes"])
+AM_CONDITIONAL([HAVE_PYGOBJECT], [test "x$with_pygobject" = "xyes"])
 
 dnl Qt
 AC_ARG_WITH([qt],
@@ -756,8 +759,11 @@ AS_IF([test "x$have_GM" = "xyes"],
   [echo "        => ImageMagick is preferred, as GraphicsMagick doesn't support https"])
 AS_IF([test "x$with_gtk" != "xyes"],
   [echo "        => the GTK+ widget will *NOT* be built"],
-  [AS_IF([test "x$with_pygtk2" != "xyes"],
-     [echo "        => the Python GTK widget wrapper will *NOT* be built"])])
+   AS_IF([test "x$PYTHON_VERSION_MAJOR" = "x2"],
+   [AS_IF([test "x$with_pygtk2" != "xyes"],
+      [echo "        => the Python GTK widget wrapper will *NOT* be built"])]),
+   [AS_IF([test "x$with_pygobject" != "xyes"],
+      [echo "        => the Python GTK widget wrapper will *NOT* be built"])])
 AS_IF([test "x$with_qt" != "xyes"],
   [echo "        => the Qt widget will *NOT* be built"])
 AS_IF([test "x$with_qt" == "xyes" && test "x$enable_static_qt" == "xyes" ],
diff --git a/pygtk/Makefile.am.inc b/pygtk/Makefile.am.inc
index 433412c8bcb5..f0ba4a9cd8d6 100644
--- a/pygtk/Makefile.am.inc
+++ b/pygtk/Makefile.am.inc
@@ -1,10 +1,10 @@
 pyexec_LTLIBRARIES += pygtk/zbarpygtk.la
 pygtk_zbarpygtk_la_CPPFLAGS = \
-    $(GTK_CFLAGS) $(PYTHON_CFLAGS) $(PYGTK_CFLAGS) $(AM_CPPFLAGS)
+    $(GTK_CFLAGS) $(PYTHON_CFLAGS) $(PYGTK_CFLAGS)$(PYGOBJECT_CFLAGS) $(AM_CPPFLAGS)
 pygtk_zbarpygtk_la_LDFLAGS = -shared -module -avoid-version -export-dynamic \
-    -export-symbols-regex initzbarpygtk
+    -export-symbols-regex '(initzbarpygtk|PyInit_zbarpygtk)'
 pygtk_zbarpygtk_la_LIBADD = \
-    $(PYTHON_LIBS) $(PYGTK_LIBS) gtk/libzbargtk.la $(AM_LIBADD)
+    $(PYTHON_LIBS) $(PYGTK_LIBS) $(PYGOBJECT_LIBS) gtk/libzbargtk.la $(AM_LIBADD)
 
 dist_pygtk_zbarpygtk_la_SOURCES = pygtk/zbarpygtkmodule.c
 nodist_pygtk_zbarpygtk_la_SOURCES = pygtk/zbarpygtk.c
diff --git a/pygtk/zbarpygtkmodule.c b/pygtk/zbarpygtkmodule.c
index 581d362e07b2..da62fa4e0741 100644
--- a/pygtk/zbarpygtkmodule.c
+++ b/pygtk/zbarpygtkmodule.c
@@ -31,16 +31,49 @@
 void zbarpygtk_register_classes(PyObject*);
 extern PyMethodDef zbarpygtk_functions[];
 
+#if PY_MAJOR_VERSION >= 3
+
+struct PyModuleDef zbarpygtk_moduledef = {
+    PyModuleDef_HEAD_INIT,
+    "zbar",
+    NULL,
+    0,
+    zbarpygtk_functions,
+    NULL,
+    NULL,
+    NULL,
+    NULL
+};
+
+PyMODINIT_FUNC
+PyInit_zbarpygtk(void)
+
+#else
+
 DL_EXPORT(void)
 initzbarpygtk(void)
+#endif
 {
-    init_pygobject();
+    PyObject *mod = NULL;
+
+    if (!pygobject_init(-1, -1, -1))
+        goto error;
+
+#if PY_MAJOR_VERSION >= 3
+    mod = PyModule_Create(&zbarpygtk_moduledef);
+#else
+    mod = Py_InitModule("zbarpygtk", zbarpygtk_functions);
+#endif
 
-    PyObject *mod = Py_InitModule("zbarpygtk", zbarpygtk_functions);
     PyObject *dict = PyModule_GetDict(mod);
 
     zbarpygtk_register_classes(dict);
 
     if(PyErr_Occurred())
         Py_FatalError("unable to initialise module zbarpygtk");
+
+error:
+#if PY_MAJOR_VERSION >= 3
+    return mod;
+#endif
 }
```
but running a modified version of test_pygtk.py importing "gi" instead of "pygtk" causes this error:

```
Traceback (most recent call last):
  File "./test/test_pygtk3.py", line 26, in <module>
    import zbarpygtk
ImportError: /usr/local/lib64/python3.7/site-packages/zbarpygtk.so: undefined symbol: _PyGObject_API
```

mchehab, 2019-05-06:
> Phew, I'm not experienced with Gtk stuff. `pygobject` is the right way though as far as I've heard.

Ok. Apparently, it is not too hard to convert zbarpygtk to use it instead of pygtk-2.0 (see above comment).

> How one does the `zbarpygtk.defs` the modern way, I have no idea.
> 
> It seems like the `defs` file is converted to generated C code. With `pygobject`, I don't think that's necessary any more.

Probably we need to do something similar, as zbarpygtk_functions is produced from the generated C file.

> 
> Maybe the best approach would be to just write some wrapper code in plain Python and reuse the `zbar` python module directly. 

Maybe. Yet, it sounds that converting it shouldn't be that hard.

> What is the `pygtk` module designed/used for anyway?

See test/test.pygtk.py. With zbar pygtk module, it will be just like the zbarcam-gtk application, except that it would be written in Python.

mchehab, 2019-05-06:
Ok, pushed an experimental (not working) patch at:

https://github.com/mchehab/zbar/tree/test_python_gtk3

Building with python 3, e. g. with:

```
autoreconf; PYTHON=/usr/bin/python3 ./configure && make && sudo make install
```

Everything builds fine, but python3 crashes with:

```
$ python3 ./test/test_pygtk.py
Fatal Python error: unable to initialise module zbarpygtk
ImportError: cannot import name Widget from gtk

Current thread 0x00007fdc025fd680 (most recent call first):
  File "<frozen importlib._bootstrap>", line 219 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 1043 in create_module
  File "<frozen importlib._bootstrap>", line 583 in module_from_spec
  File "<frozen importlib._bootstrap>", line 670 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 967 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 983 in _find_and_load
  File "./test/test_pygtk.py", line 25 in <module>
Aborted (core dumped)
```

mchehab, 2019-05-06:
> Maybe the best approach would be to just write some wrapper code in plain Python and reuse the `zbar` python module directly. 

It seems that this would be the easiest approach. There aren't many functions there:

```
$ grep define-  pygtk/zbarpygtk.defs
(define-object Gtk
(define-function zbar_gtk_get_type
(define-function zbar_gtk_new
(define-method scan_image
(define-method get_video_device
(define-method set_video_device
(define-method get_video_enabled
(define-method set_video_enabled
(define-method get_video_opened
(define-method request_video_size
(define-function zbar_gtk_image_from_pixbuf
```
I suspect that it shoudn't be hard to use the .defs file to write a wrapper. Just doesn't have any idea about how to do it ;-)

mchehab, 2019-05-06:
After digging into it, it sounds that the right approach would be to use Gobject Introspection approach, with seems to be what pgobject 3.x supports. Did already the port (still, not really tested). The current patches are at:

- https://github.com/mchehab/zbar/commits/test_python_gtk3_v2

Still, the python part is missing. It seems we need to teach pgobject how to add this to gi.repository module.

mchehab, 2019-05-08:
Just pushed the needed patches for zbarcam-gtk and libzbargtk to work with Gtk3. Didn't push yet a final patch that would change configure.ac to use Gtk3 (or to allow selecting the version) yet, due to the following reasons:

1) There is still a bug when using Gtk3: at exit, it produces an error, while trying to unmap a widget. I was unable to fix it so far.

2) For Python 3 to work with Gtk, it needs to use pygobject and Gnome GObject Introspection (GIR). I wrote the patches already but:
- it causes troubles with Debian packaging and Travis CI due to building Gtk on a separate Makefile. Shouldn't be hard to fix, but requires time and patience, as it requires debugging it remotely with Travis CI;
- I was unable to make it work: When the python script tries to load it, it warns that the ZBarGtk widget is smaller than GtkWidget (that's not true). It also is not accepting setting the signals (probably the GIR file is incomplete).

I'm currently busy with something else, so I'm pushing the remaining patches to a separate temporary branch, in case someone has time to work on those:

https://github.com/mchehab/zbar/tree/python3_gir

mchehab, 2019-05-09:
I was able to solve most of the issues with Python3 Gtk3 port:
- Debian build breakage with Travis CI due to splitting the gtk `Makefile.am` file was solved. I already committed such patch on master;
- The ZBar GObject Introspection (GIR) is now working. I converted the `test-pygtk.py` script to work with it. The GIR version is called `test-gi.py`.

It should be noticed that python has a hardcoded path for system libraries and for *.testlib files. So, in order to be able to test it, one has to install zbar at /usr (e. g. uilding it with `PYTHON=python3 ./configure --prefix=/usr`.

Pending action items:

1) Fix the core dump when closing a ZBar Gtk3 window. This is consistent with both `zbarcam-gtk` and `test-gi.py` script:

```
Gtk:ERROR:gtkwidget.c:5531:gtk_widget_unrealize: assertion failed: (!widget->priv->mapped)
```

2) Add `--with-gtk3` and `--with-python3` config options, in order to allow selecting the gtk and python versions to be used;

3) for the python bindings under `/python` dir: add a testing tool (maybe using `unittest`?);

4) Fix the `debian/` ruleset to work with python3 and/or gtk3. I did some puntual changes there just to add the needed dependencies for building gtk3, but the python dependencies are still broken, and it is not creating packages for GIR or python3 stuff.

5) The `test-gi.py` script produces some warnings due to Gtk3 API deprecation. Not a serious issue, but it would be worth fixing it somewhere in the future. We don't need to keep Gtk2 compatibility here (although I think the script could easily work with both Gtk2 and Gtk3 using `gi.repository`, with also seems to exist on python2.

alexmyczko, 2019-05-10:
Will you keep supporting py2 or can I just drop it from the Debian packaging?

mchehab, 2019-05-10:
> Will you keep supporting py2 or can I just drop it from the Debian packaging?

That's an interesting question. ZBar has 3 kinds of Python dependencies:

1. Python bindings (directory python/), with as a package with just the barcode scanner on it. No GUI integration and no webcam support;
2. Python GTK bindings via pygtk, with has a more full-featured ZBar, with GUI and webcam support. It comes with a test script: `test/test_pygtk.py`.
3. Python scripts: `examples/upcrpc.py` and `test/barcodetest.py`.

This PR addresses (1). We still need a small script to test it, in order to be added at the ZBar test logic (with runs on Travis CI).

The patches I applied yesterday addresses (2), by allowing the usage of GObject Introspection (GIR). That's actually more than just supporting Python, as it supports a lot more languages. GIR depends on Gtk3 (with currently has a bug with happens when the main window closes. It also makes sense to have a small script to test it, if it is possible to do that without installing it.

We still need to address (3) in order to not depend on python 2 package anymore. Converting `test/barcodetest.py` seems to be tricky: `2to3` only partially convert it, and the result is not backward compatible, and the result script doesn't run anymore. It seems that there's some sutile difference at `unittest` package with regards to test suites.

That's said, there are lots of changes here. So, for the upcoming version (0.23), I intend to keep support for python 2 bindings and pygtk. It probably makes sense to keep it for a while, at least while it builds on my machines at the latest versions of Fedora and Ubuntu.

From Distro-maintainer PoV, that's a different matter. I maintain ZBar on Fedora. There, we had to drop support for pygtk for Fedora 30 and upper, as some dependencies were planning to be dropped[1].

[1] ZBar pygtk still builds fine on Fedora 30. Looking at the removal patch, the problem was actually due to a dependency that was replaced by another one with does the same thing: `pyobject-codgegen-2.0`. So, we could have kept it.

mchehab, 2019-05-10:
>     3. Python scripts: `examples/upcrpc.py` and `test/barcodetest.py`.

Addressed this one too. All scripts should now run with either python2 or python3 version.

Now, the only pending issue needed to allow a "pure" python3 version is to fix the close window bug at Gtk3 support (plus configure.ac to allow selecting Gtk3 and Debian/Travis CI changes to allow testing it).

mchehab, 2019-05-11:
Finally got the Gtk3 bug addressed.

The only remaining issue is that the test-gi.py script produces some warnings with Gtk 3.22. Not really a problem, as it works properly. Also, it is just a test script, so no real harm done if kept this way (at least while we don't add support for the upcoming Gtk 4). So, I don't intend to work any time soon in it. 

If anyone wants to volunteer  cleanup those deprecated messages, go for it!

In summary, from my side, the migration to python3 and gtk3 is complete.  So, I'm closing this issue.

With this addressed, we have now several new different possibilities:

   - gtk3 or gtk3 or no-gtk;
   - python2, python3 or no-python;
   - gir or no-gir.
   - pygtk2/pygobject2 (only when python2 is selected) or no-pygtk2

For the next version, I should keep support for all the combinations of the above.

I should only drop support for the legacy features when it becomes hard to keep maintaining it.

As gtk4 is planning to be released this year, my wild guess is that distros may drop support for gtk2 soon. So I wouldn't expect keeping support for gtk2 for more than one year or so, but let's see :-)

PR #44 (2019-05-06): Port to python3
I've ported the Python bindings to Python3.

It would be better to use use Cython instead of the raw C-API :)

I'm not sure how the buildsystem should be updated in a sensible way. When Python3 is used, one needs `--without-gtk`.

Fixes #27
closed 2019-05-11 by mchehab