mchehab/zbar #31: Configure error with libv4l enabled

hosiet, 2019-02-14:
When running `./configure`, the following info indicates grammar errors in configure scripts:

```
[...]
checking libv4l2.h usability... yes
checking libv4l2.h presence... yes
checking for libv4l2.h... yes
./configure: line 15166: V4L2_CFLAGS: command not found
./configure: line 15167: V4L2_LIBS: command not found
checking for V4L2... yes
checking for X... libraries , headers 
checking for gethostbyname... yes
checking for connect... yes
checking for remove... yes
checking for shmat... yes
[...]
```

Note there are two "command not found" lines.

mchehab, 2019-02-15:
Tricky... there was a missing [] block when calling PKG_CONFIG_MODULES().
That caused the logic there to be mis-interpreted by GNU m4, producing a bad code. Those lisp-based languages are not the best for humans to code :-)

This patch should fix it: https://github.com/mchehab/zbar/commit/34b26e581dee257124091eea314f9aded06d5936

closed 2019-02-15 by mchehab

hosiet, 2019-02-15:
I guess we need to cherry-pick the fix on master branch as well...

mchehab, 2019-02-16:
> I guess we need to cherry-pick the fix on master branch as well...

True. It sounds I forgot to cherry-pick this one. I double-checked: all other stable fixes were already merged on master.