mchehab/zbar #34: Support multiple QR codes in a single image

jasp00, 2019-02-21:
This is Debian [#869234](https://bugs.debian.org/869234). Unlike #24, this is about many independent QR codes in a single image; I am not sure whether this case should be supported. Report also complains about high CPU time, which I have not experienced because the test image did not produce any symbol.

mchehab, 2019-02-21:
> This is Debian [#869234](https://bugs.debian.org/869234). Unlike #24, this is about many independent QR codes in a single image; I am not sure whether this case should be supported. Report also [...]

Supporting multiple codebars at the same image is a feature supported by ZBar.

jasp00, 2019-02-21:
>Supporting multiple codebars at the same image is a feature supported by ZBar.

Is the [test case](https://bugs.debian.org/cgi-bin/bugreport.cgi?att=2;bug=869234;filename=qrpage.png;msg=5) expected to succeed?

mchehab, 2019-02-21:
Not sure how that image was produced, but something like this should work:

```
FMT="58" # QR Code

echo "Testing QR decoding"
for i in $(seq 100 199); do
        n="$(printf "%08d" $i)"
        zint --small -b$FMT -o out$n.png -d $n
done

montage -mode concatenate -tile 10x  -geometry +5+5  out00*png output.png
rm out00*png 
zbarimg output.png 

```

mchehab, 2019-02-21:
On my tests here, ZBar is working fine decoding multiple QR codes with the above script. There's one issue there, however. If we mix normal and reversed QR codes, it doesn't work properly, as it will decode just non-reversed codes:

```
#!/bin/bash

FMT="58" # QR Code

echo "Testing QR decoding"
for i in $(seq 100 199); do
        n="$(printf "%04d" $i)"
        zint --small -b$FMT -o out$n.png -d $n
done

for i in $(seq 200 299); do
        n="$(printf "%04d" $i)"
        zint --small --bind -r -b$FMT -o out$n.png -d $n
done

montage -mode concatenate -tile 20x -geometry +5+5 -background white out01*.png normal.png
montage -mode concatenate -tile 20x -geometry +5+5 -background black out02*.png reversed.png
montage -mode concatenate -tile 1x -geometry +5 normal.png reversed.png output.png
rm out0*.png 

echo "Normal decoding"
zbarimg normal.png >/dev/null

echo "Reverse decoding"
zbarimg -Stest-inverted reversed.png >/dev/null

echo "Normal + Reverse decoding"
zbarimg -Stest-inverted output.png >/dev/null
```

The output is:

```
$ ./zint_qr_test.sh 
Testing QR decoding
Normal decoding
scanned 100 barcode symbols from 1 images in 0.07 seconds

Reverse decoding
scanned 100 barcode symbols from 1 images in 0.09 seconds

Normal + Reverse decoding
scanned 100 barcode symbols from 1 images in 0.15 seconds
```

That's because, right now, we only do reverse decoding when the normal one fails. Not sure what would be the best way to handle such scenario.

mchehab, 2019-02-21:
As reference, I placed all 3 images at:
https://www.infradead.org/~mchehab/zbar_test_images/

jasp00, 2019-02-21:
> we only do reverse decoding when the normal one fails.

Reverse decoding should be tried on the non-decoded part of the image, which may be implemented in the future.

jasp00, 2019-02-21:
The output is not sorted. Is it fine?

jasp00, 2019-02-21:
I believe the problem is when dealing with smoothed images, which is a limitation of ZBar. You should try decoding these tests.
400 %, cubic interpolation:
[image]

400 %, sinc interpolation:
(It gives an error about DataBar.)
[image]

400 %, no interpolation:
(It shows an extra DataBar symbol.)
[image]

800 %, cubic interpolation:
[image]

mchehab, 2019-02-21:
> > we only do reverse decoding when the normal one fails.
> 
> Reverse decoding should be tried on the non-decoded part of the image, which may be implemented in the future.

Yeah, that makes sense.

> The output is not sorted. Is it fine?

Good question. In thesis, ZBar stores the position where the image was found. We could place it at the output data (it would probably be easier/better to do it at the D-Bus dictionary). This way, if the position becomes relevant, an application could sort it.

mchehab, 2019-02-21:
> I believe the problem is when dealing with smoothed images, which is a limitation of ZBar. You should try decoding these tests.

Yeah, ZBar is not particularly good with smoothed images. It wouldn't be hard to have a filter for interpolated image, but it would still be dependent on a somewhat "perfect" image. With cameras, it would be a lot harder to solve such kind of issues (although an image could be artificially "improved" by adjusting bright/contrast/focus).

It probably makes sense to have some pre-processing blocks to solve such kind of issues.