mchehab/zbar #38: MinGW compatibility

LRN, 2019-03-27:
Here's a few patches that might be useful.

[0001-Include-windows.h-for-vfw.txt](https://github.com/mchehab/zbar/files/3013024/0001-Include-windows.h-for-vfw.txt)
[0002-Specify-correct-path-to-barcode.png.txt](https://github.com/mchehab/zbar/files/3013025/0002-Specify-correct-path-to-barcode.png.txt)
[0003-Check-for-clock_gettime-on-pthread-library-as-well.txt](https://github.com/mchehab/zbar/files/3013026/0003-Check-for-clock_gettime-on-pthread-library-as-well.txt)
[0004-Disable-zbarcam-gtk-on-Windows.txt](https://github.com/mchehab/zbar/files/3013027/0004-Disable-zbarcam-gtk-on-Windows.txt)
[0005-Use-no-undefined-for-libzbargtk.txt](https://github.com/mchehab/zbar/files/3013028/0005-Use-no-undefined-for-libzbargtk.txt)
[0006-Make-zbargtk-build-on-Windows.txt](https://github.com/mchehab/zbar/files/3013029/0006-Make-zbargtk-build-on-Windows.txt)
[0007-Make-use-of-glib-thread-names.txt](https://github.com/mchehab/zbar/files/3013030/0007-Make-use-of-glib-thread-names.txt)
[0008-Correctly-generate-version.xml-and-reldate.xml.txt](https://github.com/mchehab/zbar/files/3013031/0008-Correctly-generate-version.xml-and-reldate.xml.txt)

mchehab, 2019-03-27:
Could you please send those as a pull request? Also, we're using the Kernel DCO, as speficied on:

https://linuxtv.org/wiki/index.php/Development:_Submitting_Patches#Developer.27s_Certificate_of_Origin_1.1

So, could you please add your Signed-off-by: on each patch?

With regards to zbarcam-qt, It should probably be easy to make it to work with Windows. It only includes v4l2 header in order to allow discovering the video devices on Linux. I suspect that Windows APIs have other means to discover devices.

LRN, 2019-04-07:
> Could you please send those as a pull request?

No. That requires forking the repo on github, which i won't do, since github is proprietary.

You can download the patches, remove the diffs (everything starting with the first `diff`), and use the remaining info to re-implement the changes yourself. That should take care of any formalities.

jasp00, 2019-04-07:
>No. That requires forking the repo on github, which i won't do, since github is proprietary.

This makes no sense: you already signed up with GitHub.

LRN, 2019-04-07:
I had a github account way before i started paying attention to proprietary JS. I have to use github to file this issue and communicate with this project (you don't have a mailing list, do you, guys?), but forking is the line i won't cross.

mchehab, 2019-04-07:
If your problem is with forking it from github,  you could send it as a patch series to:
    linux-media@vger.kernel.org
adding the Signed-off-by to each patch. If you do so, please add [PATCH for ZBar] at the subject line, in order to better identify the patchset.
Btw, the same patches are there at LinuxTV.org too (with is, btw, older than this github repository and contains the same stuff):

https://git.linuxtv.org/zbar.git/

mchehab, 2019-04-08:
FYI, I just created a clone of it also at gitlab. While I prefer to have issues and pull requests here (just for the sake of sanity of having everything at the same place), you may either send patches to linux-media@vger.kernel.org or to submit via gitlab. The clone is at:
https://gitlab.com/mchehab_kernel/zbar

LRN, 2019-04-09:
Do i need to send one email per patch, or just attach all of them to one message? If the former, do i need to send git-format'ed patches in the body of the message or (again) as attachments? In-body might be a problem, since i'm not sure my email client won't mangle the diffs.

mchehab, 2019-04-09:
> Do i need to send one email per patch, or just attach all of them to one message? If the former, do i need to send git-format'ed patches in the body of the message or (again) as attachments? In-body might be a problem, since i'm not sure my email client won't mangle the diffs.

You need to send one e-mail per patch, using git-format, as we rely on patchwork to catch the patches. The best way to sent properly formatted git patches is to use `git send-email` (see https://git-scm.com/docs/git-send-email). It uses either the smart host set on your machine or it will use the one specified by this .git/config parameter:

```
[sendemail]
	smtpEncryption = tls
	smtpServer = smtp.gmail.com
	smtpUser = yourname@gmail.com
	smtpServerPort = 587
```

(the above example comes from its man page, and contains the setup for it to use gmail)

mchehab, 2019-04-25:
Patches applied.

closed 2019-04-25 by mchehab

LRN, 2019-04-30:
The patches were merged, i think.