mchehab/zbar #39: [QT5] unsupported request: no input or output formats available

dehy, 2019-04-01:
Hello there,

I'm fairly new to QT5 and I cannot figure why the QZBar class does not want to work. I can successfully scan barcodes with the zbarcam cli tool but cannot use it inside my app.
I do not have X server installed and use the framebuffer to display my app fullscreen. The app works well without the QZBar code.

I'm using v4l2loopback to convert the stream from my camera to a proper format (using gstreamer). I cannot figure out why it cannot find any output format. Can you help me? Thank you!

main.cpp
```c++
#include <QtCore/QtGlobal>
#include <QtWidgets>
#include "mainwindow.h"

int main(int argc, char **argv) { 
    QApplication app (argc, argv); 
    MainWindow w(nullptr);
    w.setWindowFlags(Qt::FramelessWindowHint);
    w.showFullScreen();
    return app.exec(); 
}
```

mainwindow.cpp
```c++
#include "mainwindow.h"
#include "zbar/QZBar.h"

using namespace zbar;

MainWindow::MainWindow(QWidget *parent) : QMainWindow(parent)
{
    zbar_set_verbosity(32);
    QZBar *bcr = new QZBar(this, 1);
    bcr->setVideoDevice("/dev/video0");
    bcr->setVideoEnabled(true);
    setCentralWidget(bcr);
}
```

output log
```
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QFSFileEngine::open: No file name specified
QStandardPaths: wrong ownership on runtime directory /tmp, 0 instead of 1000
Unable to query physical screen size, defaulting to 100 dpi.
To override, set QT_QPA_EGLFS_PHYSICAL_WIDTH and QT_QPA_EGLFS_PHYSICAL_HEIGHT (in millimeters).
_zbar_video_open: opened camera device /dev/video0 (fd=20)
_zbar_v4l2_probe: Dummy video device (0x0000) on platform:v4l2loopback-000 driver v4l2 loopback (version 4.4.49)
_zbar_v4l2_probe:     capabilities: CAPTURE READWRITE STREAMING
ERROR: zbar video in v4l2_reset_crop():
    system error: querying crop support (VIDIOC_CROPCAP): Inappropriate ioctl for device (25)
v4l2_probe_formats: current format: YU12(32315559) 1280 x 720 (line=0x500 size=0x151800)
v4l2_probe_formats: setting requested size: 40960 x 30720
v4l2_probe_formats: final format: YU12(32315559) 1280 x 720 (line=0x500 size=0x151800)
_zbar_v4l2_probe: using I/O mode: MMAP
ERROR: zbar video in zbar_negotiate_format():
    unsupported request: no input or output formats available
ERROR: ERROR: zbar video in zbar_negotiate_format():
    unsupported request: no input or output formats available
```

dehy, 2019-04-07:
Ok, found that I need X11 running for QZBar. I was only working with qws

closed 2019-04-07 by dehy

mchehab, 2019-04-26:
Yeah, QZBar currently assumes X (and some work has been doing for it to also suport Windows by another developer).

I suspect it shouldn't be hard to make it work headless, but it would require some changes.  Not sure if it is worth, as, if you don't have X, you could simply use the C library, with should provide you with the same functionality.