mchehab/zbar #40: Headless Zbarcam with Webcam throws errors

jshaker000, 2019-04-25:
Hi, I am running on a Macbook Air and have been passing my webcam via VirtualBox to a VM, a headless Arch Linux machine running 
zbarcam --healdless -Sdisable -Sqrcode.enable

It will usually run fine for 10-15 minutes, and then will stop scanning and then will make an endless loop of printing the following. It will also cease to scan correctly:

"Not a JPEG file: starts with 0xff 0xd9"

Running zbar not in headless mode doesnt seem to produce this problem.  
But, this required me to install a dummy X driver and a lot of other configuration I would rather not have to do.  (launch X just for zbar, then appropriately kill it)

Let me know if you need any more information about my system.

mchehab, 2019-04-26:
Hmm... we don't have a --headless option at zbarcam on our git tree at its master branch.

Perhaps you're using some ZBar fork with added such option.

mchehab, 2019-04-26:
There is a package for Arch Linux called zbar-headless:
    https://aur.archlinux.org/packages/zbar-headless/
But all it seems to be doing is to use the plain old ZBar 0.10 configured with:

./configure --prefix=/usr --without-qt --without-gtk --disable-video CFLAGS=-DNDEBUG

mchehab, 2019-04-26:
I did a build here. I'm running ```$ ./zbarcam/zbarcam --nodisplay``` for a quite a while with upstream version (0.22.1) reading some random QR codes I'm getting from the net. No problem so far.

mchehab, 2019-04-26:
I'm using this image for the test:
https://i.gifer.com/origin/17/17f1308347fd23ab0cd657b21839b352_w200.gif
It is an animated gif with lots of different QR codes inside it.

jshaker000, 2019-04-26:
I apologize, I using this package:
https://www.archlinux.org/packages/community/x86_64/zbar/

And am using --nodisplay, not headless

It will scan fine and then I get the 
"Not a JPEG file: starts with 0xff 0xd9"
loop. Have you ever seen that as an output of zbar, or could it be some of its dependancies?
I am not also getting it occassionally when I use Zbar without --nodisplay, but it is much rarer in that case

mchehab, 2019-04-26:
> I'm using this image for the test:
> https://i.gifer.com/origin/17/17f1308347fd23ab0cd657b21839b352_w200.gif
> It is an animated gif with lots of different QR codes inside it.

It is running for about 3 hours here. No problems so far.

> I apologize, I using this package:
> https://www.archlinux.org/packages/community/x86_64/zbar/

Ok.

> And am using --nodisplay, not headless

Ah! Makes a lot more sense now :-)

> It will scan fine and then I get the
> "Not a JPEG file: starts with 0xff 0xd9"
> loop. Have you ever seen that as an output of zbar, or could it be some of its dependancies?

Didn't find anything inside ZBar that would be generating it. What I suspect is that this could be a problem with your camera or with the machine where you're running it. Are you using an ARM machine? There are several USB stacks for some ARM SoC that have a broken support for ISOC transfers (with is the type of the transfer used by most cameras). After some time (~15 mins on my personal experiences), it starts dropping packages, causing corruption at the images.
 
> I am not also getting it occassionally when I use Zbar without --nodisplay, but it is much rarer in that case

Perhaps with Display it is reducing the frame rate, as it will take some time to show the frames, with may make a little easier for the USB host controller to sustain the high USB ISOC traffic.

jshaker000, 2019-04-26:
I  will try running the git version itself instead of the package and compare.  
I am using an x86_64 machine and passing my camera via VirtualBox to the VM.   
Maybe the framerate is an issue, would there be a flag in v4l or zbar to reduce the sampling rate? I know ffmpeg has options to do this also.  

In the mean time, I'll try running both the raw git vs the arch package, and also try running the package on bare metal and see if performance improves

mchehab, 2019-04-26:
> I will try running the git version itself instead of the package and compare.
> I am using an x86_64 machine and passing my camera via VirtualBox to the VM.

Hmm... sounds a tricky scenario. Could you do some tests of running
zbar at bare metal? It could be a problem at the VirtualBox USB
emulation layer. Usually, passing high traffic ISOC USB 2.0 packets
via VM host emulators don't end too well.

> Maybe the framerate is an issue, would there be a flag in v4l or zbar to reduce the sampling rate? I know ffmpeg has options to do this so I was wondering if zbar could access similar flags.

ZBar has limited support. for this to work we would need to implement
VIDIOC_S_PARAM ioctl. I guess we currently don't implement it.

This is somewhat tricky, as the camera driver needs to be able to handle
it. Not all camera drivers know how to configure the USB framing rate. Some fallback code is required.

So, right now, the only way to reduce the USB bitrate is to reduce the
resolution of the camera.

> 
> In the mean time, I'll try running both the raw git vs the arch package, and also try running the package on bare metal and see if performance improves

Yeah, testing it at bare metal may shed some light.

jshaker000, 2019-04-27:
On bare metal it works fine,  
In the VM I have been playing with reducing resolution to decrease the data rate and its  tighttrope between scanning easily and not crashing.

Would it be possible for Zbar to detect it is having this problem and crash or reset itself rather than printing an error loop? I still would like to use Zbar headless in a VM if possible.

mchehab, 2019-04-27:
> On bare metal it works fine,

Ok, so it is a VM broken driver. 

> In the VM I have been playing with reducing resolution to decrease the data rate and its tighttrope between scanning easily and not crashing.

Hmm.. why don't you try running this into a container? It could be a little tricky to set it up (as you would need to map the /dev/video* into it), but it could solve your issue.

Another alternative could be to use USB pass-through with KVM. I suspect this could give a better performance.

> Would it be possible for Zbar to detect it is having this problem and crash or reset itself rather than printing an error loop? I still would like to use Zbar headless in a VM if possible.

The thing is that it is not trivial to detect it. See, the frame drops happen at a "emulated-hardware" level. Some USB frames are dropped, not generating interruptions to the driver. For the driver, the net result is that it will take a longer
time to fill a video buffer. It should still pass those frames to ZBar.  So, ZBar can't tell if there was frame losses or if it is just a slow camera. The maximum we could do would be to setup a timeout (what would be a reasonable timeout?) and quit if no frames appear after such time.

jshaker000, 2019-04-27:
I can try a container or KVM, although one of the points of my application is to ship it as a small VM so that it is "cross platform" and can be used on any OS by begin a VM with a small linux image on it.  

Would there be a way to tell if the error messages I get come from Zbar or from the emulated hardware? Also does it make sense that once I see the error once it would be in an infinite loop? In that case I guess what happens is that the emulated hardware gets permanently behind and the frames fall out of sync? Would that be accurate.
I almost question if it would be easier to take pictures manually and then use zbarimg ...

Zbar has a fixed sampling rate though, yes? I run it without --nodisplay and it takes only a few static images. Could that be slowed down and controlled or could a timeout be an optional CLI parameter, after which it could crash itself. I could detect this with pgrep or something else and relaunch it.  

Im more or less thinking out loud here.

I appreciate your help in tracking this down and exploring solutions

mchehab, 2019-04-29:
> I can try a container or KVM, although one of the points of my application is to ship it as a small VM so that it is "cross platform" and can be used on any OS by begin a VM with a small linux image on it.

Not sure if this would work on "any OS", as it would depend on how good (or bad) would be the OS-specific driver that would emulate the hardware.

> Would there be a way to tell if the error messages I get come from Zbar or from the emulated hardware? Also does it make sense that once I see the error once it would be in an infinite loop? In that [...]

Let me try to answer that, but you should notice that I'm not the original ZBar developer. So, my answer may be incomplete. ZBar has some layers internally, as it can work even on Windows, with a different driver. So, internally, it "hides" the V4L2 code (see zbar/video/ - where zbar's logic for capturing via V4L2 is). The stuff there is responsible to talk with the Kernel driver, with, on a VM, it will end by talking with a VM internal driver with will emulate the hardware (or expose it via an USB pass-through driver), with may not support USB ISOC frames well. 

If the VM driver is bad, it will keep receiving frames, but data will be lost inside the VM driver.

Once a video frame is received from the Kernel hardware driver, it has to be decoded. The specific logic depends on what format the camera provides the image (so, basically, it depends on the camera - or on the "virtualized camera", depending on how VM handles cameras), and if ZBar itself knows how to decode it. If ZBar knows the format, it will handle internally, at zbar/convert.c. Otherwise, it will let libv4l decode it. In the specific case of JPEG, I suspect that it is passing the received data to libjpeg. I suspect that the messages you're seeing are actually coming from libjpeg.

Didn't check the specifics, but it could be possible to add some code to check if libjpeg had issues decoding the frames, adding some logic there to, for example, abort ZBar, if there are more than a certain number of errors per a time period, but this will work only if someone has a camera with outputs JPEG frames. It could be more generic, but, then you would need to add the same kind of check to all stuff under convert.c, with would be a lot of work.

> I almost question if it would be easier to take pictures manually and then use zbarimg ...

You'll have the same problem: if the VM camera driver is broken, you'll still have the same issues, as this is not a ZBar issue, but an issue at VirtualBox.

> Zbar has a fixed sampling rate though, yes? 

No.
(I'm assuming that you're talking about the frame rate. The sampling rate depends only on the video resolution, and, on most cameras, you can't even change it)

Zbar just lets the camera decide the frame rate. 

Depending on the camera, the frame rate can be fixed or adjustable by the camera hardware, depending on the lightning conditions. Those will automatically reduce the frame rate in the dark, in order to increase the exposure time.

By implementing VIDIOC_S_PARM at ZBar, it would be possible to ask the camera driver (if supported) to slow down the frame capture rate, in order to ensure that it won't capture more than a certain number of frames per second. The camera can still slow down the capture even further, due to bad light conditions. 

Not all V4L2 camera drivers support VIDIOC_S_PARM. So, the logic should check if it is supported. Adding support for this ioctl at zbar/video/v4l2.c is simple. But then you need to add support for it at the ZBar video abstraction layer and implement the max FPS logic on all library layers, going up to the applications (e. g. at least up to zbarcam and zbarcam-qt), in order to allow userspace to select the frame rate, if the camera driver supports, hiding the option (at GUI apps) if this ioctl is not supported.

If you want to see how a change at v4l2.c reflects at ZBar, see, for example
0d01d9800c8c1b68493dfd23fbe718ee63db5cf3. This basically adds support for more V4L2 control types (so the feature itself is not related to S_PARM), but it shows that any change there that would reflect at zbar apps need to touch several files, in order to first add the feature at the libraries, and then add it to the apps:
``` include/zbar.h         |   24 +++++++--
 include/zbar/QZBar.h   |   10 +++
 qt/QZBar.cpp           |    6 +-
 qt/QZBarThread.cpp     |    4 +
 qt/QZBarThread.h       |    2 
 zbar/video.c           |   11 +---
 zbar/video.h           |    2 
 zbar/video/v4l2.c      |  123 +++++++++++++++++++++++++++++++++++--------------
 zbarcam/zbarcam-qt.cpp |   39 +++++++++++----
```
(this specific feature won't need to touch zbarcam.c - a S_PARM would need to touch it as well)

> I run it without --nodisplay and it takes only a few static images. Could that be slowed down and controlled or could a timeout be an optional CLI parameter, after which it could crash itself. I could detect this with pgrep or something else and relaunch it.

The problem is the volume of frames received by the VM's USB layer. This is controlled only by the Kernel driver, which actually depends on the camera known features. So, the only way is to control it is by implementing VIDIOC_S_PARAM.

If you're willing to do that, feel free to submit us patches.
> 
> Im more or less thinking out loud here.
> 
> I appreciate your help in tracking this down and exploring solutions

mchehab, 2020-04-09:
As this doesn't seem to be a ZBar issue, but instead some weird behavior with USB on VMs, I'm closing this issue.

closed 2020-04-09 by mchehab