mchehab/zbar #45: ZBar does not handle any inclination (exception: SQ code), it only processes space/bar pairs horizontally and vertically. In QR code, it detects the corner finders like 1D barcodes.

friendlzx, 2019-05-08:
ZBar does not handle any inclination (exception: SQ code), it only processes space/bar pairs horizontally and vertically. In QR code, it detects the corner finders like 1D barcodes.

_Originally posted by @jasp00 in https://github.com/mchehab/zbar/issues/21#issuecomment-462173736_

friendlzx, 2019-05-08:
hi，can you give me a SQ code IMAGE for test？

mchehab, 2019-05-08:
> ZBar does not handle any inclination (exception: SQ code), it only processes space/bar pairs horizontally and vertically. In QR code, it detects the corner finders like 1D barcodes.
> 
> _Originally posted by @jasp00 in [#21 (comment)](https://github.com/mchehab/zbar/issues/21#issuecomment-462173736)_

This is duplicated of #21. Better to close this issue keeping everything related to inclination at just one issue.

> hi，can you give me a SQ code IMAGE for test？

See the examples directory. There are two SQ images there: `sqcode1-generated.png` and `sqcode1-scanned.png`.

closed 2019-05-08 by mchehab