mchehab/zbar #46: debian/

alexmyczko, 2019-05-08:
could you remove that from the tarball releases? it'll make it easier to maintain for debian maintainers. (or just put it in a debian branch if you want to keep it)

alexmyczko, 2019-05-08:
I was planning to ITA it? https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=898078

mchehab, 2019-05-08:
I'll apply this patch (e210a05f3c56d5fe45f289c4f670a76d771b8193). We can always revert it later, if needed.

mchehab, 2019-05-08:
I'll apply the above patch: e210a05f3c56d5fe45f289c4f670a76d771b8193
We can always remove it needed.

mchehab, 2019-05-08:
Change applied.

hosiet, 2019-05-08:
@alexmyczko It's good to know you are going to adopt this package; please let me know if you need information about my previous QA uploads for zbar in Debian.

alexmyczko, 2019-05-09:
@hosiet dunno I prefer irc, am tarzeau there...

alexmyczko, 2019-05-10:
@hosiet not sure if i should just create python3 packages, and drop py2, or add py3 and keep py2, depends on https://github.com/mchehab/zbar/issues/27 i guess

hosiet, 2019-05-10:
@alexmyczko For official Debian packages, I guess we can wait till upstream settles down to a stable state for python3 support. Several uploads onto Experimental before uploading onto Debian Sid should be needed.

mchehab, 2019-05-10:
@alexmyczko I'm uploading upstream[1] the needed bits for selecting python3 and/or gtk3 support. So, feel free to use what we have there as the basis for the Debian package at Experimental.

[1] as soon as the test branch pass on Travis CI - The Debian building logic does some very weird things, building with and without python, with exercises some uncommon `./configure` settings.

The way it is, without any parameter, `./configure` defaults to gtk2 and python2. 
So, it shouldn't require any changes under `debian/` and `travis/` directories for it to pass on Travis CI.

If you want to build it with python3 and gtk3 support instead, you can use:

```
./configure --with-gtk=auto --with-python=auto
```

And be sure that gtk3 and python3 packages will be there at the build machine. For a pure gtk3/python3 build, the needed packages are likely the ones on this patch: cb6222031887f10a370e9bb21e420ad123267a08

As pointed before, there's still a bug with Gtk3. My intention is to work there, avoiding touching at Travis CI, as whatever I would do there would be duplicated work to what you'll be doing in order to build it for Debian Sid.

Please, once you change the `debian/` rules for it to work with Gtk3/Python3, submit to us as a patch. At least here, I want to keep building for Gtk2/Python2 (at least for one version), just to get reports from Travis CI if we ever break something there.

mchehab, 2019-05-11:
As the migration is done, feel free to test and report any issues. 

@alexmyczko @hosiet @LRN: I probably won't touch at the Debian build system for now. It took a lot more time than I originally expected to finish the Gtk3/Python3 migration.

So, my plan is to let you people with more experience doing the Debian packaging stuff. Once you have it done, feel free to submit patches to add/modify the Debian build logic used to feed Travis CI.

---

With regards to Fedora, I'm planning to upgrade the rawhide package to use Gtk3 and will probably add a `zbar-gir` package as well.

As nobody opened any BZ yet asking for _The return of `ZBar-python`_ -  which was removed on Fedora 30 - I'll probably wait for a while before re-adding either pygtk and/or python bindings.

mchehab, 2019-05-11:
> With regards to Fedora, I'm planning to upgrade the rawhide package to use Gtk3 and will probably add a `zbar-gir` package as well.

It ended to be a lot harder than I expected. Producing GIR files under a chroot environment with the built on an out of tree dir had its challenges. Anyway, just finished building and testing. As the building scripts could be useful for Debian and other packagers, I'm placing the link here with the Fedora building rules:

- https://linuxtv.org/downloads/zbar/zbar.spec

I had to add extra patches for it to happen, already merged upstream.

With that, I was able to generate packages with GIR enabled for both Fedora 30 and Rawhide. I ended by creating a COPR repository with the Fedora 30 build with ZBar test versions: 
- https://copr.fedorainfracloud.org/coprs/mchehab/ZBar/

mchehab, 2019-05-12:
> @alexmyczko @hosiet @LRN: I probably won't touch at the Debian build system for now. It took a lot more time than I originally expected to finish the Gtk3/Python3 migration.

I ended by adding a new test at Travis CI doing the build with Gtk3/Python3/GIR:
   - https://travis-ci.org/mchehab/zbar

With that, we now have 4 targets:

1. xenial-gtk2 - builds with Gtk2, Python2 and pygtk2/pygobject2;
2. xenial-gtk3- builds with Gtk3, Python3 and GIR (pyobject3);
3. debian-sid with gcc;
4. debian-sid with clang.

The Debian sid targets are there mainly to smooth upstreaming ZBar on Debian, as it runs the stuff there at debian/ directory with (approximately) matches the upstream one. In order for it to be useful, whatever changes you do upstream (except for trivial ones like comments and changelogs) should also be reflected there. So, feel free to periodically send patches in order to merge back from Debian packaging system - or if you don't find it useful anymore - we can simply drop it and add a xenial target to test clang build.

It is really your call.

mchehab, 2019-05-14:
@alexmyczko @hosiet @LRN: please let me know if you find any issues with it, as otherwise I'll likely release ZBar version 0.23 at the beginning of the next week.

alexmyczko, 2019-05-15:
It looks great from reading, I'll work on it at http://phd-sid.ethz.ch/debian/zbar/ so waiting for the 0.23 tarball...

mchehab, 2019-05-23:
> It looks great from reading, I'll work on it at http://phd-sid.ethz.ch/debian/zbar/ so waiting for the 0.23 tarball...

Released the 0.23 version today. Tarball available at:

- https://linuxtv.org/downloads/zbar/

Also released a COPR package for Fedora 30 today (and a new build for rawhide).

alexmyczko, 2019-07-05:
https://ftp-master.debian.org/deferred.html it's still there for a few more hours, and then gets into debian sid... sorry it took so long. 0.23 will then also appear here: https://buildd.debian.org/status/package.php?p=zbar
and i'm dropping python2 completely, and providing python3 packages.

closed 2019-07-23 by alexmyczko