mchehab/zbar #47: Broken build, zbar.h: No such file or directory

EchterAgo, 2019-05-18:
This is caused by commit https://github.com/mchehab/zbar/commit/d09afa67a6df7c2ecbaab877d0630201799cd425

First I had a problem where it told me `ZBAR_SQCODE` is undefined, which I traced back to it using my system `zbar.h`. After I uninstalled my system zbar it complained about not finding `zbar.h`.

```
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I../../zbar -I../include -I../../zbar -I../include -mthreads -I/mingw32/include -Wall -Wno-parentheses -g -O2 -MT libzbar_la-symbol.lo -MD -MP -MF .deps/libzbar_la-symbol.Tpo -c ../../zbar/symbol.c  -DDLL_EXPORT -DPIC -o .libs/libzbar_la-symbol.o
../../zbar/config.c:31:10: fatal error: zbar.h: No such file or directory
 #include <zbar.h>
          ^~~~~~~~
compilation terminated.
```

EchterAgo, 2019-05-18:
If I add `-I$(srcdir)/../include` to `libzbar_la_CPPFLAGS` it builds again.

EchterAgo, 2019-05-18:
Also caused by the same commit:

```
make[2]: *** No rule to make target 'libzbar-rc.lo', needed by 'libzbar.la'.  Stop.
```

EchterAgo, 2019-05-18:
Ok, repeating the .rc rules in `zbar/Makefile.am` fixes that as well:

```
%-rc.o: %.rc
	$(RC) $(DEFS) $(DEFAULT_INCLUDES) $(INCLUDES) \
	$(AM_CPPFLAGS) $(CPPFLAGS) -o $@ $<
%-rc.lo: %.rc
	$(LIBTOOL) --tag=RC $(AM_LIBTOOLFLAGS) $(LIBTOOLFLAGS) \
	--mode=compile $(RC) $(DEFS) $(DEFAULT_INCLUDES) $(INCLUDES) \
	$(AM_CPPFLAGS) $(CPPFLAGS) -o $@ $<
```

Maybe this should be in an included file?

EchterAgo, 2019-05-18:
See #48

mchehab, 2019-05-18:
Thanks for catching it. I was thinking: would it be possible to add some Travis CI build(s) for Windows? It seems that there is some support for it there:

https://blog.travis-ci.com/2018-10-11-windows-early-release

EchterAgo, 2019-05-18:
That is definitely a good idea. I'm already doing reproducible build for Windows in Docker so that should be no problem. I also saw that the zbarw repo has some Travis setup, maybe we can use that.

Regarding this change, it is likely that the GTK part needs a similar change, but I haven't tested this yet as I don't build GTK. I'm not even sure if adding the include dir is the right way to do this, maybe a relative include would be preferable?

jameshilliard, 2019-05-18:
I think using [appveyor](https://www.appveyor.com/) may be the best option for windows.

mchehab, 2019-05-18:
> That is definitely a good idea. I'm already doing reproducible build for Windows in Docker so that should be no problem. I also saw that the zbarw repo has some Travis setup, maybe we can use that.

Interesting... I'm doing some tests on my test branch in order to see if it can easily be enabled. The zbarw repository uses docker for the build, probably because it was written before Windows image on Travis CI.

> Regarding this change, it is likely that the GTK part needs a similar change, but I haven't tested this yet as I don't build GTK. I'm not even sure if adding the include dir is the right way to do this, maybe a relative include would be preferable?

The problem with relative includes is that they fail with some build options used by Linux distros. Most distros create a separate directory for the build and then do an out of the tree build (e. g. the source and the build dirs are different, and the build doesn't have sources). That makes a lot harder to make a building logic that works.

> I think using [appveyor](https://www.appveyor.com/) may be the best option for windows.

If we can't make it work on Travis, we can try using it.

jameshilliard, 2019-05-18:
> If we can't make it work on Travis, we can try using it.

A number of projects use both CI systems at the same time.

mchehab, 2019-05-18:
> > If we can't make it work on Travis, we can try using it.
> 
> A number of projects use both CI systems at the same time.

Ok, but it makes the environment a little more complex, IMO. 

Anyway, the main issue (for me) is to know exactly what commands are used to build it on windows. Being either appveyor or TravisCi won't change that.

The `.travis.yml` I wrote is already able of building for Windows, and even install packages like `cygwin` and `msys2` with chocolatey (see the several tests I made: https://travis-ci.org/mchehab/zbar/builds.

 The issue is that it needs `autoconf` or `autoreconf` before running ./configure. Installing `autotools` it is not hard. The hardest part is to identify where it is installed within the windows image, as it is not at the system's path, and windows doesn't have anything similar to the `tree` comand, AFAIKT.

mchehab, 2019-05-20:
> That is definitely a good idea. I'm already doing reproducible build for Windows in Docker so that should be no problem. I also saw that the zbarw repo has some Travis setup, maybe we can use that.

From what I saw there, it is actually doing a cross build on a Linux VM, and not really a true Windows build.

It took me a while, but the Windows build is starting to work:

https://travis-ci.org/mchehab/zbar/builds/534966662

The are some issues there, though:

1) I had to add `--disable-dependency-tracking` to `./configure` command, as otherwise it complains. It sounds to be that there are some dependencies broken there. Maybe it is related to this comment at `zbar/Makefile.am`:

```
# FIXME broken
libzbar_la_LIBADD += libzbar-rc.lo
```

2) The build script fails either at `make` or `make install` with:
```
/usr/bin/sh: C:/program: No such file or directory
```

It seems that it is calling something like `C:/program foo` without enclosing it on quotation marks.

mchehab, 2019-05-20:
> It seems that it is calling something like `C:/program foo` without enclosing it on quotation marks.

The problem is that `make` calls the shell with:

```
C:/program files/git/usr/bin/sh.exe ../libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I. -I../include  -I. -I./../include -I./include -mthreads -DCOBJMACROS   -Wall -Wno-parentheses -g -O2 -c -o libzbar_la-config.lo `test -f 'config.c' || echo './'`config.c
```

inside a bash script, causing it to do the wrong thing.

mchehab, 2019-05-21:
> I think using [appveyor](https://www.appveyor.com/) may be the best option for windows.

Tried it, but didn't work. It didn't even let adding ZBar to the list of projects.

Anyway, Windows build on Travis CI is now working:

- https://travis-ci.org/mchehab/zbar/jobs/535331139

There is/was a minor issue: the Windows build is/was not ending automatically. I removed the `after_script` hook. Maybe it will work now.

@EchterAgo it would be nice to try enabling more things at the Windows build. Right now, it is just like what's enabled on Zbarw: DirectShow:

```
DirectShow driver      --with-directshow=yes
pthreads               --enable-pthread=yes
doc                    --enable-doc=yes
v4l                    --enable-video=yes
jpeg                   --with-jpeg=no
Python                 --with-python=no    python
GTK                    --with-gtk=no          Gtk
GObject introspection  --with-gir=no
Qt                     --with-qt=no            Qt
Java                   --with-java=no
Dbus                   --with-dbus=no
ImageMagick            --with-imagemagick=no
Enabled codes:         ean databar code128 code93 code39 codabar i25 qrcode sqcode
Disabled codes:        pdf417
        => libv4l will *NOT* be used
        => JPEG image conversions will *NOT* be supported
        => the zbarimg file scanner will *NOT* be built
        => GTK support will *NOT* be built
        => the Qt widget will *NOT* be built
        => the Java interface will *NOT* be built
        => the Java unit test will *NOT* be enabled
```

Everything else is disabled. The libv4l warning makes no sense on Windows. I'll write a followup patch disabling it. It probably makes sense to have a similar target using, instead, VfW, assuming that it builds at the Windows VM on Travis CI infra.

mchehab, 2019-05-21:
@EchterAgo hard to tell without a previous build, but it seems that adding support for DirectShow broke support for VfW:
- https://travis-ci.org/mchehab/zbar/jobs/535367234

```
zbarcam/zbarcam.o: In function `main':
C:\Users\travis\build\mchehab\zbar/zbarcam/zbarcam.c:147: undefined reference to `__imp_CoInitializeEx'
```

On a real quick search, it seems that this is related to DirectShow:

- https://stackoverflow.com/questions/11566007/directshow-compile-issue

mchehab, 2019-05-21:
> On a real quick search, it seems that this is related to DirectShow:
> 
>     * https://stackoverflow.com/questions/11566007/directshow-compile-issue

Something like this should fix it: 45044740b91de96a01f4c78c9c15836fbf8667db. Testing the build again:

- https://travis-ci.org/mchehab/zbar/builds/535382086

EchterAgo, 2019-05-22:
I think we should only call CoInitializeEx when we actually build with DirectShow. I had a look at your patch and it looks good. I'll give it a try in my setup.

EchterAgo, 2019-05-22:
This time `zbarcam` can not find `zbar.h`

```
make[2]: Entering directory '/home/axel/src/zbar/dist-i686'
  CC       zbarcam/zbarcam-zbarcam.o
../zbarcam/zbarcam.c:35:10: fatal error: zbar.h: No such file or directory
 #include <zbar.h>
          ^~~~~~~~
```

Caused by https://github.com/mchehab/zbar/commit/45044740b91de96a01f4c78c9c15836fbf8667db

EchterAgo, 2019-05-22:
Setting `zbarcam_zbarcam_CPPFLAGS = $(AM_CPPFLAGS)` seems to resolve this. Compilation command changes from
```
gcc -DHAVE_CONFIG_H -I. -I.. -I./include  -DDIRECTSHOW -I/mingw32/include -Wall -Wno-parentheses -g -O2 -MT zbarcam/zbarcam-zbarcam.o -MD -MP -MF zbarcam/.deps/zbarcam-zbarcam.Tpo -c -o zbarcam/zbarcam-zbarcam.o `test -f 'zbarcam/zbarcam.c' || echo '../'`zbarcam/zbarcam.c
```
to
```
gcc -DHAVE_CONFIG_H -I. -I.. -I./include  -I../include -DDIRECTSHOW -I/mingw32/include -Wall -Wno-parentheses -g -O2 -MT zbarcam/zbarcam-zbarcam.o -MD -MP -MF zbarcam/.deps/zbarcam-zbarcam.Tpo -c -o zbarcam/zbarcam-zbarcam.o `test -f 'zbarcam/zbarcam.c' || echo '../'`zbarcam/zbarcam.c
```
But it doesn't seem quite right considering we already include `-I./include`. I'm not great at autotools so this likely is the wrong approach.

EchterAgo, 2019-05-22:
Looks like the Travis build has the same issue: https://travis-ci.org/mchehab/zbar/jobs/535619168

mchehab, 2019-05-22:
> zbarcam_zbarcam_CPPFLAGS = $(AM_CPPFLAGS)

> Setting `zbarcam_zbarcam_CPPFLAGS = $(AM_CPPFLAGS)` seems to resolve this. Compilation command changes from
>
> ```
> gcc -DHAVE_CONFIG_H -I. -I.. -I./include  -DDIRECTSHOW -I/mingw32/include -Wall [...]

That's probably because the main Makefile has:

```
Makefile.am:    $(AM_CPPFLAGS) $(CPPFLAGS) -o $@ $<
```

autotools is very complex. To make things worse, each distro does the build on different ways, passing sourcedir/builddir/... differently. Nowadays, lots of projects are using cmake or mason instead, as those are aimed to make the build simpler. Maybe some day we end converting it, but I suspect that it would be easier to just add a new TravisCI target than porting it.

So, it would be great if you could send us a patch for .travis.yml and travis/* adding the exact cross-compilation environment you're using.

In the mean time, I added a patch based on your suggestion. I suspect that we could do some cleanup there on all CPPFLAGS. Anyway, with your suggestion, it is building on the current Travis CI scenarios:

https://travis-ci.org/mchehab/zbar/builds/535689412

mchehab, 2019-05-22:
I suspect that this will fix the issue:

```
[PATCH] makefiles: fix CPPFLAGS

a recent change in order to support native Windows build caused
ZBar to break on some builds due to the lack of the proper
CPPFLAGS. While debugging it, I noticed that the definition at
configure.ac assumes a single Makefile.am file. Because of that,
it was adding a "./include" on some targets at gtk and zbar
directories.

diff --git a/configure.ac b/configure.ac
index 7a4b1fe8400a..1b4358530440 100644
--- a/configure.ac
+++ b/configure.ac
@@ -53,7 +53,7 @@ AC_DEFINE_UNQUOTED([LIB_VERSION_REVISION],
   [[`echo "$LIB_VERSION" | sed -e 's/^[^:]*:\([^:]*\):.*$/\1/'`]],
   [Library revision])
 
-AM_CPPFLAGS="-I$srcdir/include"
+AM_CPPFLAGS="-I$top_srcdir/include"
 AM_CFLAGS="-Wall -Wno-parentheses"
 AM_CXXFLAGS="$AM_CFLAGS"
 AC_SUBST([AM_CPPFLAGS])
diff --git a/gtk/Makefile.am b/gtk/Makefile.am
index 93b1b3cbda85..874468246a6b 100644
--- a/gtk/Makefile.am
+++ b/gtk/Makefile.am
@@ -1,5 +1,5 @@
 lib_LTLIBRARIES = libzbargtk.la
-libzbargtk_la_CPPFLAGS = -I$(srcdir) -I$(top_srcdir)/include $(GTK_CFLAGS) $(AM_CPPFLAGS)
+libzbargtk_la_CPPFLAGS = $(GTK_CFLAGS) $(AM_CPPFLAGS)
 libzbargtk_la_LDFLAGS = -version-info $(ZGTK_LIB_VERSION) \
     -export-symbols-regex "^zbar_.*" $(AM_LDFLAGS) -no-undefined
 libzbargtk_la_LIBADD = $(GTK_LIBS) ../zbar/libzbar.la $(AM_LIBADD)
diff --git a/zbar/Makefile.am b/zbar/Makefile.am
index ce8bde3896b5..3981c299d543 100644
--- a/zbar/Makefile.am
+++ b/zbar/Makefile.am
@@ -1,5 +1,5 @@
 lib_LTLIBRARIES = libzbar.la
-libzbar_la_CPPFLAGS = -I$(srcdir) -I$(srcdir)/../include $(AM_CPPFLAGS)
+libzbar_la_CPPFLAGS = $(AM_CPPFLAGS)
 libzbar_la_LDFLAGS = -no-undefined -version-info $(LIB_VERSION) \
     -export-symbols-regex "^(zbar|_zbar.*_error)_.*" $(AM_LDFLAGS)
 libzbar_la_LIBADD = $(LTLIBICONV)
diff --git a/zbarcam/Makefile.am.inc b/zbarcam/Makefile.am.inc
index e8a529670747..02f7140725b5 100644
--- a/zbarcam/Makefile.am.inc
+++ b/zbarcam/Makefile.am.inc
@@ -1,7 +1,7 @@
 bin_PROGRAMS += zbarcam/zbarcam
 zbarcam_zbarcam_SOURCES = zbarcam/zbarcam.c
 zbarcam_zbarcam_LDADD = zbar/libzbar.la
-zbarcam_zbarcam_CPPFLAGS =
+zbarcam_zbarcam_CPPFLAGS = $(AM_CPPFLAGS)
 # automake bug in "monolithic mode"?
 CLEANFILES += zbarcam/.libs/zbarcam zbarcam/moc_zbarcam_qt.h 
```
I'm running a build with that right now:
https://travis-ci.org/mchehab/zbar/builds/535703230

mchehab, 2019-05-22:
It is almost that. At `configure.ac`, it shold be, instead:
-AM_CPPFLAGS="-I$top_srcdir/include"
+AM_CPPFLAGS="-I\$(top_srcdir)/include"

mchehab, 2019-05-22:
Seems to have worked. Cross-building for windows passed:

  - https://travis-ci.org/mchehab/zbar/jobs/535795845

closed 2019-05-22 by mchehab

mchehab, 2019-05-22:
> So, it would be great if you could send us a patch for .travis.yml and travis/* adding the exact cross-compilation environment you're using.

I added a target to cross-build it on Ubuntu Xenial: a039e69855a1e73f277f932c2463d25650f74e9d

It contains just the very basic stuff: no gtk, qt, python, gir or java. The best would be if you could improve it to add more stuff that builds for Windows, but with that, we should now be able to detect issues with cross-building.

A few minor issues:
1) I had to manually disable dbus. It probably makes sense to disable it automatically if WIN32;
2) The pthread detection at configure.ac is hacky. It should, instead, be using AX_PTHREAD;
3) Perhaps we could make ICONV dependency optional, or at least improve it, as configure.ac currently allows trying to build ZBar with it, but QR decode actually depends on such feature.

PR #48 (2019-05-18): Fix the build, include files not found and .rc rules missing
closes #47
commit 92e442f2 (2019-05-18): Fix the build, include files not found and .rc rules missing

closes #47