mchehab/zbar #54: OSX autoreconf -vfi fails

stuta, 2019-06-01:
``` bash
autoreconf -vfi

autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I config
configure.ac:141: warning: macro 'AM_ICONV' not found in library
autoreconf: configure.ac: tracing
autoreconf: running: glibtoolize --copy --force
glibtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'config'.
glibtoolize: copying file 'config/ltmain.sh'
glibtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'config'.
glibtoolize: copying file 'config/libtool.m4'
glibtoolize: copying file 'config/ltoptions.m4'
glibtoolize: copying file 'config/ltsugar.m4'
glibtoolize: copying file 'config/ltversion.m4'
glibtoolize: copying file 'config/lt~obsolete.m4'
configure.ac:141: warning: macro 'AM_ICONV' not found in library
autoreconf: running: /usr/local/Cellar/autoconf/2.69/bin/autoconf --force
configure.ac:141: error: possibly undefined macro: AM_ICONV
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
autoreconf: /usr/local/Cellar/autoconf/2.69/bin/autoconf failed with exit status: 1
```

Can't create working configure. It creates configure file but running it gives error: 
``` bash
configure: error: cannot find install-sh, install.sh, or shtool in config "."/config
```

mchehab, 2019-06-01:
I never tried to compile it on OSX, as I don't have any machines here with it installed. If we can make it work there, it would be great to be able to build it via Travis CI, in order to avoid breaking it.

> configure.ac:141: warning: macro 'AM_ICONV' not found in library

I guess you need `iconv` package in order to enable QR and SQ code.

> configure: error: cannot find install-sh, install.sh, or shtool in config "."/config

This is due to the lack of GNU libtool (a quick search at the Internet seems to indicate that the name of the package is `glibtool` there).

stuta, 2019-06-01:
Solved by:
```bash
brew install gettext autoconf automake libtool
brew link gettext --force
brew unlink libtool && brew link libtool
```

Clang does not work with openmp.

So I tried this: 
```bash
CC=/usr/local/Cellar/gcc/9.1.0/bin/gcc-9 CCLD=/usr/local/Cellar/gcc/9.1.0/bin/gcc-9 ./configure --disable-video --without-qt --without-gtk --without-python && make clean && make

Making all in .
  CC       zbarimg/zbarimg-zbarimg.o
  CCLD     zbarimg/zbarimg
ld: library not found for -lltdl
collect2: error: ld returned 1 exit status
```

How can I disable openmp so I can try clang build?

mchehab, 2019-06-01:
> How can I disable openmp so I can try clang build?

I've no idea. ZBar itself doesn't depend on it. Perhaps it is something specific to MacOS/OSX.

stuta, 2019-06-01:
Imagemagick is the problem (as always). This works with clang:

``` bash
./configure --disable-video --without-qt --without-gtk --without-python --without-imagemagick --with-graphicsmagick
```

closed 2019-06-01 by stuta

mchehab, 2019-06-01:
Thanks!

Added a TravisCI target for building with OSX: https://github.com/mchehab/zbar/commit/9d02280d5f9a14ea3b96fa3fe8c70e545507ffd1

Travis CI build should pass, just like the test one:

https://travis-ci.org/mchehab/zbar/builds/540139314

The `make install` target didn't work, though. I suspect that some work could be needed to make it more smooth on OSX.