mchehab/zbar #55: Decoding binary

matheusmoreira, 2019-06-01:
Decoding QR codes containing binary data with zbar 0.22 produces incorrect output.

Here's a script that reproduces the issue:

```bash
dd if=/dev/urandom bs=256 count=1          > test.bin
dd if=/dev/urandom bs=256 count=1 | base64 > test.b64

qrencode --8bit --output test.bin.png < test.bin
qrencode --8bit --output test.b64.png < test.b64

zbarimg --raw test.bin.png > test.bin.decoded.raw
zbarimg --raw test.b64.png > test.b64.decoded.raw

# zbarimg raw output is delimited by \n
printf "\n" >> test.bin
printf "\n" >> test.b64

sha1sum test.bin test.bin.decoded.raw test.b64 test.b64.decoded.raw

# 57f5e17a523964f26f6a8e7c006d6b0fa76e674b  test.bin
# 1012fc6b60af642cae02f354c16d856bf3281fd9  test.bin.decoded.raw
# 4172d922bfbc2f59ac350abe33d2746160ffa2c0  test.b64
# 4172d922bfbc2f59ac350abe33d2746160ffa2c0  test.b64.decoded.raw
```

closed 2020-03-13 by mchehab

hifi, 2024-02-12:
For anyone stumbling across this from search engines, you need to add `-Sbinary` option for `zbarimg` to get the fixed behavior that was released back in [0.23.1](https://github.com/mchehab/zbar/releases/tag/0.23.1).

The man page was never updated to include that option so it's very easy to miss.

matheusmoreira, 2024-02-12:
@hifi

That's true... I guess I forgot to update the manuals since I use the Arch Wiki so much. I edited the [paperkey] Arch Wiki article back in the day with instructions on how to use the binary decoding mode. It's the original use case I wanted this feature to enable. I described the new options in the `--help` output.

Anyone want to make a pull request?

[paperkey]: https://wiki.archlinux.org/title/Paperkey

tim-tx, 2025-04-22:
I reran OP's commands, including `-Sbinary` for the binary decode, but still didn't recover the original.

```
dd if=/dev/urandom bs=1024 count=1 > test.bin
base64 > test.b64                  < test.bin

qrencode -8 -o test.bin.8.png < test.bin
qrencode -8 -o test.b64.8.png < test.b64
qrencode -c -o test.b64.c.png < test.b64

zbarimg --raw -Sbinary test.bin.8.png | head -c -1 > test.bin.8.decoded.raw
zbarimg --raw test.b64.8.png | head -c -1 > test.b64.8.decoded.raw
zbarimg --raw test.b64.c.png | head -c -1 > test.b64.c.decoded.raw

sha1sum test.bin test.bin.8.decoded.raw \
        test.b64 test.b64.8.decoded.raw test.b64.c.decoded.raw

# 8063b58230a5b7240d7011e733162ba98f59e068  test.bin
# cb30370cdc6b849a2afb90d78219b4abe15a686d  test.bin.8.decoded.raw
# 28c09caed811e064d99503920fb89e3e3aeb7686  test.b64
# 28c09caed811e064d99503920fb89e3e3aeb7686  test.b64.8.decoded.raw
# 28c09caed811e064d99503920fb89e3e3aeb7686  test.b64.c.decoded.raw
```

Am I missing something? ArchLinux `zbar 0.23.93-4`, `qrencode 4.1.1-3`

matheusmoreira, 2025-04-22:
@tim-tx You also need `--oneshot`.

There might be an arbitrary number of QR codes in the image, so the tools will output separators between them in the form of a `\n` character. In my test script, I piped the output to `head -c -1` in order to remove that separator. I eventually added oneshot mode which prevents this separator from being written at all.

With `--raw --oneshot -Sqr.binary`, the output will be the exact binary contents of the first decoded QR code.

```bash
dd if=/dev/urandom bs=1024 count=1 > test.bin
base64 > test.b64                  < test.bin

qrencode -8 -o test.bin.8.png < test.bin
qrencode -8 -o test.b64.8.png < test.b64
qrencode -c -o test.b64.c.png < test.b64

zbarimg --raw --oneshot -Sqr.binary test.bin.8.png > test.bin.8.decoded.raw
zbarimg --raw test.b64.8.png | head -c -1 > test.b64.8.decoded.raw
zbarimg --raw test.b64.c.png | head -c -1 > test.b64.c.decoded.raw

sha1sum test.bin test.bin.8.decoded.raw \
        test.b64 test.b64.8.decoded.raw test.b64.c.decoded.raw

# cebbaa71a4753d27ae851781a27db6c8b95b9208  test.bin
# cebbaa71a4753d27ae851781a27db6c8b95b9208  test.bin.8.decoded.raw
# 9896a9adeb853c7f2c60afd2a89080e70d202cec  test.b64
# 9896a9adeb853c7f2c60afd2a89080e70d202cec  test.b64.8.decoded.raw
# 9896a9adeb853c7f2c60afd2a89080e70d202cec  test.b64.c.decoded.raw
```

tim-tx, 2025-04-22:
Thanks @matheusmoreira seems like `head` or `--oneshot` takes care of the separator, but more importantly `-Sqr.binary` instead of `-Sbinary` gives the expected result. Maybe time to update Arch Wiki for paperkey again, which still says `-Sbinary`!

matheusmoreira, 2025-04-22:
@tim-tx I tested `-Sbinary` as well and both gave the correct results. That form applies the `binary` option to all decoders, the `qr.` variant applies it to just the QR decoder which to my knowledge is the only one that makes use of it.

commit 59e0ed18 (2020-03-13): qr: add support for raw binary data extraction

If a QR code doesn't specify the text encoding of binary data through
Extended Channel Interpretation, ZBar tries to guess the encoding.
This unconditional character set conversion makes it impossible
to recover other types of binary data stored in the QR code.

The QR decoder now supports the ZBAR_CFG_BINARY configuration option.
If set, it will output the bytes without converting them to text.
This allows access to the actual bytes encoded in a binary QR code.

Closes: #55
Thanks: Martin von Wittich <martin.von.wittich@iserv.eu>