mchehab/zbar #56: Release precompiled binaries?

simonfranz, 2019-06-04:
Is it possible that from the current version also the /bin directory with e.g. zbarimg.exe is also checked in?
I think not everybody has the possibility to get the build running by himself/herself, so makign the binaries available would possibly be useful

best regards

mchehab, 2019-06-04:
There are two separate parts to this issue:

1) Makefile should have a target that would a "binaries" tarball/zip file with the produced files (something similar to `make dist` - except that this make target generates a source tarball). Right now, ZBar doesn't have any target to pack the generated libraries and binary files. 

2) Once the build succeeds on Travis CI, it should upload the tarballs to some place. Right now, the built stuff is kept only at the Travis CI VM.

For (2), I suspect Travis CI allows uploading the generated files to some place, but never tried to do that. It may have some issues on Windows, as, right now, support for it is on experimental phase. Eventually, it could, instead, generate those from the windows cross-compilation build on Ubuntu (e. g. from this kind of build: https://travis-ci.org/mchehab/zbar/jobs/540272649).

(1) is a lot harder, though. 

The problem is that the ZBar build relies on dynamic library link (`.so`/`.dll` files), instead on static ones (`.a` or `.lib` files). A program only runs with such libraries if the symbols there are compatible with the ones used on build time. In other words, it will only run if you run ZBar on a machine with the same libraries as the one used to build it.

So, if one builds it on Ubuntu Xenial, the binaries won't run on Fedora (with uses a different C library), or even on a newer Ubuntu version. The same applies to windows: if the build is done using a certain Windows version and a certain minGw version, it won't run on a machine with different runtime libraries. Just as an example, the Travis CI machine comes with two different minGw libraries. I had to add some tricks there in order to ensure that some commands were using the right library, as otherwise the application crashes and the build fails.

As you asked for `zbarimg.exe`, I'm assuming that you're more interested on Windows. Working on it it is a lot harder for me, as I don't have any development machine with Windows. Yet, if you're willing to work on it, feel free to submit patches and I'll integrate on our build system.

filler56789, 2019-08-01:
Therefore you should erase the file "README-windows.md" from Zbar's page.

OR update it.

jameshilliard, 2019-08-01:
> The same applies to windows: if the build is done using a certain Windows version and a certain minGw version, it won't run on a machine with different runtime libraries.

AFAIK this is much less of a problem for windows than Linux since Windows runtime libraries are usually very backwards compatible with any libraries that aren't typically being bundled with the application(maybe they aren't being bundled properly?).

Github releases would probably work for distributing binaries.

mchehab, 2019-08-01:
> AFAIK this is much less of a problem for windows than Linux since Windows runtime libraries are usually very backwards compatible with any libraries that aren't typically being bundled with the [...]

I'm not so sure about that. The Windows build environment on Travis CI has two different minGw libraries, and we had to do some tricks in order to make the build work over there, as, if we change the PATH for it to use the one where both autoconf and gcc works, bash breaks and vice-versa.

> Github releases would probably work for distributing binaries.

If someone provides a patchset that will produce a Windows package, I can integrate that at the building system.

mchehab, 2019-08-01:
> README-windows.md

Good point.

 Even if we add a Windows build, it has to be updated, as it mentions specific versions of the libraries that are from the time this was maintained at Sourceforge.

mchehab, 2019-08-01:
> > Therefore you should erase the file "README-windows.md" from Zbar's directory.

> Even if we add a Windows build, it has to be updated, as it mentions specific versions of the libraries that are from the time this was maintained at Sourceforge.

README-windows.md updated to reflect the current status. 

I added a section about building and removed the install section from the precompiled binaries.

If we end adding a precompiled package for Windows, I'll add the INSTALL instructions again.

Ortega-Dan, 2020-01-31:
I just want to link this issue with #77 since I'm trying to build according to the instructions with no success. 
I will appreciate so much your help. @mchehab

mchehab, 2021-02-15:
The binaries for Zbar will now be released sinc eversions 0.23.90 at:
https://linuxtv.org/downloads/zbar/binaries/

closed 2021-02-15 by mchehab

filler56789, 2021-02-15:
> The binaries for Zbar will now be released sinc eversions 0.23.90 at:
> https://linuxtv.org/downloads/zbar/binaries/

The new .EXE now requires libMagickWand7*.dll, which is not included in the .ZIP archive.

Ortega-Dan, 2021-07-08:
@filler56789 I faced the same thing but it is asking me for many more DLLs. Do you know if there is an easy way to get them all ?
[image]
[image]
[image]
And many more.
Or is there a way to get a static build not dependent of other DLLs ?

filler56789, 2021-07-08:
> 
> 
> @filler56789 I faced the same thing but it is asking me for many more DLLs. Do you know if there is an easy way to get them all ?
> ........
> And many more.
> Or is there a way to get a static build not dependent of other DLLs ?

ONE possible way is: _to use a MinGW-w64 build of gcc that ENFORCES the STATIC linking of runtime libraries such as seh-1.dll_.

filler56789, 2021-07-09:
@Ortega-Dan — I deleted my previous answer because it was incomplete.

1) POSSIBLY using a **NON-msys2** build of gcc would be sufficient solve the problem regarding the dynamically-linked lib*seh-1.dll.

2) As for the "many more" DLLs, _the makefile itself_ should be re-written in order to enforce the static linking to those libraries.

mchehab, 2021-07-09:
> @Ortega-Dan — I deleted my previous answer because it was incomplete.
>
>     1. POSSIBLY using a **NON-msys2** build of gcc would be sufficient solve the problem regarding the dynamically-linked [...]

The autoconf tools have some ways to generate a makefile that will use static libraries. Maybe this would work:

```./configure LDFLAGS="-static"```