mchehab/zbar #58: Add one shot mode

xaverdh, 2019-07-29:
Add a mode where the application quits after reading one qr code, rather than continue scanning.
This is useful for scripting, where you may not want to receive duplicate data.

matheusmoreira, 2019-09-12:
The `zbar` programs currently output a line feed to separate each QR code. This makes decoding binary data more complicated: the trailing byte must be discarded by piping the output to `head -c -1`.

A one shot mode would eliminate the need for this line feed processing, making it easier to consume the output of the decoding tools.

matheusmoreira, 2019-09-15:
I opened a pull request for this: #60. It adds `-1` and `--oneshot` options to `zbarimg` that cause it to exit after scanning one barcode. With `--raw` it will also omit the trailing line feed.

I'm working on `zbarcam` as well but I haven't been able to make it work yet.

commit dd0b1d9f (2019-12-23): Merge branch 'oneshot-tools' into oneshot

Add one shot mode to both zbarimg and zbarcam.

Closes: #58
Suggested-by: Dominik Xaver Hörl <hoe.dom@gmx.de>
closed 2019-12-23 by mchehab