mchehab/zbar #61: zbar crashes on QR code decoding

l3u, 2019-10-16:
I recently tried to decode a QR code of the chip tan stuff my bank does (out of curiosity what's encoded in it). Scanning such a QR code makes zbar crash. See the attached demo QR code:
[image]

The content seems to be some binary data, using https://zxing.org/w/decode.jspx , I get:

    DK
    X@Ã\2'NrND;�]uywgp~[rì

Perhaps, zbar expects the QR code content not to be bytes? Anyway, would be cool if it wouldn't crash ;-)

matheusmoreira, 2019-11-02:
I was able to reproduce the issue with `zbarimg 0.23`.

    double free or corruption (top)
    Aborted (core dumped)

However, I was able to decode the QR code with a `zbarimg` compiled from git.
The issue was fixed in commit 3b9c242a2975d3fdf4237fa2bdd05ef8aa89b6f1 which is newer than version 0.23.

---

This QR code contains Extended Channel Interpretation entry for `QR_ECI_GLI0` which signals that the data is encoded in CP437. It is followed by a binary representation of the following data:

    DK
    X@Ã\2'NrND;]uywgp~[rì

l3u, 2019-11-02:
Ah okay, the lastest we have here on Gentoo is 0.20.1. Thanks for the info!

closed 2020-04-09 by mchehab