mchehab/zbar #66: zbarimg fail to decode large QR code created by qrencode

petterreinholdtsen, 2019-11-26:
This issue is originally reported to https://bugs.debian.org/719013 .

I just tested this with zbar version 0.22, and sure enough, the problem still exist:

```
$ yes | dd bs=1 count=3000 | qrencode -S -v 32 -o /tmp/foo.png
$ zbarimg /tmp/foo-0*
scanned 0 barcode symbols from 2 images in 0.16 seconds

WARNING: barcode data was not detected in some image(s)
Things to check:
  - is the barcode type supported? Currently supported symbologies are:
    . EAN/UPC (EAN-13, EAN-8, EAN-2, EAN-5, UPC-A, UPC-E, ISBN-10, ISBN-13)
    . DataBar, DataBar Expanded
    . Code 128
    . Code 93
    . Code 39
    . Codabar
    . Interleaved 2 of 5
    . QR code
    . SQ code
  - is the barcode large enough in the image?
  - is the barcode mostly in focus?
  - is there sufficient contrast/illumination?
  - If the symbol is split in several barcodes, are they combined in one image?
  - Did you enable the barcode type?
    some EAN/UPC codes are disabled by default. To enable all, use:
    $ zbarimg -S*.enable <files>
    Please also notice that some variants take precedence over others.
    Due to that, if you want, for example, ISBN-10, you should do:
    $ zbarimg -Sisbn10.enable <files>

$
```

I expected zbar to handle QR codes generated by qrencode.

matheusmoreira, 2019-11-27:
I was able to reproduce this issue with `zbarimg 0.23` as well as the master branch.

jose1711, 2019-12-15:
possibly related: https://github.com/mchehab/zbar/issues/65

jose1711, 2020-01-22:
by looking at `debug.svg` it seems that grid finder got totally confused:
[image]

bitcoinhodler, 2020-02-04:
I noticed this problem recently when attempting to split a large block of data into multiple QR codes to solve [this issue](https://github.com/GlacierProtocol/GlacierProtocol/issues/78). My workaround was to split the data myself and run `qrencode` multiple times to create separate QR codes, instead of running `qrencode` once to create a series of "structured append" codes.

It would be nice if zbarimg could decode these, and even better if it could somehow report the structured append metadata (sequence indicator and parity data).

monperrus, 2020-05-02:
looking forward to seeing support for structured append in zbar!

maglub, 2021-02-26:
@jose1711 how do you generate an `debug.svg` file?

ionum, 2021-06-26:
@jose1711 You have to set DEBUG_SVG and recompile...

markuzzi, 2021-09-23:
What is the status of this issue? Is there any chance to recover a QR code that is created with qrencode? It is very large.