mchehab/zbar #68: Python 2/3: import zbar throws Import Error undefined symbol: zbar_image_get_crop

luxifeo, 2019-12-05:
I followed the building process at [README.md](../blob/master/README.md) and installed three binary program and python package successfully. However, when i try to import zbar, it throws the above error. Have tried on 4 different machines, both python 2 and 3, I still couldn't get past this. Currently i have to use subprocess, but i don't think it is the ideal way to do. 

OS tested:
Ubuntu 16.04
Ubuntu 18.04
LinuxMint 19.04

mchehab, 2019-12-23:
> I followed the building process at [README.md](../blob/master/README.md) and installed three binary program and python package successfully. However, when i try to import zbar, it throws the above [...]

Did you install the ZBar library? ZBar depends on a library on native code (written in C) for it to work.

You need it for the python bindings (or GIR) to work. Also, you could need to run ldconfig for it to update the dynamic load cache.

luxifeo, 2019-12-30:
> Did you install the ZBar library? ZBar depends on a library on native code (written in C) for it to work.
>
> You need it for the python bindings (or GIR) to work. Also, you could need to run ldconfig [...]

Thank you. I have figured it out. The python setup script refers to the default libzbar0.so rather than the newly installed one. Uninstall the old file solves the problem

closed 2019-12-30 by luxifeo