mchehab/zbar #69: Cannot disable QR code detection

luxifeo, 2019-12-05:
I removed the qrcode in enable_codes string at line 102 in [configure.ac](../blob/master/configure.ac) then built the program. Still detects QR code. Also, when i try to add some printf statements in function zbar_process_image in [processor.c](../blob/master/zbar/processor.c), it doesn't recognize the changes, and the new added codes don't work.
These two issues appear to happen at the same instance, so i think they are related.
Steps to reproduce:

- Modify enable_codes string in [configure.ac](../blob/master/configure.ac) or Modify codes in [processor.c](../blob/master/zbar/processor.c)
- Build the program 

Reproducible rate: 75%
Os tested:

- Ubuntu 16.04 gcc 5.4.0  (worked)
- Ubuntu 18.04 gcc 7.4.0
- Linuxmint 19.04 gcc 7.4.0

mchehab, 2019-12-23:
You don't need to change configure.ac to disable qrcode. Something like this should do the same:

```
./configure --with-gtk=auto --with-python=auto --enable-codes=ean,databar,code128,code93,code39,codabar,i25,sqcode
```

Here with Fedora 31, both removing qrcode or using the above works fine:

```
please verify that the detected configuration matches your expectations:
------------------------------------------------------------------------
X                      --with-x=yes
pthreads               --enable-pthread=yes
doc                    --enable-doc=yes
v4l                    --enable-video=yes
jpeg                   --with-jpeg=yes
Python                 --with-python=python3    python3.7
GTK                    --with-gtk=gtk3          Gtk3.24.13
GObject introspection  --with-gir=yes
Qt                     --with-qt=yes            Qt5.12.5
Java                   --with-java=yes
Dbus                   --with-dbus=yes
ImageMagick            --with-imagemagick=check
Enabled codes:         ean databar code128 code93 code39 codabar i25 sqcode
Disabled codes:        qrcode pdf417
JAVA_HOME              /usr/lib/jvm/java-11-openjdk-11.0.5.10-0.fc31.x86_64
```

Currently, I can't test Linuxmint or Ubuntu 18, as, for now, I'm using just a notebook without space for installing VMs on it.

From your comments, it seems that something is getting wrong with the logic which handles ```ZBAR_CHK_CODE```, but I was not able to find anything wrong there so far. Are you using the latest version from github git tree? If so, could you please post the output of ```./configure --enable-codes=i25 --with-gtk=auto --with-python=auto```?

luxifeo, 2019-12-30:
> You don't need to change configure.ac to disable qrcode. Something like this should do the same:
>
> ```
> ./configure --with-gtk=auto --with-python=auto [...]

The problem is still with the old libzbar0.so. The configure output is correct,  but the binary doesn't. Uninstalling old file solves.

closed 2019-12-30 by luxifeo