mchehab/zbar #70: Code-93 barcode not decoded

demolitions, 2019-12-05:
Hello,
I'm using zbar 0.20 from the command line (invoked from a PHP script, if that matters), to decode a png image that contains a single Code-93 barcode (extracted from a PDF).
Every hundred barcodes, I get one that is not being decoded, a failing sample is attached.

[image]

zbar does not crash nor segfaults, it simply answers me: 
`scanned 0 barcode symbols from 1 images in 0.02 seconds`

I have tried with v0.23, and I get the same result.

Any ideas on how to proceed from here?

mchehab, 2019-12-23:
Image size seems too small... increasing scale does the trick:

```
$ convert /tmp/70225199-7ec1c880-1746-11ea-90b5-7a99e14e4eec.png -scale 200% /tmp/new.png
$ zbarimg /tmp/new.png 
CODE-93:T11413AQW1KT111194ARMP6
scanned 1 barcode symbols from 1 images in 0.01 seconds
```

demolitions, 2020-01-13:
Implemented a rescale before invoking zbar, thanks.

closed 2020-01-13 by demolitions