mchehab/zbar #71: iPhone: ITMS-90809: Deprecated API Usage - UIWebView

Animal451, 2019-12-19:
Hi there,

  ZBarHelpController.h/.m use UIWebview, and this API is on Apple's deprecation list:
https://developer.apple.com/documentation/uikit/uiwebview

  When uploading the app to appstoreconnect you get a warning that future submissions containing these API references will be refused automatically.

  Apple's direction is to replace use of UIWebView with WKWebView.

  I'm working on fixing this locally.  But, just in case I missed something, I'm raising this issue ticket for awareness and fair warning in case there's a clamour of 'It's fixed already - look more closely!'

  Thank you,

  Ben.

mchehab, 2019-12-23:
> Hi there,
>
> ZBarHelpController.h/.m use UIWebview, and this API is on Apple's deprecation list:
> https://developer.apple.com/documentation/uikit/uiwebview
>
> When uploading the app to appstoreconnect [...]

Once you have a fix, please submit. I don't have any Apple's device for testing the changes.

daniel4123, 2020-02-12:
@Animal451, are you still working in this issue? 
It seems that the adaptation in master from `UIWebView` to `WKWebView` is partially incorrect; 
```webViewDidFinishLoad``` is part of `UIWebViewDelegate` protocol, and the correct replacement would be its WKWebView's analogous delegate method, the `didFinishNavigation` of the WKNavigationDelegate protocol. 
Thanks!

Animal451, 2020-02-12:
Hi Daniel,

thanks for pointing this out!  I missed it entirely, so will happily update it once I get a chance to rebuild and test.  Unless, of course, someone else beats me to it :)

  Ben.

ghr1234, 2020-04-07:
I've just sent changes to Mauro which fix this. Included are some other changes so the reader and help screens appear better on iPhoneX family.

I missed this problem as the previous partial change  mentioned ticket #17 rather than ticket #71

Godfrey

mchehab, 2020-04-07:
Patches applied. Please test.

Btw, right now, our Continuous Integration tests won't be able to build it. On a quick look at TravisCI (with we currently use for such tests), it seems that the MacOS support may work for iPhone builds:

https://docs.travis-ci.com/user/languages/objective-c/#default-test-script

closed 2020-04-09 by mchehab