mchehab/zbar #76: How To Start Zbar

KitchM, 2020-01-29:
Can anyone tell me how to start the application?  I have it installed on Debian 10 but there is no menu item.

Thanks.

Ortega-Dan, 2020-01-31:
Hi, 

From a terminal do:
$ zbarimg 
[image]

Or you can also do zbarcam to run it from the default camera device.

Hope it helps

KitchM, 2020-02-10:
Thank you.  I guess I thought something else.  Is there a GUI for it, or another that has one?

matheusmoreira, 2020-03-06:
> Is there a GUI for it, or another that has one?

There's `zbarcam-gtk` and `zbarcam-qt`. These programs should be packaged by Debian as well.

KitchM, 2020-03-06:
Thanks matheusmoreira.  I agree that it should be, but sadly it is not.  Couldn't find the .deb package anywhere else either.  Bummer.

closed 2020-04-09 by mchehab