mchehab/zbar #77: Not Building for Windows "cannot find -lMagickWand-6.Q16"

Ortega-Dan, 2020-01-31:
Hi, 

I really will appreciate your help. Thanks in advance for the great work done here:

I'm trying to build for windows following all instructions but adding imagemagick since I need to build zbarimg ... for Windows.

Then the instructions I am following from a brand new Ubuntu 18.04 are:

    sudo apt-get install -y \
        autoconf automake autotools-dev libdbus-1-dev \
        pkg-config binutils-mingw-w64-i686 gcc-mingw-w64 \
        mingw-w64-i686-dev mingw-w64-common win-iconv-mingw-w64-dev \
        xmlto imagemagick libmagickwand-dev gettext libtool

Then, build Zbar with:

Then, build Zbar with:

    export PKG_CONFIG_PATH=/usr/x86_64-w64-mingw32/lib/pkgconfig

    autoreconf -vfi

    CC=i686-w64-mingw32-gcc ./configure --host=i686-w64-mingw32 \
        --prefix=/usr/local/win32 --with-directshow \
        --without-gtk --without-python --without-qt --without-java \
        --enable-pthread

    make

But after compiling everything else to that point I am getting the following error:
  CCLD     zbarimg/zbarimg.exe
/usr/bin/i686-w64-mingw32-ld: cannot find -lMagickWand-6.Q16

I will appreciate so much your help. We have a great need to build this one. 
Thanks in advance again.

Ortega-Dan, 2020-01-31:
The full error lines are:

make  all-recursive
make[1]: Entering directory '/home/danort/code/zbar'
Making all in zbar
make[2]: Entering directory '/home/danort/code/zbar/zbar'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/danort/code/zbar/zbar'
Making all in .
make[2]: Entering directory '/home/danort/code/zbar'
  CCLD     zbarimg/zbarimg.exe
/usr/bin/i686-w64-mingw32-ld: cannot find -lMagickWand-6.Q16
/usr/bin/i686-w64-mingw32-ld: cannot find -lMagickCore-6.Q16
collect2: error: ld returned 1 exit status
Makefile:1298: recipe for target 'zbarimg/zbarimg.exe' failed
make[2]: *** [zbarimg/zbarimg.exe] Error 1
make[2]: Leaving directory '/home/danort/code/zbar'
Makefile:1831: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/danort/code/zbar'
Makefile:917: recipe for target 'all' failed
make: *** [all] Error 2

mchehab, 2020-04-06:
You can build ZBar for windows either on an Ubuntu host or directly on Windows.

We currently have some scripts that do that for TravisCI. Take a look at those files for building it directly on windows:
```
    travis/windows.pc.before_install 
    travis/windows.pc.install
    travis/windows.pc.script
```

And those files to use mingw to build it from Ubuntu:
```
   travis/linux.mingw-w64.before_install
   travis/linux.mingw-w64.install
   travis/linux.mingw-w64.script
```

mchehab, 2020-04-07:
> The full error lines are:
>
> make all-recursive
> make[1]: Entering directory '/home/danort/code/zbar'
> Making all in zbar
> make[2]: Entering directory '/home/danort/code/zbar/zbar'
> make[2]: Nothing to be [...]

Btw, MagickWand is an external library. It needs to be manually installed in a place that GCC will look when trying to merge with the compiled code.

mchehab, 2020-04-09:
I double-checked the dependencies when building natively on Windows. There, I had do manually disable ImageMagick with:

```
    --without-imagemagick 
```

The drawback is that ZBar will lose the capability of working with lots of different image formats.

In any case, as this is not a bug, but, instead, a configuration option (that requires installing an extra library), I'm closing this issue.

closed 2020-04-09 by mchehab