mchehab/zbar #82: Question: when are new releases made?

peter-moran, 2020-04-03:
When are new releases made?

I would like to use the new binary qr scanning feature, but this is not in the latest release. For now, I am using the the latest commit, but sake of configuration management it would be nice to have a new release at some point.

Thanks!

linxon, 2020-04-04:
.

mchehab, 2020-04-06:
We don't have a strict release policy. I try to do at least one or two releases per year, but that depends on the number of features and bug fixes I receive.

That's said, it is probably about the time to do another release. 

As there are some work going on by some developers, I'll be waiting for a couple of days to receive any new stuff for the upcoming new version.

alexmyczko, 2020-04-09:
Instead of opening plenty new issues, would you mind looking at https://bugs.debian.org/cgi-bin/pkgreport.cgi?src=zbar before doing the new release?

mchehab, 2020-04-09:
> Instead of opening plenty new issues, would you mind looking at https://bugs.debian.org/cgi-bin/pkgreport.cgi?src=zbar before doing the new release?

I reviewed the bugs at the Debian bug list and submitted some answers via e-mail (not displayed there yet).

From those, I was unable to reproduce those:

```
    #803967 [n|  |  ] [zbar-tools] zbar-tools: Regression: 0.10+doc-10 cannot read barcodes from images that 0.10+doc-8 can
    #869234 [i|  |  ] [zbar-tools] /usr/bin/zbarimg: Took much to long and failed to decode image with 123 qrcodes
```
The first one, the Debian maintainer wasn't capable of reproducing it either and already asked the reporter to make the broken image file available for testing.

The second one has an Image that zbarimg can't recognize the codes, probably because of the low resolution of the uploaded version. Even if we had a good version of it, this is probably something that we can't do much anyway: handling an image with 123 qrcodes will likely spend some time.

---

Those are already solved upstream:

```
    #703234 [n|  |  ] [libzbar0] libzbar0 recodes 8-bit QR-encoded data into UTF-8
    #710009 [n|  |  ] [zbar-tools] /usr/bin/zbarimg: Decoding binary data fails
```

So, it is just a matter of upgrading Debian package to upstream.

---

This one is not really a bug:

```
    #642896 [m|  |  ] [zbar-tools] zbar-tools: --xml as last parameter generates empty xml
```

In any case, I agree with the report that the way ZBar handles it is not intuitive, so I changed the parameter parsing logic today.

---

So, from that list, the only pending issue is this one:

```
    #719013 [i|u|   ^f^}] [zbar-tools] zbar-tools: zbarimg fail to decode structured qr code
```

Which has already an open issue: https://github.com/mchehab/zbar/issues/66

Fixing it may not be trivial. Volunteers?

mchehab, 2020-04-09:
> I reviewed the bugs at the Debian bug list and submitted some answers via e-mail (not displayed there yet).
> ```
>     #869234 [i|  |  ] [zbar-tools] /usr/bin/zbarimg: Took much to long and failed to decode image with 123 qrcodes
> ```

This one has actually an issue associated to it:

https://github.com/mchehab/zbar/issues/34

mchehab, 2020-04-20:
Just released version 0.23.1:

```0.23.1:```

  * Add i18n translation support and add a pt_BR translation
  * Change default to autodetect python and gtk versions
  * Improve tests and builds on travisCI
  * Some documentation updates
  * Add support for binary data extraction
  * Add support for raw decoding without charset conversions
  * Add one shot scanning mode
  * Improved OSX and iphone support
  * Fix help messages
  * Fix some makefile issues
  * Fix error detection in video4linux read
  * Fix pkgconfig for zbar-qt5
  * Fix a double-free condition
```

closed 2020-04-20 by mchehab

hosiet, 2020-04-21:
@mchehab  It looks like we miss the tag ov 0.23.1. Any chance of tagging the release in https://github.com/mchehab/zbar/releases and/or https://github.com/mchehab/zbar/tags ?

matheusmoreira, 2020-04-22:
I've notified the [Arch Linux package] maintainer. The new version should be available there soon.

[Arch Linux package]: https://www.archlinux.org/packages/extra/x86_64/zbar/

mchehab, 2020-04-22:
> @mchehab It looks like we miss the tag ov 0.23.1. Any chance of tagging the release in https://github.com/mchehab/zbar/releases and/or https://github.com/mchehab/zbar/tags ?

Tag added.