mchehab/zbar #84: raspberry pi install (with java) - issues

petercwallis, 2020-04-12:
Trying to run configure on the latest raspbian image, these issues occurred:

needed to 'apt-get install gettext' to get past an issue with AM_ICONV().  It then could tell me it wasn't using gettext

needed to 'apt-get install libgtk2.0-dev' Tried libgtk-3-dev but that failed

cannot install the java examples because javah is missing under jdk11.  These days the header files are produced from the .java files using javac -h
See this article for the gory details: https://stackoverflow.com/questions/48816188/kotlin-replacement-for-javah
It seems the only solution is to use java 8 :-/

While I am here, two questions:
what is an "SQ Code" as opposed to a QR Code, and how would I generate one? 

and does anyone know what the codes in the boston dynamics warehouse video are?
https://www.youtube.com/watch?v=5iV_hB08Uns

mchehab, 2020-04-12:
> Trying to run configure on the latest raspbian image, these issues occurred:
> 
> needed to 'apt-get install gettext' to get past an issue with AM_ICONV(). It then could tell me it wasn't using gettext

ZBar itself doesn't support yet internationalization. So, it doesn't require gettext. That doesn't mean that dependent packages like iconv would not require it.

> needed to 'apt-get install libgtk2.0-dev' Tried libgtk-3-dev but that failed

If you want to use gtk3, you need to pass an option to ./configure, as the current default is gtk2. I'll likely change the default to auto before releasing the next version.

> cannot install the java examples because javah is missing under jdk11. These days the header files are produced from the .java files using javac -h
> See this article for the gory details: [...]

ZBar (upstream) has a logic with checks Java. See HAVE_JAVAH at java/Makefile.am.  It can work either with java <= 8 or with java 11. Perhaps you tries to build an older version of ZBar. If not, then configure.ac Java detection may need some fixes to work with raspbian (or with whatever distro you're using).

> While I am here, two questions:
> what is an "SQ Code" as opposed to a QR Code, and how would I generate one?

SQ code is another 2D QR code. See `examples/sqcode1-generated.png`.
See https://github.com/mchehab/zbar/pull/2 for more details.

> and does anyone know what the codes in the boston dynamics warehouse video are?
> https://www.youtube.com/watch?v=5iV_hB08Uns

petercwallis, 2020-04-13:
Thanks. the configure file relies on JAVA_HOME being set.  Fair enough, although please note raspberry pis don't set it by default.  That is what happens when the computer is £30 and an SD card a 5er. A message to that effect or, if not set, looking in /usr/bin/ for javah would be icing on the already very nice cake.

However I still get the message 
the java unit test will *NOT* be enabled
looking at the log I get the impression the java stuff is all there but not the test. Will press on and close this later if there are no issues.

alexmyczko, 2020-04-16:
raspbian does have binaries, built from sources derived from debian: https://repology.org/project/zbar/versions

can you not just use their package? what version do you get?