mchehab/zbar #94: VERY OLD version of zbarimg in macOS

johfpardo, 2020-06-23:
I created a bash script to scan QR codes and I want it to be a multi-platform script to use in linux and macOS. However, the macOS's zbarimg can't read some QR codes well as linux's zbarimg does.

I searched why was the reason of this and I found that the version of zbarimg that macOS uses is lower than linux uses (0.10 < 0.23).

So my question is how I can upgrade the zbarimg to 0.23 version in macOS and It would work as expected?

Please help me

jasp00, 2020-06-23:
You may ask your macOS package provider to upgrade to the new version or you may build it from source.

johfpardo, 2020-06-23:
I installed manually using the scripts in the travis folder, thanks !

closed 2020-06-23 by johfpardo