mchehab/zbar #98: License issue due to zbar/gettext.h

ffontaine, 2020-07-27:
zbar/gettext.h which has been added by commit d05911f8d5fb8c1e064bd93ed9ec9f038c5da096 is licensed under GPL-3.0 and not LGPL-2.1+. As a result, since this change, zbar should be licensed under GPL-3.0. Is it intentional?

hosiet, 2020-07-27:
FYI: https://savannah.gnu.org/bugs/?56863

mchehab, 2020-07-28:
No, this was not intentional. It should be noticed that, so far, the only files affected by this are the ones which actually use gettext.h:

```
$ grep "#:" po/zbar.pot
#: zbarcam/zbarcam.c:50
#: zbarcam/zbarcam.c:73 zbarimg/zbarimg.c:103
#: zbarimg/zbarimg.c:81
#: zbarimg/zbarimg.c:107
#: zbarimg/zbarimg.c:113
#: zbarimg/zbarimg.c:501
#: zbarimg/zbarimg.c:504
#: zbarimg/zbarimg.c:507
#: zbarimg/zbarimg.c:510
#: zbarimg/zbarimg.c:513
#: zbarimg/zbarimg.c:516
#: zbarimg/zbarimg.c:519
#: zbarimg/zbarimg.c:522
#: zbarimg/zbarimg.c:525
#: zbarimg/zbarimg.c:528
```

I would prefer to keep them licensed under LGPL (or GPLv2 - which IMO makes more sense for the non-library part). Do you have any suggestions about how to keep the translations without contaminating the translated parts with gettext licensing issues?

mchehab, 2020-07-28:
Hmm... I was expecting it to be harder, but just removing the header is enough: a133aea7880bbb56d75535e534716d0e16a3b61a.

So, I guess the issue is solved.

closed 2020-07-28 by mchehab