mchehab/zbar #104: Name error when using zbarimg

DamienCassou, 2020-08-11:
I always get this error when using zbarimg:

```
$ zbarimg -q --raw firefox.combyK5uV.png
Name Error (Connection ":1.699" is not allowed to own the service "org.linuxtv.Zbar" due to security policies in the configuration file)
otpauth://totp/Firefox:damien%40cassou.me?secret=MRDXMULLKA4HAKZSJBLGKULLHBMGMVLV&issuer=Firefox
```

DamienCassou, 2020-08-11:
- zbarimg --version : 0.23.1
- OS : NixOS

acc987, 2021-07-02:
I am encountering the same issue when using zbarimg (0.23.1 and also with 0.23.90 from the Application:Geo repository) with openSUSE Leap 15.3.

lechner, 2022-05-26:
Hi, I also saw that in Guix when installing `zbar` as a user.

Thanks for this great piece of software!

lrustand, 2024-12-04:
This problem is still here in Guix