mchehab/zbar #106: [Question] What is the difference between version 0.23 and version 0.10 for QR code recognition?

hyojoo-cho, 2020-08-20:
Hello.

I'm looking for a way to improve QR code recognition speed.
I'm going to use  Ordroid H2 / Linux 18.04 / c++

Q1. In the case of Zbar, there are two versions.  Compared to past version (0.10), is there any improvement on QR code recognition in the latest version?

Q2. As a result of testing with minimal code, zbar::scan(img) took 10ms to recognize QR marker about 35mm size. If I want to recognize only QR code, is there a way to speed it up?

Thank you.

tobltobs, 2020-10-07:
Version 0.23 seems buggy to me.
The attached QRcode decoded with 0.10: DICDAn Ch Er On Qu Sch Sh St Th an and are ati atio cb ch ck con ee ent er ere ff fi fl for
With version 0.23: n Ch Er On Qu Sch Sh St Th an and are ati atio cb ch ck con ee ent er ere ff fi fl for
The result from v0.10 is the correct one.

[image]

mchehab, 2021-02-15:
> Hello.
> 
> I'm looking for a way to improve QR code recognition speed.
> I'm going to use Ordroid H2 / Linux 18.04 / c++
> 
> Q1. In the case of Zbar, there are two versions. Compared to past version (0.10), is there any improvement on QR code recognition in the latest version?

Version 0.10 were the original version, written many years ago. Version 0.20 and above is an effort of keeping maintaining ZBar. There are lots of differences and bug changes everywhere.

> Q2. As a result of testing with minimal code, zbar::scan(img) took 10ms to recognize QR marker about 35mm size. If I want to recognize only QR code, is there a way to speed it up?

ZBar allows to disable other decoders. That could give you some performance improvements, e. g.:

```
./zbarimg/zbarimg -S*.disable -Sqr.enable examples/qr-code.png
```

ldoolitt, 2022-11-25:
> Version 0.23 seems buggy to me.

The one-line patch I posted in #237 appears to fix this case.  Please check.