mchehab/zbar #121: Zbarimg failes to decode a Swiss invoice QR code when it contains a logo

KlaasJelmer, 2020-09-27:
Hello,

Zbarimg fails to detect the QR code format that is often found on Swiss invoices. I created the follow QR code using https://www.codecrete.net/qrbill/bill :

[image]

When I remove the logo from the image, Zbarimg detects the QR code correctly:

[image]

The command that I used for both images: `zbarimg -Sbinary -q filename.png`. 
The expected output for both QR codes is:
```
SPC
0200
1
CH450023023099999999A
S
Schreinerei Habegger & Söhne
Uetlibergstrasse
138
8045
Zürich
CH

287.30
EUR
S
Simon Glarner
Bächliwis
55
8184
Bachenbülach
CH
SCOR
RF192320QF02T3234UI234
Rechnungsnr. 10978 / Auftragsrnr. 3987
EPD
```

ionum, 2021-02-19:
Same problem for me

ionum, 2021-06-26:
Generated a debug.svg. 3 qrcode-markers have been recognized. Seems to be an error correction problem. As described in #167 it seems to depend on the size of the swiss cross.... @mchehab Do you fix problems on qrcode source code?

mchehab, 2021-06-26:
> @mchehab Do you fix problems on qrcode source code?

Yes, but only when I have spare time. See, my main focus is to maintain the media subsystem at the Linux Kernel. Those days, I'm not having much time, so, I'm just ensuring that ZBar doesn't get bit-rotted and I'm applying patches developed by others.

dpocock, 2022-10-03:
ZXing is able to scan these QR codes

Is there anything in the ZXing bug tracker or repository history that may give us ideas about the easiest way to handle this?