mchehab/zbar #122: Homebrew `brew install zbar` doesn't include zbarcam

sunknudsen, 2020-10-10:
First, thanks for maintaining ZBar.

Is it normal that `zbarcam` is not installed when running `brew install zbar`?

jameshilliard, 2020-10-10:
@sunknudsen You could try removing [this](https://github.com/Homebrew/homebrew-core/blob/923457d5c49fdbc97d5d07fe8c36d67a3edae58b/Formula/zbar.rb#L42) line from the formula.

sunknudsen, 2020-10-10:
@jameshilliard Thanks for your help... unfortunately that doesn't work.

```shell
==> ./configure --prefix=/usr/local/Cellar/zbar/0.23.1_10 --without-python --wit
Last 15 lines from /Users/sunknudsen/Library/Logs/Homebrew/zbar/02.configure:
checking for pthread_create in -lpthread... yes
checking linux/videodev.h usability... no
checking linux/videodev.h presence... no
checking for linux/videodev.h... no
checking linux/videodev2.h usability... no
checking linux/videodev2.h presence... no
checking for linux/videodev2.h... no
checking libv4l2.h usability... no
checking libv4l2.h presence... no
checking for libv4l2.h... no
configure: error: in `/private/tmp/zbar-20201010-5261-8g44zn/zbar-0.23.1':
configure: error: test for video support failed!
rebuild your kernel to include video4linux support or
configure --disable-video to skip building video support.
See `config.log' for more details
```

mchehab, 2020-10-10:
I'm assuming that you want to build it for Mac OS (and/or iOS), right? 

Currently, `zbarcam` knows only those ways to talk with camera devices:

- On Windows: DirectShow (Windows) or VFW (Windows);
- On Linux: libv4l, V4L2 (and V4L1, although this is very old)

Currently, there's no support for talking with the camera/video API on Mac OS/iOS.

If someone wants to add support for it, this would be a great addition to ZBar.

jasp00, 2020-10-10:
AVFoundation may be a good start. FFmpeg is using this framework.

sunknudsen, 2020-10-11:
@mchehab Yes, macOS...

> If someone wants to add support for it, this would be a great addition to ZBar.

That would be amazing!

@jasp00 Interesting. Thanks for sharing.

Hope someone with the right skill set figures out a way to support zbarcam on macOS.

hdfsyu, 2021-09-04:
Try compiling on a virtual machine and see if that works