mchehab/zbar #123: QR Code not detected

pmichalek, 2020-10-12:
Attached QR is not detected by zbarimg 0.23. Scan by phone (QRDroid) from screen is OK.
QR code image was extracted from PDF.
[image]

jreason, 2020-10-29:
I am currently having the same problem.
It appears that whenever i scan an image in with an A4 scanner, the QR Code is never detected.
The scan is saved to a PDF by the scanner software.
I then convert the PDF file to a PNG file in order to use zbarimg.

I have used various barcode readers on mobile devices on the PNG document to make sure the QR Code is not corrupt and they all read the barcode fine.

I would love to know how to fix this. Can provide sample documents if needed.

jounathaen, 2021-02-01:
Same here. This is an example of a QR code which is not detected by zbar:
[image]

Maxdamantus, 2021-02-02:
In case it's useful, @pmichalek's image can be made readable by copying all black pixels 1 px to the right and then again copying all black pixels 1 px down:
[image]

maglub, 2021-02-26:
I have one more example, Swiss QR Bill codes:

[image]

These are used on bills in Switzerland, and can be generated here: https://qrportal.ch/one#paymentForm

I found this docu mentioning "QR-IID": https://www.credit-suisse.com/media/assets/private-banking/docs/ch/unternehmen/kmugrossunternehmen/c2174-qr-rechnung-uebersicht-en.pdf

```
(venv) [maglub@Magnuss-MacBook-Pro-2:~/dev/wue/dani]$ zbarimg magnus2.jpg 
scanned 0 barcode symbols from 1 images in 0.02 seconds

WARNING: barcode data was not detected in some image(s)
Things to check:
  - is the barcode type supported? Currently supported symbologies are:
	. EAN/UPC (EAN-13, EAN-8, EAN-2, EAN-5, UPC-A, UPC-E, ISBN-10, ISBN-13)
	. DataBar, DataBar Expanded
	. Code 128
	. Code 93
	. Code 39
	. Codabar
	. Interleaved 2 of 5
	. QR code
	. SQ code
  - is the barcode large enough in the image?
  - is the barcode mostly in focus?
  - is there sufficient contrast/illumination?
  - If the symbol is split in several barcodes, are they combined in one image?
  - Did you enable the barcode type?
    some EAN/UPC codes are disabled by default. To enable all, use:
    $ zbarimg -S*.enable <files>
    Please also notice that some variants take precedence over others.
    Due to that, if you want, for example, ISBN-10, you should do:
    $ zbarimg -Sisbn10.enable <files>
```

maglub, 2021-02-26:
Interesting, when I removed the "swiss" sign in the middle of the QR code, zbarimg can read it.

Bad image:

[image]

Edited image:

[image]

```
(venv) [maglub@Magnuss-MacBook-Pro-2:~/dev/wue/dani]$ zbarimg -q removed_swiss_mark.png 
QR-Code:SPC
0200
1
CH760000000000000
S
Some person
Somewhere
10
1234
Swisstown
CH

1234
CHF
S
Other person
Somewhere
11
9876
Swisstropolis
CH
NON

Please send money
EPD
```

kwrobel, 2021-04-21:
I am able to provide a few more images that zbarimg does not decode, but my iPhone has zero issue detecting.

[image]
[image]
[image]

We are scanning documents that we stick a 1"x1" label with a QR Code. Our success rate is fairly good, but every once in a while we also have documents that our software does not process because the QR Code wasn't detected. I am curious why my iPhone can decode these, but zbarimg can't. If there's anything I can do to help, please let me know.

jounathaen, 2021-04-22:
I also tried to recognize scans via QR Codes, and my conclusion was, that QR codes are not reliable enough for recognizing documents. 
They seem to be rather sensitive about the thickness, shape and orientation of the markers. The reason your phone can recognize them so reliable is, that it has multiple tries. With a standard camera you can scan like 30 images per second and due to the movement of the hand, you'll get a result fairly quickly.

 But for a document you have only a single shot. I switched to barcodes, and they are far better recognizable for scans. Maybe there are more fancy kind of codes which also work reliable for scans but I stopped researching there.

adeliktas, 2022-01-05:
just tried/tested and failed to decode a qrcode of a screenshot with zbarimg of zbar-0.23.1 (but works online with zxing.org)
[image]

zeusin, 2022-03-16:
I use this solution and it helps me: https://stackoverflow.com/questions/69689869/improve-zbarimg-qrcode-recognition?newreg=25681a87a68e4de3baf9f2cd28000883

convert input.png +repage -threshold 50% -morphology open square:1 output.png
The important parameter is the morphology one

jreason, 2022-03-18:
@zeusin im afraid this command still does not work for me

[qr-code.pdf](https://github.com/mchehab/zbar/files/8305564/qr-code.pdf)

sarfulsheikh09-hub, 2026-09-03:
> Attached QR is not detected by zbarimg 0.23. Scan by phone (QRDroid) from screen is OK. QR code image was extracted from PDF. [image]

123@123

sarfulsheikh09-hub, 2026-09-03:
123@123

ykne, 2026-09-03:
six years later, no changes. the 1px gap between black squares is the root cause, nothing else.