mchehab/zbar #128: FourCC format codes

xloem, 2020-11-14:
I'm trying to feed data into the library and I'm unsure what fourcc format code to provide.  Is there a list anywhere?

mchehab, 2021-01-09:
No, but you can do a grep to get what fourcc's are supported: `git grep -i fourcc`.

It should be noticed that ZBar core library is capable of using lots of different formats, but the clients may support a smaller set of fourccs. Also, it depends on the OS used. 

In the specific case of Linux, ZBar will use by default `libv4l2`, with does format conversion internally if needed. This allow it to support a wider range of cameras that only output video on proprietary fourccs.

A small perl script can help to check what's supported:

```#!/usr/bin/perl
use strict;

my %fourccs;
open IN,"git grep -i fourcc |";
while (<IN>) {
	$fourccs{$1}{"$2$3$4$5"} = 1 if (m/^(\S+):.*fourcc\(\s*\'(\w)\'\s*,\s*\'(\w)\'\s*,\s*\'(\w)\'\s*,\s*\'(\w)\'\s*\)/);
}
close IN;

foreach my $file (sort keys %fourccs) {
	foreach my $f (sort keys %{$fourccs{$file}}) {
		print "$file: $f\n";
	}
}
```
Such script returns:

```examples/scan_image.c: Y800
gtk/zbargtk.c: BGR4
gtk/zbargtk.c: R444
gtk/zbargtk.c: RGB3
gtk/zbargtk.c: RGBR
gtk/zbargtk.c: Y800
include/zbar/QZBarImage.h: BGR4
iphone/ZBarCVImage.m: CV2P
iphone/ZBarImage.m: RGB3
iphone/ZBarImage.m: RGB4
iphone/ZBarImage.m: RGBQ
iphone/ZBarImage.m: Y800
test/test_convert.c: I420
test/test_cpp_img.cpp: GREY
test/test_cpp_img.cpp: RGB3
test/test_images.c: 411P
test/test_images.c: 422P
test/test_images.c: BGR3
test/test_images.c: BGR4
test/test_images.c: GREY
test/test_images.c: I420
test/test_images.c: NV12
test/test_images.c: NV21
test/test_images.c: RGB3
test/test_images.c: RGB4
test/test_images.c: RGBO
test/test_images.c: RGBP
test/test_images.c: RGBQ
test/test_images.c: RGBR
test/test_images.c: UYVY
test/test_images.c: Y800
test/test_images.c: YU12
test/test_images.c: YUV9
test/test_images.c: YUY2
test/test_images.c: YUYV
test/test_images.c: YV12
test/test_images.c: YVU9
test/test_images.c: YVYU
test/test_jpeg.c: JPEG
test/test_jpeg.c: Y800
test/test_proc.c: BGR3
test/test_video.c: BGR3
test/test_window.c: BGR4
test/test_window.c: I420
test/test_window.c: RGB3
test/test_window.c: UYVY
test/test_window.c: Y800
test/test_window.c: YUYV
test/test_window.c: YV12
zbar/convert.c: 411P
zbar/convert.c: 422P
zbar/convert.c: BA81
zbar/convert.c: BGR1
zbar/convert.c: BGR3
zbar/convert.c: BGR4
zbar/convert.c: GREY
zbar/convert.c: HI24
zbar/convert.c: HM12
zbar/convert.c: I420
zbar/convert.c: JPEG
zbar/convert.c: MJPG
zbar/convert.c: MPEG
zbar/convert.c: NV12
zbar/convert.c: NV21
zbar/convert.c: R444
zbar/convert.c: RGB1
zbar/convert.c: RGB3
zbar/convert.c: RGB4
zbar/convert.c: RGBO
zbar/convert.c: RGBP
zbar/convert.c: RGBQ
zbar/convert.c: RGBR
zbar/convert.c: UYVY
zbar/convert.c: Y41P
zbar/convert.c: Y444
zbar/convert.c: Y800
zbar/convert.c: YU12
zbar/convert.c: YUV4
zbar/convert.c: YUV9
zbar/convert.c: YUVO
zbar/convert.c: YUY2
zbar/convert.c: YUYV
zbar/convert.c: YV12
zbar/convert.c: YVU9
zbar/convert.c: YVYU
zbar/image.h: GREY
zbar/image.h: Y800
zbar/img_scanner.c: GREY
zbar/img_scanner.c: Y800
zbar/processor.c: Y800
zbar/sqcode.c: Y800
zbar/video.c: Y800
zbar/video/dshow.c: BGR3
zbar/video/dshow.c: BGR4
zbar/video/dshow.c: MJPG
zbar/video/v4l1.c: 411P
zbar/video/v4l1.c: 422P
zbar/video/v4l1.c: BGR3
zbar/video/v4l1.c: BGR4
zbar/video/v4l1.c: GREY
zbar/video/v4l1.c: HI24
zbar/video/v4l1.c: RGBO
zbar/video/v4l1.c: RGBP
zbar/video/v4l1.c: UYVY
zbar/video/v4l1.c: Y41P
zbar/video/v4l1.c: YU12
zbar/video/v4l1.c: YUV9
zbar/video/v4l1.c: YUY2
zbar/video/v4l1.c: YUYV
zbar/video/vfw.c: BGR3
zbar/video/vfw.c: BGR4
zbar/video/vfw.c: GREY
zbar/video/vfw.c: I420
zbar/video/vfw.c: JPEG
zbar/video/vfw.c: NV12
zbar/video/vfw.c: UYVY
zbar/video/vfw.c: Y800
zbar/video/vfw.c: YUY2
zbar/video/vfw.c: YV12
zbar/video/vfw.c: YVU9
zbar/window/dib.c: BGR3
zbar/window/dib.c: BGR4
zbar/window/dib.c: JPEG
zbar/window/vfw.c: BGR3
zbar/window/vfw.c: BGR4
zbar/window/vfw.c: JPEG
zbar/window/win.c: BGR3
zbar/window/win.c: BGR4
zbar/window/win.c: JPEG
zbar/window/ximage.c: BGR1
zbar/window/ximage.c: BGR3
zbar/window/ximage.c: BGR4
zbar/window/ximage.c: RGB1
zbar/window/ximage.c: RGB3
zbar/window/ximage.c: RGB4
zbar/window/ximage.c: RGBO
zbar/window/ximage.c: RGBQ
zbar/window/ximage.c: Y800
zbarimg/zbarimg.c: Y800
```

wqh17101, 2021-01-25:
[image]
when using zbar_image_scanner , "GRAY" and "Y800" are the format supported only.

mchehab, 2021-01-25:
> when using zbar_image_scanner , "GRAY" and "Y800" are the format supported only.

It is not that simple. The image is converted to a gray format before zbar_scan_image is called. On Linux, libv4l does such conversions internally.

ZBar itself has its own conversion code. See zbar_negotiate_format(). The implementation is inside zbar/convert.c, and this is called at zbar/processor.c.

closed 2021-02-15 by mchehab