mchehab/zbar #131: Want to merge my modify?

ssrlive, 2020-12-16:
Hi @mchehab, 
I have get tons of modify to fit with C90 and VS 2010,
Do you think it is necessary to merge into the master branch?

wqh17101, 2021-01-26:
@ssrlive Yeah. we will merge it when we are free. Thank you! Maybe you can adjust your directories structure in the branch to merge for less aggressive modification like the VS properties file.
Also https://github.com/dani4/ZBarWin64/tree/master/zbar and https://github.com/NaturalHistoryMuseum/ZBarWin64 provide the windows support.  Also `libiconv`  is necessary for the Windows.
@mchehab

ssrlive, 2021-01-26:
Hi @wqh17101 and @mchehab , please see #137 . The C code has committed.

mchehab, 2021-01-26:
@wqh17101 could you please improve patch descriptions, and add a SoB, as described at:
   https://www.linuxtv.org/wiki/index.php/Development:_Submitting_Patches#Developer.27s_Certificate_of_Origin_1.1

When ready, please send us a Pull Request.

wqh17101, 2021-01-26:
@ssrlive could you please improve patch descriptions, and add a SoB, as described at:
https://www.linuxtv.org/wiki/index.php/Development:_Submitting_Patches#Developer.27s_Certificate_of_Origin_1.1

When ready, please send us a Pull Request.

ssrlive, 2021-01-26:
I don't know how to do it. Add the lines to the comment?

ssrlive, 2021-01-26:
Done. It's OK? #137

mchehab, 2021-01-26:
> I don't know how to do it. Add the lines to the comment?

Just add:

See, for instance:

https://github.com/mchehab/zbar/commit/ab66c2c8821357321df2c2326b1543f1f2a30325

closed 2021-01-26 by ssrlive