mchehab/zbar #141: update setup.py for bug fixing

wqh17101, 2021-02-01:
we should update setup.py as below to support building .whl
```
#!/usr/bin/env python
from setuptools import setup
from distutils.extension import Extension

setup(
    name = 'zbar',
    version = '0.23.1',
    author = 'Jeff Brown',
    author_email = 'spadix@users.sourceforge.net',
    url = 'https://github.com/mchehab/zbar',
    description = 'read barcodes from images or video',
    license = 'LGPL',
    long_description = open('README').read(),
    classifiers = [
        'License :: OSI Approved :: GNU Library or Lesser General Public License (LGPL)',
        'Development Status :: 4 - Beta',
        'Intended Audience :: Developers',
        'Environment :: Console',
        'Environment :: X11 Applications',
        'Environment :: Win32 (MS Windows)',
        'Operating System :: POSIX',
        'Operating System :: Unix',
        'Operating System :: Microsoft :: Windows',
        'Topic :: Communications',
        'Topic :: Multimedia :: Graphics',
        'Topic :: Software Development :: Libraries',
    ],
    ext_modules = [
        Extension('zbar', [
                'zbarmodule.c',
                'enum.c',
                'exception.c',
                'symbol.c',
                'symbolset.c',
                'symboliter.c',
                'image.c',
                'processor.c',
                'imagescanner.c',
                'decoder.c',
                'scanner.c',
                ],
            libraries = [ 'zbar' ],
            include_dirs = ['../include']
        ),
    ],
)

```

wqh17101, 2021-02-03:
First i want to build the wheel with `libzbar` so that we can upload it to pypi and easy to install.
Second, using a pythonic way to wrap zbar maybe better such as  what pyzbar did.

wqh17101, 2021-02-03:
for the first thing, we can modify the setup.py like below
```
#!/usr/bin/env python
# -*- coding: utf-8 -*-
import os

from setuptools import Extension
from setuptools import find_packages
from setuptools import setup

CUR_DIR = os.path.relpath(os.path.dirname(__file__))
ZBAR_SO = "libzbar.so"
ZBAR_SO_PATH = os.path.join(CUR_DIR, ZBAR_SO)
with_so = os.path.exists(ZBAR_SO_PATH)
setup(
    name='zbar',
    version='0.23.1',
    author='Jeff Brown',
    author_email='spadix@users.sourceforge.net',
    url='https://github.com/mchehab/zbar',
    description='read barcodes from images or video',
    license='LGPL',
    long_description=open('README').read(),
    classifiers=[
        'License :: OSI Approved :: GNU Library or Lesser General Public License (LGPL)',
        'Development Status :: 4 - Beta',
        'Intended Audience :: Developers',
        'Environment :: Console',
        'Environment :: X11 Applications',
        'Environment :: Win32 (MS Windows)',
        'Operating System :: POSIX',
        'Operating System :: Unix',
        'Operating System :: Microsoft :: Windows',
        'Topic :: Communications',
        'Topic :: Multimedia :: Graphics',
        'Topic :: Software Development :: Libraries',
    ],
    packages=find_packages() if with_so else [],
    ext_modules=[
        Extension('zbar', [
            'zbarmodule.c',
            'enum.c',
            'exception.c',
            'symbol.c',
            'symbolset.c',
            'symboliter.c',
            'image.c',
            'processor.c',
            'imagescanner.c',
            'decoder.c',
            'scanner.c',
        ],
                  libraries=['zbar'],
                  include_dirs=['../include']
                  ),
    ],
    data_files=[("lib", [ZBAR_SO_PATH])] if with_so else []
)
```
And then we can move the .so here and rename to libzbar.so if we want to build a wheel with .so to migrate conveniently.
After that we meet a problem that how can python find the `libzbar.so` in the wheel.
so that i make a init module to load it  explicitly. like below
```
#!/usr/bin/env python
# -*- coding: utf-8 -*-
import os
from ctypes import CDLL

cur_dir = os.path.realpath(os.path.dirname(__file__))
lib_path = os.path.join(cur_dir, "../lib")
zbar_so = os.path.join(lib_path, "libzbar.so")
print("hello")
try:
    import zbar
except OSError as err:
    print("warning : {}".format(err))
    print("try to load {} ".format(zbar_so))
    CDLL(zbar_so)
    import zbar
```

if you have `libzar` in your environment, you can use
```
import zbar
print(zbar.version())
```
to check whether it works.

if you don`t have libzar ,and install a  wheel with libzbar.so
you can use
```
import zbar_init as zbar
print(zbar.version())
```
to check

mchehab, 2021-02-15:
For this to work, we need to first define how a zbar-lite will work properly:

1. a new `configure.ac` option is needed in order to enable/disable such feature;
2. a new test to be run with `make tests` should test the functionality provided by the new python modules;
3. it doesn't make sense to copy source files. Instead, `./configure --with-zbar-lite` should be setting the makefiles to properly build it;
4. any new functions that may be needed for this to work should be inside the already-existing ZBar libraries, as we don't want the same symbols to be added on both `libzbar` and on a `libzbar-lite` that would contain a sub/superset of `libzbar`;
5. whatever is needed for `setup.py`, it should never include `config.h`, `Makefile` or any other symbol produced by `autotools`, as this would be  a mess to maintain and probably useless for the user, as their enviroment will be different than the ones used by developers;
6. Let's try to use the same version number for zbar-lite, as keeping a different one for it is a maintainance nightmare.

wqh17101, 2021-02-15:
6. The version number of zbar-lite is a software number,and libzbar version is a core version number. These are two differnt things. Whenever i add some new feature by Python to zbar-lite, it should not update the core version for core code of zbar is not changed. In the other hand, if java add some new feature, i don`t think my zbar-lite version should be changed, because nothing changed for Python