mchehab/zbar #142: Complie zbar error

bestyuan, 2021-02-01:
Hi,

I'm running ./configure encountered some problem.

`
checking for strcasecmp... yes
checking for strchr... yes
checking for strdup... yes
checking for strerror... yes
checking for strrchr... yes
checking for strstr... yes
checking for strtol... yes
checking for strtoul... yes
checking for malloc... yes
checking for realloc... yes
Generating config files
checking that generated files are newer than configure... done
configure: creating ./config.status
.in'ig.status: error: cannot find input file: `
`
Best Regards,
bestyuan

wqh17101, 2021-02-04:
@bestyuan you should provide your system info

wqh17101, 2021-02-05:
you can try to follow my process in the latest issue.
For the most common conditions, you don't need all configure options. 
Also check your env , if the dependcy your option needed can be found there

发自我的iPhone

  

 my system info :
 windows subsystem for linux + Ubuntu 20.04

mchehab, 2021-02-15:
There were several issues at `configure.ac` that got fixed on release https://github.com/mchehab/zbar/releases/tag/0.23.91. Please test. If the issues keep happening, feel free to re-open, providing a full log of the `./configure` output.

closed 2021-02-15 by mchehab