mchehab/zbar #144: win64 build process

wqh17101, 2021-02-05:
first install MSYS2
then 
```
pacman -Syu --noconfirm autoconf libtool automake make autoconf-archive pkg-config
pacman -S base-devel
cd zbar
./configure  --host=x86_64-pc-msys --without-java --without-gtk --without-qt  --without-imagemagick  --disable-video --without-python
make
make install
```
and get a msys-zbar-0.dll in /usr/bin

then build python
```
pacman -S python-pip
pacman -S python-setuptools
pip install wheel
pacman -S libcrypt libcrypt-devel
cd python
python setup_plus.py bdist_wheel
```
then i can use zbar normally in the MSYS2 env not the win64 env.
So i decide to build it in mingw64 instead of MSYS2 env.

wqh17101, 2021-02-05:
open 
[image]

```
pacman -S mingw-w64-x86_64-toolchain
cd zbar
autoreconf -vfi
./configure  --host=x86_64-w64-mingw32 --without-java --without-gtk --without-qt  --without-imagemagick  --disable-video --without-python
make
make install
```

```

wqh17101, 2021-02-05:
to set host
you can use `gcc -v` to see what host you should use
[image]

wqh17101, 2021-02-07:
i am a Python user so that i used setup.py to build zbar.
Most problems i met is that Windows doesn`t have some .h which can be found only in the Linux. So i made a zbar-lite involving the minium denpendency only support scanning image for Python can be anything else such as read image or video process on its own and .
i make a preparation.sh
```
mkdir -p src/zbar/video
cp ../zbar/video/ v4l.c src/zbar/video
cp -r ../zbar/decoder src/zbar/
cp -r ../zbar/qrcode src/zbar/
mkdir zbar
cp ../python/*.c zbar
cp ../python/*.h zbar
```
and a setup.py like below
```
#!/usr/bin/env python
# -*- coding: utf-8 -*-
import os

try:
    import setuptools
    from setuptools import setup, Extension

    setuptools_opts = dict()
except ImportError:
    from distutils.core import setup
    from distutils.extension import Extension

    setuptools_opts = {}

import os
import ctypes
import ctypes.util

def get_c_file(target_dir):
    return [os.path.join(target_dir, x) for x in os.listdir(target_dir) if x.endswith(".c")]

SRCS = get_c_file("src/zbar") + \
       get_c_file("src/zbar/decoder") + \
       get_c_file("src/zbar/video") + \
       get_c_file("src/zbar/qrcode") + get_c_file("zbar")
SRCS.remove(os.path.join("zbar", "processor.c"))

INCLUDE = ['./src', './src/zbar', "./src/zbar/decoder", "./src/zbar/qrcode", "./zbar"]

def has_libc_iconv():
    if os.name != 'posix':
        return False
    libc = ctypes.CDLL(ctypes.util.find_library('c'))
    return hasattr(libc, 'iconv')

# don't try to link to standalone iconv library if it's already in libc
# (iconv is in glibc, but on OS X one needs a stanalone libiconv)
LIBS = [] if has_libc_iconv() else ['iconv']
zbar = Extension('zbar', sources=SRCS, include_dirs=INCLUDE, libraries=LIBS)

setup(name='zbar-lite',
      version='0.23.1',
      description='zbar package',
      url='https://github.com/mchehab/zbar',
      author='Qinghua Wang',
      author_email='597935261@qq.com',
      ext_modules=[zbar],
      # packages=['zbar'],
      license='MIT',
      **setuptools_opts)
```

also add a Macro  ZBAR_LITE to zbarmodule.h and add controller like below
```
/*------------------------------------------------------------------------
[... 546 lines cut ...]
    const void *zobj = ((zbarProcessor*)self)->zproc;
    zbar_error_t err = _zbar_get_error_code(zobj);

    struct module_state *st = GETMODSTATE();

    if(err == ZBAR_ERR_NOMEM)
        PyErr_NoMemory();
    else if(err < ZBAR_ERR_NUM) {
        PyObject *type = st->zbar_exc[err];
        assert(type);
        PyErr_SetObject(type, self);
    }
    else
        PyErr_SetObject(st->zbar_exc[0], self);
    return(NULL);
}
#endif

```

you only need to do below
```
sh preparation.sh
python setup.py install
```

wqh17101, 2021-02-07:
```

G:\msys64\mingw64\bin\gcc.exe -shared -s build\temp.win-amd64-3.8\Release\src\zb ar\config.o build\temp.win-amd64-3.8\Release\src\zbar\convert.o build\temp.win-a md64-3.8\Release\src\zbar\decoder.o build\temp.win-amd64-3.8\Release\src\zbar\er ror.o build\temp.win-amd64-3.8\Release\src\zbar\image.o build\temp.win-amd64-3.8 \Release\src\zbar\img_scanner.o build\temp.win-amd64-3.8\Release\src\zbar\refcnt .o build\temp.win-amd64-3.8\Release\src\zbar\scanner.o build\temp.win-amd64-3.8\ Release\src\zbar\sqcode.o build\temp.win-amd64-3.8\Release\src\zbar\symbol.o bui ld\temp.win-amd64-3.8\Release\src\zbar\video.o build\temp.win-amd64-3.8\Release\ src\zbar\decoder\codabar.o build\temp.win-amd64-3.8\Release\src\zbar\decoder\cod e128.o build\temp.win-amd64-3.8\Release\src\zbar\decoder\code39.o build\temp.win -amd64-3.8\Release\src\zbar\decoder\code93.o build\temp.win-amd64-3.8\Release\sr c\zbar\decoder\databar.o build\temp.win-amd64-3.8\Release\src\zbar\decoder\ean.o  build\temp.win-amd64-3.8\Release\src\zbar\decoder\i25.o build\temp.win-amd64-3. 8\Release\src\zbar\decoder\pdf417.o build\temp.win-amd64-3.8\Release\src\zbar\de coder\qr_finder.o build\temp.win-amd64-3.8\Release\src\zbar\decoder\sq_finder.o  build\temp.win-amd64-3.8\Release\src\zbar\video\v4l.o build\temp.win-amd64-3.8\R elease\src\zbar\qrcode\bch15_5.o build\temp.win-amd64-3.8\Release\src\zbar\qrcod e\binarize.o build\temp.win-amd64-3.8\Release\src\zbar\qrcode\isaac.o build\temp .win-amd64-3.8\Release\src\zbar\qrcode\qrdec.o build\temp.win-amd64-3.8\Release\ src\zbar\qrcode\qrdectxt.o build\temp.win-amd64-3.8\Release\src\zbar\qrcode\rs.o  build\temp.win-amd64-3.8\Release\src\zbar\qrcode\util.o build\temp.win-amd64-3. 8\Release\zbar\decoder.o build\temp.win-amd64-3.8\Release\zbar\enum.o build\temp .win-amd64-3.8\Release\zbar\exception.o build\temp.win-amd64-3.8\Release\zbar\im age.o build\temp.win-amd64-3.8\Release\zbar\imagescanner.o build\temp.win-amd64- 3.8\Release\zbar\scanner.o build\temp.win-amd64-3.8\Release\zbar\symbol.o build\ temp.win-amd64-3.8\Release\zbar\symboliter.o build\temp.win-amd64-3.8\Release\zb ar\symbolset.o build\temp.win-amd64-3.8\Release\zbar\zbarmodule.o build\temp.win -amd64-3.8\Release\src\zbar\zbar.cp38-win_amd64.def -LC:\Users\wqh\anaconda3\lib s -LC:\Users\wqh\anaconda3\PCbuild\amd64 -liconv -lpython38 -lmsvcr140 -o build\ lib.win-amd64-3.8\zbar.cp38-win_amd64.pyd
```
failed

wqh17101, 2021-02-08:
success
```
D:\mingw64\bin\gcc.exe -shared -s build\temp.win-amd64-3.7\Release\src\zbar\config.o build\temp.win-amd64-3.7\Release\src\zbar\convert.o build\temp.win-amd64-3.7\Release\src\zbar\decoder.o build\temp.win-amd64-3.7\Release\src\zbar\error.o build\temp.win-amd64-3.7\Release\src\zbar\image.o build\temp.win-amd64-3.7\Release\src\zbar\img_scanner.o build\temp.win-amd64-3.7\Release\src\zbar\refcnt.o build\temp.win-amd64-3.7\Release\src\zbar\scanner.o build\temp.win-amd64-3.7\Release\src\zbar\sqcode.o build\temp.win-amd64-3.7\Release\src\zbar\symbol.o build\temp.win-amd64-3.7\Release\src\zbar\video.o build\temp.win-amd64-3.7\Release\src\zbar\decoder\codabar.o build\temp.win-amd64-3.7\Release\src\zbar\decoder\code128.o build\temp.win-amd64-3.7\Release\src\zbar\decoder\code39.o build\temp.win-amd64-3.7\Release\src\zbar\decoder\code93.o build\temp.win-amd64-3.7\Release\src\zbar\decoder\databar.o build\temp.win-amd64-3.7\Release\src\zbar\decoder\ean.o build\temp.win-amd64-3.7\Release\src\zbar\decoder\i25.o build\temp.win-amd64-3.7\Release\src\zbar\decoder\pdf417.o build\temp.win-amd64-3.7\Release\src\zbar\decoder\qr_finder.o build\temp.win-amd64-3.7\Release\src\zbar\decoder\sq_finder.o build\temp.win-amd64-3.7\Release\src\zbar\video\v4l.o build\temp.win-amd64-3.7\Release\src\zbar\qrcode\bch15_5.o build\temp.win-amd64-3.7\Release\src\zbar\qrcode\binarize.o build\temp.win-amd64-3.7\Release\src\zbar\qrcode\isaac.o build\temp.win-amd64-3.7\Release\src\zbar\qrcode\qrdec.o build\temp.win-amd64-3.7\Release\src\zbar\qrcode\qrdectxt.o build\temp.win-amd64-3.7\Release\src\zbar\qrcode\rs.o build\temp.win-amd64-3.7\Release\src\zbar\qrcode\util.o build\temp.win-amd64-3.7\Release\zbar\decoder.o build\temp.win-amd64-3.7\Release\zbar\enum.o build\temp.win-amd64-3.7\Release\zbar\exception.o build\temp.win-amd64-3.7\Release\zbar\image.o build\temp.win-amd64-3.7\Release\zbar\imagescanner.o build\temp.win-amd64-3.7\Release\zbar\scanner.o build\temp.win-amd64-3.7\Release\zbar\symbol.o build\temp.win-amd64-3.7\Release\zbar\symboliter.o build\temp.win-amd64-3.7\Release\zbar\symbolset.o build\temp.win-amd64-3.7\Release\zbar\zbarmodule.o build\temp.win-amd64-3.7\Release\src\zbar\zbar.cp37-win_amd64.def -LC:\Users\Anaconda3\libs -LC:\Users\Anaconda3\PCbuild\amd64 -liconv -lpython37 -lmsvcr140 -o build\lib.win-amd64-3.7\zbar.cp37-win_amd64.pyd
```
gcc -v 
```
Using built-in specs.
COLLECT_GCC=D:\mingw64\bin\gcc.exe
COLLECT_LTO_WRAPPER=D:/mingw64/bin/../libexec/gcc/x86_64-w64-mingw32/7.3.0/lto-wrapper.exe
Target: x86_64-w64-mingw32
Configured with: ../../../src/gcc-7.3.0/configure --host=x86_64-w64-mingw32 --build=x86_64-w64-mingw32 --target=x86_64-w64-mingw32 --prefix=/mingw64 --with-sysroot=/c/mingw730/x86_64-730-posix-seh-rt_v5-rev0/mingw64 --enable-shared --enable-static --disable-multilib --enable-languages=c,c++,fortran,lto --enable-libstdcxx-time=yes --enable-threads=posix --enable-libgomp --enable-libatomic --enable-lto --enable-graphite --enable-checking=release --enable-fully-dynamic-string --enable-version-specific-runtime-libs --enable-libstdcxx-filesystem-ts=yes --disable-libstdcxx-pch --disable-libstdcxx-debug --enable-bootstrap --disable-rpath --disable-win32-registry --disable-nls --disable-werror --disable-symvers --with-gnu-as --with-gnu-ld --with-arch=nocona --with-tune=core2 --with-libiconv --with-system-zlib --with-gmp=/c/mingw730/prerequisites/x86_64-w64-mingw32-static --with-mpfr=/c/mingw730/prerequisites/x86_64-w64-mingw32-static --with-mpc=/c/mingw730/prerequisites/x86_64-w64-mingw32-static --with-isl=/c/mingw730/prerequisites/x86_64-w64-mingw32-static --with-pkgversion='x86_64-posix-seh-rev0, Built by MinGW-W64 project' --with-bugurl=https://sourceforge.net/projects/mingw-w64 CFLAGS='-O2 -pipe -fno-ident -I/c/mingw730/x86_64-730-posix-seh-rt_v5-rev0/mingw64/opt/include -I/c/mingw730/prerequisites/x86_64-zlib-static/include -I/c/mingw730/prerequisites/x86_64-w64-mingw32-static/include' CXXFLAGS='-O2 -pipe -fno-ident -I/c/mingw730/x86_64-730-posix-seh-rt_v5-rev0/mingw64/opt/include -I/c/mingw730/prerequisites/x86_64-zlib-static/include -I/c/mingw730/prerequisites/x86_64-w64-mingw32-static/include' CPPFLAGS=' -I/c/mingw730/x86_64-730-posix-seh-rt_v5-rev0/mingw64/opt/include -I/c/mingw730/prerequisites/x86_64-zlib-static/include -I/c/mingw730/prerequisites/x86_64-w64-mingw32-static/include' LDFLAGS='-pipe -fno-ident -L/c/mingw730/x86_64-730-posix-seh-rt_v5-rev0/mingw64/opt/lib -L/c/mingw730/prerequisites/x86_64-zlib-static/lib -L/c/mingw730/prerequisites/x86_64-w64-mingw32-static/lib '
Thread model: posix
gcc version 7.3.0 (x86_64-posix-seh-rev0, Built by MinGW-W64 project)
```

wqh17101, 2021-02-09:
in the setup.py you can use
```
from setuptools.command.build_ext import build_ext

class custom_build_ext(build_ext):
    def build_extensions(self):
        self.compiler.dll_libraries = []
        build_ext.build_extensions(self)
```
to make no extra libs of dll

for linux config.h  is like
```
/* include/config.h.  Generated from config.h.in by configure.  */
/* include/config.h.in.  Generated from configure.ac by autoheader.  */

/* Define to one of `_getb67', `GETB67', `getb67' for Cray-2 and Cray-YMP
   systems. This function is required for `alloca.c' support on those systems.
   */
/* #undef CRAY_STACKSEG_END */

/* Define to 1 if using `alloca.c'. */
/* #undef C_ALLOCA */

/* whether to build support for Codabar symbology */
#define ENABLE_CODABAR 1

/* whether to build support for Code 128 symbology */
#define ENABLE_CODE128 1

/* whether to build support for Code 39 symbology */
#define ENABLE_CODE39 1

/* whether to build support for Code 93 symbology */
#define ENABLE_CODE93 1

/* whether to build support for DataBar symbology */
#define ENABLE_DATABAR 1

/* whether to build support for EAN symbologies */
#define ENABLE_EAN 1

/* whether to build support for Interleaved 2 of 5 symbology */
#define ENABLE_I25 1

/* Define to 1 if translation of program messages to the user's native
   language is requested. */
#define ENABLE_NLS 1

/* whether to build support for PDF417 symbology (incomplete) */
#define ENABLE_PDF417 1 

/* whether to build support for QR Code */
#define ENABLE_QRCODE 1

/* whether to build support for SQ Code */
#define ENABLE_SQCODE 1

/* Define to 1 if you have the `alarm' function. */
#define HAVE_ALARM 1

/* Define to 1 if you have `alloca', as a function or macro. */
/* #undef HAVE_ALLOCA */

/* Define to 1 if you have <alloca.h> and it should be used (not on Ultrix).
   */
/* #undef HAVE_ALLOCA_H */

/* Define to 1 if you have the <arpa/inet.h> header file. */
/* #undef HAVE_ARPA_INET_H */
[... 392 lines cut ...]
   such a type exists and the standard includes do not define it. */
/* #undef uint16_t */

/* Define to the type of an unsigned integer type of width exactly 32 bits if
   such a type exists and the standard includes do not define it. */
/* #undef uint32_t */

/* Define to the type of an unsigned integer type of width exactly 64 bits if
   such a type exists and the standard includes do not define it. */
/* #undef uint64_t */

/* Define to the type of an unsigned integer type of width exactly 8 bits if
   such a type exists and the standard includes do not define it. */
/* #undef uint8_t */

/* Define to the type of an unsigned integer type wide enough to hold a
   pointer, if such a type exists, and if the system does not define it. */
/* #undef uintptr_t */

#ifndef X_DISPLAY_MISSING
# define HAVE_X
#endif

```
which is really different from Windows

hosiet, 2021-02-09:
I see no real value simply posting all your exploration history on GitHub.

If you only want C/C++ library on msys2/mingw platform, please use prebuilt https://packages.msys2.org/package/mingw-w64-x86_64-zbar .

If you want extra bindings of other languages (e.g., Python), please work with existing MSYS2 packagers.

(P.S. These are just my personal suggestions. I am neither zbar upstream nor msys2 packager.)

wqh17101, 2021-02-09:
> I see no real value simply posting all your exploration history on GitHub.
>
> If you only want C/C++ library on msys2/mingw platform, please use prebuilt [...]

# This is *not* a problem but my personal process to show how to build a zbar lib can be used in Windows System instead of  MSYS2 or any other subsystem.
Also if i want to use zbar via python binding . for now , i must install lib zbar first ,and then install zbar-python binding.  That is not convinient.

So i found a way to build python whl with zbar-python binding and zbar lib all in one lib. So that a python users can just install once to use zbar. Also for python`s flexibility , i just want to support the basic functions  of zbar---scanning and recognization images to python`s user.

mchehab, 2021-02-15:
I'll close it, as this is not really an issue.

Also, we have already an actions workflow to build it using Github's infra. The produced patches are both at the Zbar releases page (https://github.com/mchehab/zbar/releases), where the built files will be retained from 90 days (the maximum policy at Github) and at:

https://linuxtv.org/downloads/zbar/binaries/

closed 2021-02-15 by mchehab