mchehab/zbar #146: Build error: no man-pages build.

sassy-crick, 2021-02-09:
I am trying to build the latest zbar-0.23.1 with GCC-10.2.0. I am using:

`$ ./configure --with-gtk=auto --with-python=python3 --prefix=/foo/baa/zbar-0.23.1`

The configure step works well:

please verify that the detected configuration matches your expectations:

------------------------------------------------------------------------
gettext                yes
X                      --with-x=yes
pthreads               --enable-pthread=yes
doc                    --enable-doc=yes
v4l                    --enable-video=yes
jpeg                   --with-jpeg=yes
Python                 --with-python=python3    python3.8
GTK                    --with-gtk=gtk3          Gtk3.24.23
GObject introspection  --with-gir=no
Qt                     --with-qt=no         
Java                   --with-java=yes
Dbus                   --with-dbus=yes
ImageMagick            --with-imagemagick=check
Enabled codes:         ean databar code128 code93 code39 codabar i25 qrcode sqcode
Disabled codes:        pdf417
JAVA_HOME              /opt/software/build/software/Java/11.0.2

        => libv4l will *NOT* be used
        => the Qt widget will *NOT* be built
        => the Java unit test will *NOT* be enabled

------------------------------------------------------------------------

However, `make install` fails here:

------------------------------------------------------------------------

 /usr/bin/install -c -m 644 include/zbar.h '/tmp/zbar-0.23.1/include'
make[4]: Entering directory '/home/sassy/easybuild/OTPClient/zbar/zbar-0.23.1'
: --searchpath /home/sassy/easybuild/OTPClient/zbar/zbar-0.23.1/doc -m /home/sassy/easybuild/OTPClient/zbar/zbar-0.23.1/doc/style.xsl --skip-validation  -o doc/man man doc/manual.xml
make[4]: Leaving directory '/home/sassy/easybuild/OTPClient/zbar/zbar-0.23.1'
make[4]: Entering directory '/home/sassy/easybuild/OTPClient/zbar/zbar-0.23.1'
: --searchpath /home/sassy/easybuild/OTPClient/zbar/zbar-0.23.1/doc -m /home/sassy/easybuild/OTPClient/zbar/zbar-0.23.1/doc/style.xsl --skip-validation  -o doc/man man doc/manual.xml
make[4]: Leaving directory '/home/sassy/easybuild/OTPClient/zbar/zbar-0.23.1'
 /bin/mkdir -p '/tmp/zbar-0.23.1/share/man/man1'
 /usr/bin/install -c -m 644 ./doc/man/zbarimg.1 ./doc/man/zbarcam.1 '/tmp/zbar-0.23.1/share/man/man1'
/usr/bin/install: cannot stat './doc/man/zbarimg.1': No such file or directory
/usr/bin/install: cannot stat './doc/man/zbarcam.1': No such file or directory
make[3]: *** [Makefile:1722: install-man1] Error 1
make[3]: Leaving directory '/home/sassy/easybuild/OTPClient/zbar/zbar-0.23.1'
make[2]: *** [Makefile:2209: install-am] Error 2
make[2]: Leaving directory '/home/sassy/easybuild/OTPClient/zbar/zbar-0.23.1'
make[1]: *** [Makefile:1897: install-recursive] Error 1
make[1]: Leaving directory '/home/sassy/easybuild/OTPClient/zbar/zbar-0.23.1'
make: *** [Makefile:2203: install] Error 2

------------------------------------------------------------------------

(ignore I am trying to install it in /tmp/zbar-0.23.1 this is for debugging only)

Any idea where I get it wrong?

Thanks

sassy-crick, 2021-02-10:
Update: the problem was a missing system `xmlto`. However, one would have thought as `configure` is checking for that, and if it is not installed the man-pages will simply not being build, rather then trying to build them and then failing when `make install` is executed.

berkaytuna, 2021-02-10:
thanks sassy,

sudo apt-get install xmlto
./configure
make
sudo make install

solved this for me, as well.

sassy-crick, 2021-02-11:
Hi @berkaytuna

yes, thanks. 
What I am after is that `config` should pick that up and not try to install the man-pages and then fails. 
I wrote an EasyBuild Configuration (EC) file for that and this is where I tripped over that.
That is what I am after in the long run. :-) 

Jörg

mchehab, 2021-02-11:
> What I am after is that `config` should pick that up and not try to install the man-pages and then fails.

Fair enough. I'm working on a patchset to improve the configure script to better handle things like this.

sassy-crick, 2021-02-11:
Happy to try the patch and update the EC once it is done. Just let me know. 
Thanks for your help.

mchehab, 2021-02-11:
> Happy to try the patch and update the EC once it is done. Just let me know.
> Thanks for your help.

Patches applied. Please test.

closed 2021-02-15 by mchehab