mchehab/zbar #161: ./configure: ${JAVAC/...}: Bad substitution

VVD, 2021-02-20:
Error during configure stage, part of buildlog:
```
checking for MAGICK... yes
configure: trying ImageMagick version 7.0.10
checking wand/MagickWand.h usability... no
checking wand/MagickWand.h presence... no
checking for wand/MagickWand.h... no
checking MagickWand/MagickWand.h usability... yes
checking MagickWand/MagickWand.h presence... yes
checking for MagickWand/MagickWand.h... yes
configure: using ImageMagick version 7.0.10
checking for javac... /usr/local/bin/javac
./configure: ${JAVAC/...}: Bad substitution
===>  Script "configure" failed unexpectedly.
```
./configure --without-dbus --without-gir --without-java --with-jpeg --enable-nls --without-npapi --without-python --without-qt --without-qt5 --disable-doc --with-gtk=no --with-imagemagick --without-graphicsmagick --enable-video --with-x --with-xv --with-xshm --x-libraries=/usr/local/lib --x-includes=/usr/local/include --prefix=/usr/local --localstatedir=/var --mandir=/usr/local/man --disable-silent-rules --infodir=/usr/local/share/info/ --build=amd64-portbld-freebsd12.2

FreeBSD 12.2 amd64.

VVD, 2021-02-20:
This patch fixed configure stage:
```
--- configure.ac.orig
+++ configure.ac
@@ -728,7 +728,7 @@
   [],
   [with_java="check"])
 
-JAVAC=${JAVAC/ecj/ecj -1.5}
+JAVAC=${JAVAC}/ecj/ecj-1.5
 
 dnl Javah was obsoleted on Java 8 and removed on Java 11. So, we need to
 dnl look strictly at the $JAVA_HOME in order to avoid mixing different versions
```

VVD, 2021-02-20:
Ofc it's incorrect patch, but it fixed configure for me.

mchehab, 2021-02-21:
Hmm... I suspect that the issue is that this string substitution can be bash-specific.

One alternative would be to re-write it as:

```
JAVAC=$(echo $JAVAC |sed 's/ecj/ecj -1.5/');
```

but I'm not sure if sed is present on all platforms

mchehab, 2021-02-21:
Fix applied. it should use `sed` command for string replacement.

closed 2021-02-21 by mchehab

VVD, 2021-02-21:
Work fine! Thanks!