mchehab/zbar #167: Swiss QR not readable before removing the center square image

maglub, 2021-02-26:
Thanks for an amazing tool!

I have one example of a non readable qrcode, a Swiss QR Bill code. These are used on bills in Switzerland, and can be generated here: https://qrportal.ch/one#paymentForm

I found a not so practical workaround in https://github.com/mchehab/zbar/issues/65, meaning that I would edit the QR code to be able to read it.

These are following the ISO standard, and are valid QR code.

[image]

Bad, non working image:

[image]

```
(venv) [maglub@Magnuss-MacBook-Pro-2:~/dev/wue/dani]$ zbarimg magnus2.jpg 
scanned 0 barcode symbols from 1 images in 0.02 seconds

WARNING: barcode data was not detected in some image(s)
Things to check:
  - is the barcode type supported? Currently supported symbologies are:
	. EAN/UPC (EAN-13, EAN-8, EAN-2, EAN-5, UPC-A, UPC-E, ISBN-10, ISBN-13)
	. DataBar, DataBar Expanded
	. Code 128
	. Code 93
	. Code 39
	. Codabar
	. Interleaved 2 of 5
	. QR code
	. SQ code
  - is the barcode large enough in the image?
  - is the barcode mostly in focus?
  - is there sufficient contrast/illumination?
  - If the symbol is split in several barcodes, are they combined in one image?
  - Did you enable the barcode type?
    some EAN/UPC codes are disabled by default. To enable all, use:
    $ zbarimg -S*.enable <files>
    Please also notice that some variants take precedence over others.
    Due to that, if you want, for example, ISBN-10, you should do:
    $ zbarimg -Sisbn10.enable <files>
```

When I removed the "swiss" sign in the middle of the QR code, zbarimg can read it.

Working image:

[image]

```
(venv) [maglub@Magnuss-MacBook-Pro-2:~/dev/wue/dani]$ zbarimg -q removed_swiss_mark.png 
QR-Code:SPC
0200
1
CH760000000000000
S
Some person
Somewhere
10
1234
Swisstown
CH

1234
CHF
S
Other person
Somewhere
11
9876
Swisstropolis
CH
NON

Please send money
EPD
```

The QR code on page 11 in the specification works:

[image]

But the QR code on page 42 does not work:

[image]

```
(venv) [maglub@Magnuss-MacBook-Pro-2:~/dev/wue/dani]$ zbarimg -q page11.png 
QR-Code:QRType
Version
Coding Type
Konto (IBAN ZE)
Adresstyp (ZE)
Name (ZE)
Strasse oder Adresszeile 1 (ZE)
Hausnr. oder Adresszeile 2 (ZE)
PLZ (ZE)
Ort (ZE)
Land (ZE)
Adresstyp (EZE)
Name (EZE)
Strasse oder Adresszeile 1 (EZE)
Hausnr. oder Adresszeile 2 (EZE)
PLZ (EZE)
Ort (EZE)
Land (EZE)
Betrag
Währung
Adresstyp (EZP)
Name (EZP)
Strasse oder Adresszeile 1 (EZP)
Hausnr. oder Adresszeile 2 (EZP)
PLZ (EZP)
Ort (EZP)
Land (EZP)
Referenztyp
Referenz
Unstrukturierte Mitteilungen
Trailer
Rechnungsinformationen
AV1 – Parameter
AV2 – Parameter
(venv) [maglub@Magnuss-MacBook-Pro-2:~/dev/wue/dani]$ zbarimg -q page42.png 
(venv) [maglub@Magnuss-MacBook-Pro-2:~/dev/wue/dani]$ 
(venv) [maglub@Magnuss-MacBook-Pro-2:~/dev/wue/dani]$ echo $?
4
```

# References

* Standards document (the QR code on page 42 can be read on my iPhone, but not with zbarimg): https://www.paymentstandards.ch/dam/downloads/ig-qr-bill-de.pdf

ionum, 2021-06-26:
Same Issue #121. It seems that recognition fails if the swiss cross bars have similar height/width as qrcode-squares.

maslick, 2022-01-30:
Hej @maglub! I was able to decode all of the images you provided with my OSS project named [Koder](https://github.com/maslick/koder). It is based on zbar 0.23.90 (compiled into WebAssembly). You can check out the demo here: https://qr.maslick.tech

ionum, 2022-02-01:
But it works not on commandline eg "zbarimg pic.png". Tested with lastest version from here (0.23.92).

izidormatusov, 2022-07-04:
It seems to have something to do with QR code error correction:

```
$ zbarimg -v -q qrbill.png
qr_code_decode: Number of errors corrected: -1 (data irrecoverable)
qr_code_decode: Number of errors corrected: -1 (data irrecoverable)
dump_stats: symbol sets allocated   = 1
dump_stats:     scanner syms in use = 0   	recycled  = 0
dump_stats:     image syms in use   = 0   	recycled  = 0
dump_stats: symbols allocated       = 0
dump_stats:      recycled[0]        = 0
dump_stats:      recycled[1]        = 0
dump_stats:      recycled[2]        = 0
dump_stats:      recycled[3]        = 0
dump_stats:      recycled[4]        = 0
_zbar_qr_destroy: max finder lines = 127x127
```

SemGa92, 2023-02-09:
Hi, thank you for the great work.
I have the same problem tested on Ubuntu 22.04 and 0.23.92 zbar version.

Is there any news regarding this issue?