mchehab/zbar #173: QR code can't be detected

xuhdev, 2021-04-08:
zbar reports `WARNING: barcode data was not detected in some image(s)` with the following image:

[image]

But I guess it should be a QR code?

wqh17101, 2021-04-09:
you should invert the color first in your preprocessing. and you can get the result.
before invert i got nothing.
i got below after i inverted the color
`decoded QRCODE symbol "https://developer.ibm.com/exchanges/data/all/airline/"`

xuhdev, 2021-04-12:
Hmm, thanks. Do you think the image was made improperly, or this is a bug in zbar for not being able to decode?

wqh17101, 2021-04-12:
As you can see after comparing the ori one with the inverted one, zbar can not found the bound of the ori one.
[image]
It is white.

xuhdev, 2021-04-12:
Got it. I tried to run zbar again with a black background this time:

    zbarimg '-S*.enable' black-background.png

However, I still got the same error, for both of the following images:

[image]

[image]

George-Miles, 2021-04-12:
I just invert the image and it works fine.

wqh17101, 2021-04-13:
> Got it. I tried to run zbar again with a black background this time:

So why did you think that inverting the img is the same as setting the background black ?
here is some info for QR CODE , and you can read it to learn more about QRCODE.
As i metioned before, inverting can work fine.
[image]

As we all known that QR Code uses color to store info, and for the default , we can assume that *black or other dark colors* for '1' and *white* for '0'. So after inverted, it means different thing. For example , you will get 0 after inverting 1 not 1 itself.

In a word,  your original picture uses the inverted setting which is **made improperly**. So we can not decode by default setting, and then trying to invert is a good idea.

wqh17101, 2021-04-13:
When the background is darker than the QR Code's foreground, it's called an inverted Code. These types of Codes typically have a dark background such as black, navy or dark grey. While a few scanners can read an inverted Code, some apps are not able to scan them.

xuhdev, 2021-04-13:
I tried the black-backgrounded version because you said the reason it could not be recognized is

> As you can see after comparing the ori one with the inverted one, zbar can not found the bound of the ori one.

It's just good to be sure that zbar can't recognize inverted QR code. Consider this as a feature request to recognize an inverted QR code and feel free to close if this is out of scope :)

diegodorado, 2024-08-18:
I run into this issue, and appearently there is support for inverted images now, only not enabled by default (not sure why)

run it with `zbarimg -Stest-inverted`

kaurgurpreet002101-jpg, 2026-03-31:
Nsbjdjakmsmlwm

shabraizail, 2026-05-04:
https://d1v3egekkfybx4.cloudfront.net/?dl=5f3c3q