mchehab/zbar #179: Please add the wayland support for zbarcam

telychko, 2021-05-08:
Please add the wayland support for zbarcam application. Currently the application works with "--nodisplay" option only. It successfully scans the codes but that really hard without seeing display window.

$ zbarcam --version
0.23.1

$ LANG= zbarcam -v
_zbar_video_open: opened camera device /dev/video0 (fd=5)
_zbar_v4l2_probe: Integrated Camera on usb-0000:00:13.2-5 driver uvcvideo (version 5.11.14)
_zbar_v4l2_probe:     capabilities: CAPTURE READWRITE STREAMING
v4l2_reset_crop: crop bounds: 1280 x 720 @ (0, 0)
v4l2_reset_crop: current crop win: 1280 x 720 @ (0, 0) aspect 1 / 1
v4l2_probe_formats: Max supported size: 1280 x 720
v4l2_probe_formats: current format: YUYV(56595559) 1280 x 720 (line=0xa00 size=0x1c2000)
v4l2_query_controls: Control class User:
v4l2_query_controls: int        Brightness                       - 0x980900
v4l2_query_controls: int        Contrast                         - 0x980901
v4l2_query_controls: int        Saturation                       - 0x980902
v4l2_query_controls: int        Hue                              - 0x980903
v4l2_query_controls: bool       White Balance Temperature, Auto  - 0x98090c
v4l2_query_controls: int        Gamma                            - 0x980910
v4l2_query_controls: menu       Power Line Frequency             - 0x980918
v4l2_query_controls:            0: Disabled
v4l2_query_controls:            1: 50 Hz
v4l2_query_controls:            2: 60 Hz
v4l2_query_controls: int        White Balance Temperature        - 0x98091a
v4l2_query_controls: int        Sharpness                        - 0x98091b
v4l2_query_controls: int        Backlight Compensation           - 0x98091c
v4l2_query_controls: Control class Camera:
v4l2_query_controls: menu       Exposure, Auto                   - 0x9a0901
v4l2_query_controls:            1: Manual Mode
v4l2_query_controls:            3: Aperture Priority Mode
v4l2_query_controls: int        Exposure (Absolute)              - 0x9a0902
v4l2_query_controls: bool       Exposure, Auto Priority          - 0x9a0903
v4l2_query_controls: int        Pan (Absolute)                   - 0x9a0908
v4l2_query_controls: int        Tilt (Absolute)                  - 0x9a0909
v4l2_query_controls: int        Zoom, Absolute                   - 0x9a090d
v4l2_get_supported_resolutions: 1280x720 (7.00 fps)
v4l2_get_supported_resolutions: 640x480 (27.50 fps)
v4l2_get_supported_resolutions: 640x360 (27.50 fps)
v4l2_get_supported_resolutions: 352x288 (27.50 fps)
v4l2_get_supported_resolutions: 320x240 (27.50 fps)
v4l2_get_supported_resolutions: 176x144 (27.50 fps)
v4l2_get_supported_resolutions: 160x120 (27.50 fps)
WARNING: zbar video in v4l2_probe_iomode():
    system error: USERPTR failed. Falling back to mmap: Invalid argument (22)
_zbar_v4l2_probe: using I/O mode: MMAP
ERROR: zbar processor in _zbar_processor_open():
    X11 display error: unable to open X display
ERROR: zbar processor in zbar_processor_init():
    system error: spawning input thread: Invalid argument (22)
ERROR: zbar processor in zbar_processor_init():
    system error: spawning input thread: Invalid argument (22)

dleidert, 2021-09-21:
You can force the X11 backend on wayland: `GDK_BACKEND=x11 <command>`.

hongquan, 2022-01-07:
If you want Wayland support, you can try [CoBang](https://github.com/hongquan/CoBang), which also uses ZBar under the hood.