mchehab/zbar #182: QRs frequently not recognized (2% error rate?)

za3k, 2021-06-02:
I wrote [qr-backup](https://github.com/za3k/qr-backup/), a tool to back up files to a sequence of QR codes. It uses zbar very heavily for restore.

As the final step of my program, I run `qbarimg --raw PDF_FILE` to make sure the backup was generated correctly. I find that qbarimg is frequently missing a QR code, even though the image is 8px-per-module, crisp black and white, and digitally generated. It's probably something like a 2% error rate. I'm generating version 10 codes, with L or M error correction, and binary data.

The QRs are OK -- I can print the QR codes out using the same PDF, and it scans correctly using zbarcam.

Also, I've generally found zbarcam pretty finicky--the codes have to be at least twice the size I'd expect for them to be recognized.

If you want to test using my program directly:
- Install python 3.6+, python-pillow and python-qrcode
- Run `qr-backup -v FILENAME` on a file of about 200K (it does compression, so avoid all-zeros etc--dd from /dev/urandom or use any real file). Make sure not to run it using stdin/stdout.
- An error is printed about `!!Automatic digital restore verification FAILED` if you have reproduced.
- The PDF `FILENAME.pdf` is a repro case. Normally, it will be something like 40 QR codes, and 39 will be recognized.

Or, I can provide host and link some repro PDF files if you don't want to bother running `qr-backup` yourself.

za3k, 2021-06-05:
Attached is a repro case for convenience, a 1-page PDF with 13 codes. zbarimg detects only 12.

I can reproduce with 2K of data, but not with 1K of data (no matter how many times I run). Therefore this might have something to do with the grid layout and number of QR codes on the page?

[failure.pdf](https://github.com/mchehab/zbar/files/6602671/failure.pdf)

misa-ki, 2021-07-31:
I confirm this bug and the error rate of around 2% (something like 5 errors per 250 QR codes, so it means that I can't rely on `zbarimg` for bulk process). Here's a [PNG sample file to reproduce the issue](https://user-images.githubusercontent.com/64830812/127732474-ff7fa13e-d913-4926-8361-ace8cf14d1f8.png). The PNG image contains 16 QR codes, but `zbarimg` only sees 15 QR codes (it doesn't see the QR code 38):

```
$ zbarimg image-with-16-QR-codes.png | sort
scanned 15 barcode symbols from 1 images in 1,6 seconds

QR-Code:N33/56 6qvMOTVxoUubYs68V7S1xQLLgCsm15IzxKMgCO2xEa1O6cyIF9TwsE1evZkUGNJQRlKMlpcDUT7SBhcoAOGlvRLpTugsD+WGpjx9rpGIXGxVkizb
QR-Code:N34/56 hL5L9KSEna+corgLA8dptZamGOaBnF0JBZdpG8Z83PYCOryRZRpIkjR9j2mLRkjuJlr157kmVJCwe7G5mLIpEaWJSckY2sLgQWNztkAaqd5/0TrO
QR-Code:N35/56 R/gpfMXBUSipKtYhziX1mKk1qYYZqhaUstSY1kJsq21d6Mjc6s8G/Rae2mN+epIOmVdVRaES7QFkgqLqSsCWG4jaUWFNTrhhkm3KbFoHieQREClB
QR-Code:N36/56 CAvn4k91M5dyHoolNdsi7UPrh77RV2Kk94J2/udg4s/bAJ7+lRNvyuRwa0qdnGNoJ9HA0oDRtG0usgmzxEsxjQ4UymBXQ0iIyadi0L+cK6J/Fjg8
QR-Code:N37/56 pZRiO+Nsb+1gjCJRSmXaP3c3e7g4OuQOPSwi50htGWCHqRVkm4Hf3UaoU7x3oRZTlSZFmEsqvwhkijsvTQcPHhnhRarAXhAgudsRvW8cogiQxHtR
QR-Code:N39/56 01qanWThaIbFXEI5CZrS4iNAHgKWNCRZYErFyv0yzno/z6xOqt4PZGka1fjXOiGNv4gKjUrSKSA5qEAIUc8aFHDsyZfOTB89cD+ZsqFV+Y58oAdP
QR-Code:N40/56 dFTO3b1IyliBppijeEKL42dpg6dOnMgbaVviUj2rU1M/jS2aOSiIQ4trTAtYB3mH2xtMWLUnWxvpCwp84qzcrIJJ2D2d5kBLAnOmaO76kHByuQ4c
QR-Code:N41/56 hfIUA/qgbJqzls2a8lROQqf0w6uyr97tUQN3N3WCvXGmi/6zZpXjHdzOPdBROVuU538a06gcrGzJA1NZd1Qk0+FlqYLxS46hxsu8ZnDpRQdqdJKY
QR-Code:N42/56 9Ni5xQ+JmBOIkTStJ5+fsiX1tDyxvXcO0QZJI175mgc+Es7WiUEA5EqdDZYWpaW5QGpz3XE9b6jGxMWn+Q9asNci0tQVumd6eALMoULv5fB63Ppf
QR-Code:N43/56 Lvq9amXu0MM1It5MM6AQi5No0lJLR26UKVktZGZCrzQpnz3mo6ZOtND04UPS2y3t08RRPaIvY/FtVuKOBgxwmqLZmyKER094YtorPVwrJTlAKqPE
QR-Code:N44/56 OPEoJEooz/177ofUMRp2CNvzyIPDO0MU/dcvywWk2oRivaKRq1WnHQzCMUadz6S55i1VWPnAr7Iw29QkXpwc0tzkoxbowegR6IyMLzpLwHKiQYhi
QR-Code:N45/56 hfKs8hdSzc8bC8f6SDtzTNIM7nXrnujquY5jaRlK8Tr6FLNo9wqHB0xx2Zz85g+1ZKep0TjgtBnZXpT0FrpAgxV4Wj8oy5v68xFQtFrZ+9o9VTir
QR-Code:N46/56 qSEMcFjHj9Hc9OSVdDyjKrTTEXLBZ1RtoUhkjEv/fnc63GnuM69KM7sxmM0fVJ5YlrJDG/m8ox7K1DOcjrXodXnG5vUa9YI9qq9G5KY3aoWcUMvb
QR-Code:N47/56 GN+WYMrc2jBofSOCQM9GZN4qgNcVpnkQAsYyX0vZrKF1qXeNAnWXOnYr57BmBtGs6ScDHHk2P2QXW7Mdpx8sEtM0F2UTimacS9FUUGwCwlHoqsiy
QR-Code:N48/56 2ehekGykZ3DtME4gwDxXECnpREtUedrnkMOTVlqi5VLzbAINLwyf3+1gAsDjvom6LHJUChCDdJ75oqPmtSv+30ti01Cxo9N5MuhnhYsmQAvvJBv0
$ zbarimg --version
0.23.90
```
However, if the [QR code 38](https://user-images.githubusercontent.com/64830812/127732793-306d93bd-52b5-4f15-9d27-279073541bd9.png) is analyzed individually, `zbarimg` sees it:

```
$ zbarimg qr-code-38.png
QR-Code:N38/56 Ef7oefU/vhvPQc2jqegywM/plJOmtKrt0rGUGEb7iV4pljdJZNuDqnN2lWBPPndfuCjCHGtv58Zk8IdurC0lKXBP9D/UhkaVyI2u2VFn87o1CoD5
scanned 1 barcode symbols from 1 images in 0,08 seconds
```

misa-ki, 2021-07-31:
For the record, if I increase margin between the 16 QR codes on the sample image, `zbarimg` still sees only 15 codes, but the missed QR code is not the same (code 44 instead of code 38):

[sample image with increased margin between QR codes](https://user-images.githubusercontent.com/64830812/127733463-6833ec72-9cb7-438b-8aa4-70efca68c7d6.png)

```
$ zbarimg image-with-16-QR-codes-and-increased-margin.png | sort
scanned 15 barcode symbols from 1 images in 1,6 seconds

QR-Code:N33/56 6qvMOTVxoUubYs68V7S1xQLLgCsm15IzxKMgCO2xEa1O6cyIF9TwsE1evZkUGNJQRlKMlpcDUT7SBhcoAOGlvRLpTugsD+WGpjx9rpGIXGxVkizb
QR-Code:N34/56 hL5L9KSEna+corgLA8dptZamGOaBnF0JBZdpG8Z83PYCOryRZRpIkjR9j2mLRkjuJlr157kmVJCwe7G5mLIpEaWJSckY2sLgQWNztkAaqd5/0TrO
QR-Code:N35/56 R/gpfMXBUSipKtYhziX1mKk1qYYZqhaUstSY1kJsq21d6Mjc6s8G/Rae2mN+epIOmVdVRaES7QFkgqLqSsCWG4jaUWFNTrhhkm3KbFoHieQREClB
QR-Code:N36/56 CAvn4k91M5dyHoolNdsi7UPrh77RV2Kk94J2/udg4s/bAJ7+lRNvyuRwa0qdnGNoJ9HA0oDRtG0usgmzxEsxjQ4UymBXQ0iIyadi0L+cK6J/Fjg8
QR-Code:N37/56 pZRiO+Nsb+1gjCJRSmXaP3c3e7g4OuQOPSwi50htGWCHqRVkm4Hf3UaoU7x3oRZTlSZFmEsqvwhkijsvTQcPHhnhRarAXhAgudsRvW8cogiQxHtR
QR-Code:N38/56 Ef7oefU/vhvPQc2jqegywM/plJOmtKrt0rGUGEb7iV4pljdJZNuDqnN2lWBPPndfuCjCHGtv58Zk8IdurC0lKXBP9D/UhkaVyI2u2VFn87o1CoD5
QR-Code:N39/56 01qanWThaIbFXEI5CZrS4iNAHgKWNCRZYErFyv0yzno/z6xOqt4PZGka1fjXOiGNv4gKjUrSKSA5qEAIUc8aFHDsyZfOTB89cD+ZsqFV+Y58oAdP
QR-Code:N40/56 dFTO3b1IyliBppijeEKL42dpg6dOnMgbaVviUj2rU1M/jS2aOSiIQ4trTAtYB3mH2xtMWLUnWxvpCwp84qzcrIJJ2D2d5kBLAnOmaO76kHByuQ4c
QR-Code:N41/56 hfIUA/qgbJqzls2a8lROQqf0w6uyr97tUQN3N3WCvXGmi/6zZpXjHdzOPdBROVuU538a06gcrGzJA1NZd1Qk0+FlqYLxS46hxsu8ZnDpRQdqdJKY
QR-Code:N42/56 9Ni5xQ+JmBOIkTStJ5+fsiX1tDyxvXcO0QZJI175mgc+Es7WiUEA5EqdDZYWpaW5QGpz3XE9b6jGxMWn+Q9asNci0tQVumd6eALMoULv5fB63Ppf
QR-Code:N43/56 Lvq9amXu0MM1It5MM6AQi5No0lJLR26UKVktZGZCrzQpnz3mo6ZOtND04UPS2y3t08RRPaIvY/FtVuKOBgxwmqLZmyKER094YtorPVwrJTlAKqPE
QR-Code:N45/56 hfKs8hdSzc8bC8f6SDtzTNIM7nXrnujquY5jaRlK8Tr6FLNo9wqHB0xx2Zz85g+1ZKep0TjgtBnZXpT0FrpAgxV4Wj8oy5v68xFQtFrZ+9o9VTir
QR-Code:N46/56 qSEMcFjHj9Hc9OSVdDyjKrTTEXLBZ1RtoUhkjEv/fnc63GnuM69KM7sxmM0fVJ5YlrJDG/m8ox7K1DOcjrXodXnG5vUa9YI9qq9G5KY3aoWcUMvb
QR-Code:N47/56 GN+WYMrc2jBofSOCQM9GZN4qgNcVpnkQAsYyX0vZrKF1qXeNAnWXOnYr57BmBtGs6ScDHHk2P2QXW7Mdpx8sEtM0F2UTimacS9FUUGwCwlHoqsiy
QR-Code:N48/56 2ehekGykZ3DtME4gwDxXECnpREtUedrnkMOTVlqi5VLzbAINLwyf3+1gAsDjvom6LHJUChCDdJ75oqPmtSv+30ti01Cxo9N5MuhnhYsmQAvvJBv0
```

jerabaul29, 2021-09-02:
Very nice project. I discovered this just now, working on a kind of similar but "bash only" tool to do the same, just FYI :) . The bash-only motivation was to reduce the software stack and make installation easier (no python envs etc to manage), but definitely bash is not the best programming language ^^ .

See: https://github.com/jerabaul29/qrdump .

jerabaul29, 2021-09-02:
I have not seen problems with error rates, but maybe just not tested enough.

za3k, 2021-09-02:
That's interesting jerabaul, but let's keep things on-topic to this specific bug. Feel free to open an issue in /my/ bug  tracker if you want to talk.

za3k, 2022-05-02:
I had some personal correspondence with Timothy Terriberry (author of the QR code inside zbar) and he determined that the problem is that the PDF is being significantly downscaled before it gets to the QR decoder:

> Oh, I misunderstood the report. This sounds more likely to be an issue with zbar's PDF handling (I didn't even know it handled PDFs! ...hence the misunderstanding). For example, the image I attached (created using your convert command from the issue) is 2079x2500. The one handed to qrdec from the PDF is 499x600.
> 
> FWIW, it looks like it is code 13/13 that is getting slightly misaligned in the middle at that resolution, which causes the decoding error. Given that most rows are two pixels high, but four of the rows are only one pixel high, that seems inevitable, and frankly the decoder is lucky it was able to decode as many codes as it did with this input.

I will look into this as soon as I have time, but I suspect a lot of these bugs are duplicates now that I understand what's going on better

za3k, 2022-09-07:
OK with my current settings, I ran 1000 samples, and tried to png conversion. After conversion, I had a 100% success rate. My issue seems to be definitely a pdf-only issue. 

@misaki-web it seems like you had a reproduction case in png only. I think we should split these into two bugs.

closed 2026-09-30 by mchehab