mchehab/zbar #186: Missing DLLs for Windows runtime

Ortega-Dan, 2021-07-08:
When running on Windows it is asking for magickWand DLL and many more DLLs. 
Do we know if there is an easy way to get them all ?
[image]
[image]
[image]
And many more.
Or is there a way to get a static build not dependent of other DLLs ?

_Originally posted by @Ortega-Dan in https://github.com/mchehab/zbar/issues/56#issuecomment-876099523_

jamesmacwhite, 2021-07-09:
An easier way is to install the zbar package through MSYS2

https://packages.msys2.org/package/mingw-w64-x86_64-zbar

Ortega-Dan, 2021-07-10:
> An easier way is to install the zbar package through MSYS2
> 
> https://packages.msys2.org/package/mingw-w64-x86_64-zbar
> 
> I have found the binaries and building from source to be broken foe Windows. The building on Windows steps requires MSYS2 anyway, you can install it and then run pacman to install zbar and the exe should work fine. You will need to run zbarcam.exe from a MINGW shell, for the binary to be in the PATH. Otherwise you can manually run it from C:\msys64\mingw64\bin or 32 if not on a 64-bit machine.

Or just add C:\[...]mingw64\bin to Path environment variable.

Ortega-Dan, 2021-07-10:
@jamesmacwhite Thank you very much for the pointer. 

I've tried several times to build it, and it seems like it built successfully but then had troubles finding delegates for image formats processing and some other stuff. This is the very first time that I've got it to work successfully on Windows. 

Great help :+1:

jamesmacwhite, 2021-07-10:
@Ortega-Dan no worries. I found similar issues. The prebuilt package is much easier with MSYS2 and pacman I did get a build of zbarcam compiled but still got dll errors and took the easy way out in the end. I get the impression some of building from source info is outdated looking at other GitHub issues.

Adding ming64\bin to PATH is something you can do as you say.

Obviously the downside is using the prebuilt means the version might be slightly outdated. I think though that zbarcam binary issue on Windows that @itsmattkc found is fixed in the pacman package.

rollingmoai, 2022-07-20:
For QR Codes, I have found https://github.com/sayanarijit/qrscan to work well for me.

Elemecca, 2023-10-16:
Yeah, the existing build needs a lot of finagling to make it build actually static executables that only depend on DLLs that ship with Windows rather than needing an MSYS2 installation with a bunch of packages.

Here are the instructions I wrote for my own project on how to build an actual statically-linked copy of `zbarimg`. The resulting executable is about 6 MB.

# Windows Build

We build our own `zbarimg.exe` because the official ones aren't statically linked and depend on a lot of DLLs that they don't ship with.

The build is done on Windows in an [MSYS2] UCRT64 environment.

## Install Dependencies

```plain
pacman -S \
    mingw-w64-ucrt-x86_64-autotools \
    mingw-w64-ucrt-x86_64-gcc \
    mingw-w64-ucrt-x86_64-libjpeg-turbo \
    mingw-w64-ucrt-x86_64-libpng \
    mingw-w64-ucrt-x86_64-libwebp \
    mingw-w64-ucrt-x86_64-zlib \
    upx
```

## GraphicsMagick

The ImageMagick package in MSYS2 has broken static archives, and the GraphicsMagick package doesn't have static archives at all, so we have to build our own. We use GraphicsMagick because it's a little easier to build and results in a smaller final executable.

Download and extract the source tarball from [GraphicsMagick], then `cd` into its root directory.

```plain
./configure \
    --prefix=$(pwd)/../install \
    PKG_CONFIG_PATH=$(pwd)/../install/lib/pkgconfig \
    --disable-shared \
    --enable-static \
    --disable-dependency-tracking \
    --disable-installed \
    --disable-openmp \
    --without-magick-plus-plus \
    --without-threads \
    --without-bzlib \
    --without-dps \
    --without-gdi32 \
    --without-jbig \
    --without-heif \
    --without-jp2 \
    --without-jxl \
    --without-lcms2 \
    --without-lzma \
    --without-tiff \
    --without-trio \
    --without-ttf \
    --without-wmf \
    --without-xml \
    --without-zstd \
    --without-x
    
make
make install
```

It might be nice to remove:
- `--without-heif` for iPhone photo (HEIF) support
- `--without-xml`  for SVG support

They're currently disabled because their MSYS2 packages lack static archives.

## ZBar

Download and extract the source tarball from [ZBar], then `cd` into its root directory.

```plain
autoreconf -vfi

./configure \
    --prefix=$(pwd)/../install \
    PKG_CONFIG_PATH=$(pwd)/../install/lib/pkgconfig \
    --disable-dependency-tracking \
    --disable-shared \
    --enable-static \
    --disable-rpath \
    --disable-nls \
    --disable-pthread \
    --disable-doc \
    --disable-video \
    --with-graphicsmagick \
    --without-python \
    --without-java \
    --without-gtk \
    --without-qt
    
make LDFLAGS=-all-static
make install
```

We really need to pass `-all-static` to `libtool` in order to produce a static executable: with just `-static` it only statically links local libraries, but still dynamically links any DLLs that are already installed. However, passing `LDFLAGS=-all-static` to `./configure` makes the gcc detection fail, so we pass it to `make` instead where it can't affect the compiler checks.

## Make it Smaller

We can use UPX to shrink the executable significantly:

```plain
upx --best ../install/bin/zbarimg.exe
```

## Copy the Files

Set `TARGET` to wherever you want your installation to live.

```plain
mkdir -p $TARGET/magick
cp ../install/bin/zbarimg.exe $TARGET/
cp -r ../install/{lib,share}/GraphicsMagick-*/config/* $TARGET/magick/
```

You will need to set the environment variable `MAGICK_CONFIGURE_PATH` to `$TARGET/magick` at runtime if you want to use GraphicsMagick delegate-based file formats like PDF.

[MSYS2]: https://www.msys2.org/
[GraphicsMagick]: http://www.graphicsmagick.org/download.htm,
[ZBar]: https://github.com/mchehab/zbar

sitiom, 2026-10-11:
I have created a PR for this:
- https://github.com/mchehab/zbar/pull/344

sitiom, 2026-10-11:
@mchehab Can you mark https://github.com/mchehab/zbar/releases/tag/0.23.95 as a release so this issue can be closed? Thanks!