mchehab/zbar #192: Updating meta::cpan Barcode::Zbar

thomasSDK, 2021-08-25:
The Perl interface on meta::cpan is from before the project was forked and is thus ancient.

Would it be possible to take ownership of that and updated it to match the perl interface that is in this repo?

timlegge, 2022-02-03:
Hi @mchehab 

I recently asked and was granted "comaint" access to the Barcode::ZBar module on cpan and would like to upload the current version from the master branch.  The current version on cpan is from Oct 23, 2009 and does not build against the current zbar version.

When I asked the cpan admins (who managed to contact Jeff Brown (SPADIX) who approved) to grant me access I also stated that while I volunteer to do it if there is someone already in the ZBar community that has a pause account and is familiar with uploading perl releases to cpan I am happy for them to do it instead.  That person would simply need to contact the cpan admins and I am sure approval could be granted.

In lieu of direction from you to the contrary I would plan to basically package and release the Module as it is today when I get time.  I have done a little work on modernizing the packaging of the perl module if you are receptive to a Pull Request I can do that.

Tim

timlegge, 2022-02-18:
This issue can be closed.  The CPAN module has been updated