mchehab/zbar #194: zbar cannot scan this rather obvious QR code

vi, 2021-09-01:
Zxing barcode scanner on smartphone scans this one well:

[image]

I tried manually improving the code in raster editor to make it look more like the canonical variant (that zbar handles well):

[image]

, but `zbarimg` still fails to find anything.

Version 0.23.90-1 from Debian.

vi, 2021-09-01:
Downscaling the image makes zbar correctly decode the sample.

maslick, 2022-01-30:
My OSS project named [Koder](https://github.com/maslick/koder) which is based on zbar 0.23.90 (compiled into WebAssembly) is able to decode. Check up the demo https://qr.maslick.tech