mchehab/zbar #195: release: provide also as a linux packet?

jerabaul29, 2021-09-02:
It would be great to provide the release also as a linux packet, like a .deb or a .rpm or similar. That would allow linux users to update to the most recent release without the need to compile themselves (for example, without this possibility, not sure how I can get a recent zbarimg in Ubuntu 20.04 without compiling from source, which sounds a bit painful? ...).

jerabaul29, 2021-09-02:
Note for myself: Ubuntu 21.04 may have a relatively modern packet here: https://www.ubuntuupdates.org/package/core/hirsute/universe/base/zbar-tools .

petterreinholdtsen, 2021-10-05:
The versions available in Debian can be tracked via https://tracker.debian.org/zbar .

jerabaul29, 2021-10-05:
I only find code snapshots there (but I may just be missing something :) ), is there a link to a .deb that I can just ```sudo apt install``` without more to do?

duck-rh, 2022-02-21:
@jerabaul29 sorry, I tried not to sound patronizing but I'm not a native English speaker so I guess I got it wrong. I don't see anything wrong in being a beginner, that's why I gave some commands.

Indeed I misunderstood your goal. I don't know Ubuntu that well but in Debian you can still get newer versions through backports and for most people that's probably easier than manually and regularly checking the repo and downloading since you would get the upgrades automatically after the first time. I think that's easier for upstream authors if they can collaborate with the distro instead of having to provide so many flavors themselves and handle breakage when some distro-wide change happens etc. Now as a DD myself I guess I'm a little biased :-).

jerabaul29, 2022-04-25:
No worries, thanks :) .