mchehab/zbar #197: Can this application be compiled on Windows 7?

AnonimFan, 2021-09-11:
Since I am running Windows 7, and there's no pre-built binaries for zbar-23.92 or above(I am tryna to decode a qr-code with binary data inside), I tried compiling, only to see that it can only be compiled on Linux.

Can it be compiled on Windows 7 without **WSL**?

ssrlive, 2021-09-12:
Well, here is the path.

1. download and install VS 2019, following [this link](https://visualstudio.microsoft.com/downloads/)
2. install [cmake](https://cmake.org/)
3. download / install [git](https://git-scm.com/) and install `vcpkg` on `D disk`
```
d:
cd /
git clone https://github.com/microsoft/vcpkg.git
cd d:/vcpkg
./bootstrap-vcpkg.bat

./vcpkg integrate install

./vcpkg install libpng libpng:x64-windows
./vcpkg install libiconv libiconv:x64-windows
./vcpkg install zlib zlib:x64-windows
```
4. clone zbar source code
```
d:
cd /
git clone https://github.com/mchehab/zbar.git
```
5. download [this zip file](https://github.com/mchehab/zbar/files/7149052/zbar-cmake-msvc.zip) and extract all files and folders to the `zbar` root folder (d:/zbar).
6. use `cmake` make the build environment.
```
d:
cd d:/zbar/build
cmake .. -DCMAKE_TOOLCHAIN_FILE=D:/vcpkg/scripts/buildsystems/vcpkg.cmake
```
7. open `Visual Studio 2019 Developer Command Prompt`, then build zbar with it.
```
d:
cd d:/zbar/build
devenv zbar.sln /Build
```
8. done

AnonimFan, 2021-09-12:
Followed the steps, but, I get a lot, and I mean a lot

Build log:
[log.txt](https://github.com/mchehab/zbar/files/7149611/log.txt)

ssrlive, 2021-09-12:
you must `cd c:/zbar\build` and run `cmake`.

EliaTolin, 2021-11-19:
Hello, is it possible compile for win32?

ssrlive, 2021-11-19:
Yes https://github.com/ShadowsocksR-Live/zbar/blob/master/README-windows.md#building-with-cmake-on-windows @EliaTolin

EliaTolin, 2021-11-19:
> Yes https://github.com/ShadowsocksR-Live/zbar/blob/master/README-windows.md#building-with-cmake-on-windows @EliaTolin

Yes but i need the version x86, not x64. Is it a fork with x86?

ssrlive, 2021-11-19:
https://stackoverflow.com/questions/28350214/how-to-build-x86-and-or-x64-on-windows-from-command-line-with-cmake

EliaTolin, 2021-11-19:
Thanks so much!

rollingmoai, 2022-07-20:
@ssrlive Please create a PR to this repo for CMake support. Thanks!

sc0rp10n-py, 2022-11-04:
Hey @ssrlive I tried your cmake method in MSYS2 in windows 10
I got this error.
```
[ 50%] Building C object CMakeFiles/zbar.dir/zbar/video/v4l.c.obj
[ 52%] Building C object CMakeFiles/zbar.dir/zbar/video/v4l2.c.obj
/c/Users/princ/Downloads/projects/freelancing/fiverr/trash/zbar/zbar/video/v4l2.c:50:10: fatal error: linux/videodev2.h: No such file or directory
   50 | #include <linux/videodev2.h>
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/zbar.dir/build.make:370: CMakeFiles/zbar.dir/zbar/video/v4l2.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:87: CMakeFiles/zbar.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

ssrlive, 2022-11-04:
Please use Visual Studio 2019.

sc0rp10n-py, 2022-11-04:
@ssrlive 
I got this error while installing vcpkg

```
vcpkg install libjpeg-turbo libjpeg-turbo:x64-windows
```
error
```
-- Configuring x64-windows
CMake Error at scripts/cmake/vcpkg_execute_required_process.cmake:96 (message):
    Command failed: C:/vcpkg/downloads/tools/ninja/1.10.2-windows/ninja.exe -v
    Working Directory: C:/vcpkg/buildtrees/libjpeg-turbo/x64-windows-rel/vcpkg-parallel-configure
    Error code: 1
    See logs for more information:
      C:\vcpkg\buildtrees\libjpeg-turbo\config-x64-windows-CMakeCache.txt.log
      C:\vcpkg\buildtrees\libjpeg-turbo\config-x64-windows-CMakeCache.txt.log
      C:\vcpkg\buildtrees\libjpeg-turbo\config-x64-windows-out.log

Call Stack (most recent call first):
  installed/x64-windows/share/vcpkg-cmake/vcpkg_cmake_configure.cmake:247 (vcpkg_execute_required_process)
  ports/libjpeg-turbo/portfile.cmake:40 (vcpkg_cmake_configure)
  scripts/ports.cmake:147 (include)

error: building libjpeg-turbo:x64-windows failed with: BUILD_FAILED
error: Please ensure you're using the latest port files with `git pull` and `vcpkg update`.
Then check for known issues at:
    https://github.com/microsoft/vcpkg/issues?q=is%3Aissue+is%3Aopen+in%3Atitle+libjpeg-turbo
You can submit a new issue at:
    https://github.com/microsoft/vcpkg/issues/new?template=report-package-build-failure.md&title=[libjpeg-turbo]+Build+error
Include '[libjpeg-turbo] Build error' in your bug report title, the following version information in your bug description, and attach any relevant failure logs from above.
    vcpkg-tool version: 2022-10-17-3247920fbdd47d08f36cbd480addd9890d3c2435
    vcpkg-scripts version: a4013afe6 2022-11-03 (13 hours ago)
```