mchehab/zbar #203: Extend zbar to recognize QR codes created by swish.nu?

petterreinholdtsen, 2021-10-05:
I just visited https://www.swish.nu/skapa-qr-kod and created the attached QR code.  When I asked zbarimg to give me the QR code content, I got this message instead:

```
scanned 0 barcode symbols from 1 images in 0.21 seconds

WARNING: barcode data was not detected in some image(s)
Things to check:
  - is the barcode type supported? Currently supported symbologies are:
        . EAN/UPC (EAN-13, EAN-8, EAN-2, EAN-5, UPC-A, UPC-E, ISBN-10, ISBN-13)
        . DataBar, DataBar Expanded
        . Code 128
        . Code 93
        . Code 39
        . Codabar
        . Interleaved 2 of 5
        . QR code
        . SQ code
  - is the barcode large enough in the image?
  - is the barcode mostly in focus?
  - is there sufficient contrast/illumination?
  - If the symbol is split in several barcodes, are they combined in one image?
  - Did you enable the barcode type?
    some EAN/UPC codes are disabled by default. To enable all, use:
    $ zbarimg -S*.enable <files>
    Please also notice that some variants take precedence over others.
    Due to that, if you want, for example, ISBN-10, you should do:
    $ zbarimg -Sisbn10.enable <files>
```

Can zbarimg be improved to recognize this QR code?  This was using zbar 0.23.90-1 in Debian Bullseye.

[image]

maslick, 2022-01-30:
My OSS project named Koder which is based on zbar 0.23.90 (compiled into WebAssembly) is able to decode this QR. Check out the demo here: https://qr.maslick.tech

maslick, 2022-01-30:
[image]
[image]

petterreinholdtsen, 2022-02-02:
Can it also decode it if you give it the image directly, instead of pointing a camera at the image?

maslick, 2022-02-02:
For this you need to find the best spot (angle, focus, distance, etc.), which is easier to do live, and not by feeding the algorithm just 1 image, even in great resolution. You can use this script and experiment with a number of samples (focus, distance, image quality):

```
npm i @maslick/koder
```

```js
const Koder = require('@maslick/koder');
const {loadImage, createCanvas} = require("canvas");

const getImageData = async (src) => {
  const img = await loadImage(src);
  const canvas = createCanvas(img.width, img.height);
  const ctx = canvas.getContext('2d');
  ctx.drawImage(img, 0, 0);
  return {
    data: ctx.getImageData(0, 0, img.width, img.height).data,
    width: img.width,
    height: img.height
  };
};

(async () => {
  const url = 'https://user-images.githubusercontent.com/731031/135975403-321a58d8-a64d-4fc1-93fb-a38144685b69.png';
  const koder = await new Koder().initialized;
  const {data, width, height} = await getImageData(url);

  const t0 = new Date().getTime();
  const res = koder.decode(data, width, height);
  const t1 = new Date().getTime();

  console.log(`Scanned in ${t1-t0} ms`);
  console.log(res);
})();
```

petterreinholdtsen, 2022-02-02:
[Pavel Maslov]
> For this you need to find the best spot (angle, focus, distance,
> etc.), which is easier to do live, and not by feeding the algorithm
> just 1 image, even in great resolution. You can use this script and
> experiment with a number of samples (focus, distance, image quality):

I was just curious how your program work with one image in perfect
focus, as it is the setup I am interested in, ie running QR code
detection on prescanned images.

--
Happy hacking
Petter Reinholdtsen