mchehab/zbar #219: crash on windows while decoding QR-code

kameamea, 2022-05-04:
When I try to decode the attached image with the zbar library under windows (built by using https://github.com/dani4/ZBarWin64.git ), I get this error:

Debug-Error!

Program: ...libzbar64-0.dll

Run-Time Check Failure #2: Stack around the variable 'seq' was corrupted

Debugging shows, that the following line in databar.cpp:663 in the function 
static inline unsigned lookup_sequence(databar_segment_t *seg, int fixed, int seq[22])

 `   seq[n] = -1;`

is hit with n=22 while seq is an array of size 22.

So my solution is to surround this line by

`if (n<22) {seq[n]=-1;}`

which solves the problem. As this method is not documented and I don't really understand whats going on there, I'm not sure, whether this is the appropriate fix.

FalsinSoft, 2023-01-13:
Hi

Me too found this problem that happens very frequently only by process  frames, no need to have a qrcode inside. This issue has been reported in another post here but seems still unfixed. Can you confirm your fix work without problems?

Thank you

kameamea, 2023-01-14:
I haven't used it lately, but also never saw any problems with my fix. Sorry that's all I can say.

FalsinSoft, 2023-01-15:
Don't worry, thank you to take time to reply me. After a quick check of the code it seems -1 is used as marker for some condition than I tried to increase the size of seq array (also in the calling function) to 23. I'm testing the change, It's really strange nobody noted this problem before...

dg0yt, 2024-12-15:
Fixed by https://github.com/mchehab/zbar/commit/012a030250a203e5529d09caedea7ad7173dacfd?